coolsnowwolf/lede #10723: make[2] -C package/utils/util-linux download 下载不动，

yuzhe1129, 2023-01-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
make download -j8  下载dl库， make[2] -C package/utils/util-linux download 一直卡这不动，
make[2] -C package/libs/uclient download
 make[2] -C package/libs/ustream-ssl download
 make[2] -C package/libs/zlib download
 make[2] -C package/network/config/firewall download
 make[2] -C package/network/config/netifd download
 make[2] -C package/network/config/swconfig download
 make[2] -C package/network/services/dnsmasq download
 make[2] -C package/network/services/dropbear download
 make[2] -C package/network/services/fullconenat download
 make[2] -C package/network/services/ppp download
 make[2] -C package/network/services/shellsync download
 make[2] -C package/network/services/uhttpd download
 make[2] -C package/network/utils/bpftools download
 make[2] -C package/network/utils/iproute2 download
 make[2] -C package/network/utils/ipset download
 make[2] -C package/network/utils/iptables download
 make[2] -C package/network/utils/iwinfo download
 make[2] -C package/network/utils/wireless-tools download
 make[2] -C package/system/ca-certificates download
 make[2] -C package/system/fstools download
 make[2] -C package/system/fwtool download
 make[2] -C package/system/mtd download
 make[2] -C package/system/openwrt-keyring download
 make[2] -C package/system/opkg download
 make[2] -C package/system/rpcd download
 make[2] -C package/system/procd download
 make[2] -C package/system/ubox download
 make[2] -C package/system/ubus download
 make[2] -C package/system/uci download
 make[2] -C package/system/urandom-seed download
 make[2] -C package/system/urngd download
 make[2] -C package/system/usign download
 make[2] -C package/utils/busybox download
 make[2] -C package/utils/bzip2 download
 make[2] -C package/utils/jsonfilter download
 make[2] -C package/utils/lua download
 make[2] -C package/utils/mtd-utils download
 make[2] -C package/utils/util-linux download
^C    ERROR: package/feeds/packages/boost failed to build.
make: *** [/home/c/lede/include/toplevel.mk:203: download] 中断

然后make -j8 download V=s     就是在time: package/utils/util-linux/download#0.26#0.05#0.34
  2  113M    2 3423k    0     0  12260      0  2:41:29  0:04:45  2:36:44    这下载很久
插件SSR 节点没问题的，请问怎么解决啊
make[2]: Leaving directory '/home/c/lede/package/system/uci'
time: package/system/uci/download#0.12#0.08#0.38
make[2]: Entering directory '/home/c/lede/package/utils/jsonfilter'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/system/ubox'
time: package/system/ubox/download#0.31#0.07#0.69
make[2]: Entering directory '/home/c/lede/package/utils/lua'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/system/urandom-seed'
time: package/system/urandom-seed/download#0.11#0.10#0.48
make[2]: Entering directory '/home/c/lede/package/utils/mtd-utils'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/system/urngd'
time: package/system/urngd/download#0.12#0.11#0.51
make[2]: Entering directory '/home/c/lede/package/utils/util-linux'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/utils/jsonfilter'
time: package/utils/jsonfilter/download#0.13#0.09#0.36
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/utils/busybox'
time: package/utils/busybox/default/download#0.20#0.06#0.47
  0  113M    0 98053    0     0  10269      0  3:12:48  0:00:09  3:12:39 13454make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/system/usign'
time: package/system/usign/download#0.21#0.10#0.60
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/utils/bzip2'
time: package/utils/bzip2/download#0.20#0.11#0.53
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/utils/lua'
time: package/utils/lua/download#0.15#0.16#0.45
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/utils/mtd-utils'
time: package/utils/mtd-utils/download#0.14#0.05#0.34
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/utils/util-linux'
time: package/utils/util-linux/download#0.26#0.05#0.34
  2  113M    2 3423k    0     0  12260      0  2:41:29  0:04:45  2:36:44 12274^Cmake[2]: *** [Makefile:510: /home/c/lede/dl/boost_1_81_0.tar.bz2] Interrupt
time: package/feeds/packages/boost/download#0.24#0.10#287.70
make: *** [/home/c/lede/include/toplevel.mk:203: download] 中断

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

1054009064, 2023-01-03:
需要 `make download -j1 V=s`

yuzhe1129, 2023-01-03:
> make download -j1 V=s

time: package/network/services/shellsync/download#0.11#0.03#0.13
make[2]: Entering directory '/home/c/lede/package/network/services/uhttpd'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/network/services/uhttpd'
time: package/network/services/uhttpd/download#0.12#0.03#0.19
make[2]: Entering directory '/home/c/lede/package/network/utils/bpftools'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/network/utils/bpftools'
time: package/network/utils/bpftools/lib/download#0.13#0.05#0.19
make[2]: Entering directory '/home/c/lede/package/network/utils/iproute2'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/network/utils/iproute2'
time: package/network/utils/iproute2/ipfull/download#0.15#0.07#0.23
make[2]: Entering directory '/home/c/lede/package/network/utils/ipset'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/network/utils/ipset'
time: package/network/utils/ipset/download#0.12#0.04#0.15
make[2]: Entering directory '/home/c/lede/package/network/utils/iptables'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/network/utils/iptables'
time: package/network/utils/iptables/download#0.13#0.04#0.18
make[2]: Entering directory '/home/c/lede/package/network/utils/iwinfo'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/network/utils/iwinfo'
time: package/network/utils/iwinfo/download#0.11#0.03#0.15
make[2]: Entering directory '/home/c/lede/package/network/utils/wireless-tools'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/network/utils/wireless-tools'
time: package/network/utils/wireless-tools/download#0.13#0.04#0.19
make[2]: Entering directory '/home/c/lede/package/system/ca-certificates'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/system/ca-certificates'
time: package/system/ca-certificates/download#0.10#0.04#0.14
make[2]: Entering directory '/home/c/lede/package/system/fstools'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/system/fstools'
time: package/system/fstools/download#0.17#0.04#0.20
make[2]: Entering directory '/home/c/lede/package/system/fwtool'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/system/fwtool'
time: package/system/fwtool/download#0.16#0.05#0.19
make[2]: Entering directory '/home/c/lede/package/system/mtd'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/system/mtd'
time: package/system/mtd/download#0.11#0.04#0.16
make[2]: Entering directory '/home/c/lede/package/system/openwrt-keyring'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/system/openwrt-keyring'
time: package/system/openwrt-keyring/download#0.12#0.03#0.15
make[2]: Entering directory '/home/c/lede/package/system/opkg'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/system/opkg'
time: package/system/opkg/download#0.24#0.06#0.29
make[2]: Entering directory '/home/c/lede/package/system/procd'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/system/procd'
time: package/system/procd/default/download#0.15#0.05#0.19
make[2]: Entering directory '/home/c/lede/package/system/rpcd'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/system/rpcd'
time: package/system/rpcd/download#0.14#0.03#0.17
make[2]: Entering directory '/home/c/lede/package/system/ubox'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/system/ubox'
time: package/system/ubox/download#0.21#0.06#0.26
make[2]: Entering directory '/home/c/lede/package/system/ubus'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/system/ubus'
[... 28 lines cut ...]
time: package/utils/jsonfilter/download#0.12#0.04#0.15
make[2]: Entering directory '/home/c/lede/package/utils/lua'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/utils/lua'
time: package/utils/lua/download#0.14#0.04#0.16
make[2]: Entering directory '/home/c/lede/package/utils/mtd-utils'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/utils/mtd-utils'
time: package/utils/mtd-utils/download#0.10#0.05#0.15
make[2]: Entering directory '/home/c/lede/package/utils/util-linux'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/c/lede/package/utils/util-linux'
time: package/utils/util-linux/download#0.22#0.08#0.30
make[1]: Leaving directory '/home/c/lede'
make[1]: Entering directory '/home/c/lede'
make[2]: Entering directory '/home/c/lede/target/linux'
make[3]: Entering directory '/home/c/lede/target/linux/ramips'
make[3]: Nothing to be done for 'download'.
make[3]: Leaving directory '/home/c/lede/target/linux/ramips'
make[2]: Leaving directory '/home/c/lede/target/linux'
time: target/linux/download#0.06#0.05#0.13
make[1]: Leaving directory '/home/c/lede'
c@ubuntu:~/lede$ 

这算下载成功dl库？

closed 2023-01-03 by yuzhe1129