coolsnowwolf/lede #10724: X86 6.1 编译出错

baodp, 2023-01-04:
X86 6.1 编译出错
  默认配置 编译 ，只多点选了  IPV6。

更新今天的更新后编译错误如下：

Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.2-1-48e942db8795e8c80aeceb205c9f4d48) for kmod-button-hotplug
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-button-hotplug found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-button-hotplug.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.2-1-48e942db8795e8c80aeceb205c9f4d48) for kmod-cryptodev
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-cryptodev found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-cryptodev.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.2-1-48e942db8795e8c80aeceb205c9f4d48) for kmod-usb-net-rtl8152-vendor
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-usb-net-rtl8152-vendor found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-usb-net-rtl8152-vendor.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.2-1-48e942db8795e8c80aeceb205c9f4d48) for kmod-r8168
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-r8168 found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-r8168.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.2-1-48e942db8795e8c80aeceb205c9f4d48) for kmod-shortcut-fe
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-shortcut-fe found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-shortcut-fe.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.2-1-48e942db8795e8c80aeceb205c9f4d48) for kmod-shortcut-fe-cm
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-shortcut-fe-cm found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-shortcut-fe-cm.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.2-1-48e942db8795e8c80aeceb205c9f4d48) for kmod-ipt-fullconenat
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ipt-fullconenat found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-ipt-fullconenat.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ddns-scripts_aliyun:
 * 	kernel (= 6.1.2-1-48e942db8795e8c80aeceb205c9f4d48)
 * opkg_install_cmd: Cannot install package ddns-scripts_aliyun.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for default-settings:
 * 	kernel (= 6.1.2-1-48e942db8795e8c80aeceb205c9f4d48)
 * opkg_install_cmd: Cannot install package default-settings.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-turboacc:
 * 	kernel (= 6.1.2-1-48e942db8795e8c80aeceb205c9f4d48)
 * opkg_install_cmd: Cannot install package luci-app-turboacc.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/bdp/lede'
make[1]: *** [package/Makefile:111: /home/bdp/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/bdp/lede'
make: *** [/home/bdp/lede/include/toplevel.mk:231：world] 错误 2

GreatMichaelLee, 2023-01-04:
一样问题，但报错位置不同，具体见输出

Configuring libnl-genl200.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.2-1-0c3e5cb8a0e5a7787e4e2de58346ac2e) for kmod-cryptodev
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-cryptodev found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for pgyvpn:
 *      kernel (= 6.1.2-1-0c3e5cb8a0e5a7787e4e2de58346ac2e)
 * opkg_install_cmd: Cannot install package pgyvpn.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.2-1-0c3e5cb8a0e5a7787e4e2de58346ac2e) for kmod-ipt-geoip
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ipt-geoip found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-ipt-geoip.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.2-1-0c3e5cb8a0e5a7787e4e2de58346ac2e) for kmod-ipt-lua
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ipt-lua found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-ipt-lua.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.2-1-0c3e5cb8a0e5a7787e4e2de58346ac2e) for kmod-button-hotplug
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-button-hotplug found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-button-hotplug.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.2-1-0c3e5cb8a0e5a7787e4e2de58346ac2e) for kmod-fs-ksmbd
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-fs-ksmbd found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-fs-ksmbd.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.2-1-0c3e5cb8a0e5a7787e4e2de58346ac2e) for kmod-r8125
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-r8125 found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-r8125.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.2-1-0c3e5cb8a0e5a7787e4e2de58346ac2e) for kmod-usb-net-rtl8152-vendor
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-usb-net-rtl8152-vendor found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-usb-net-rtl8152-vendor.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.2-1-0c3e5cb8a0e5a7787e4e2de58346ac2e) for kmod-r8168
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-r8168 found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-r8168.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.2-1-0c3e5cb8a0e5a7787e4e2de58346ac2e) for kmod-ipt-fullconenat
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ipt-fullconenat found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-ipt-fullconenat.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for default-settings:
 *      kernel (= 6.1.2-1-0c3e5cb8a0e5a7787e4e2de58346ac2e)
 * opkg_install_cmd: Cannot install package default-settings.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ksmbd-server:
 *      kernel (= 6.1.2-1-0c3e5cb8a0e5a7787e4e2de58346ac2e)
 * opkg_install_cmd: Cannot install package ksmbd-server.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for iptables-mod-geoip:
 *      kernel (= 6.1.2-1-0c3e5cb8a0e5a7787e4e2de58346ac2e)
 * opkg_install_cmd: Cannot install package iptables-mod-geoip.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for iptables-mod-lua:
 *      kernel (= 6.1.2-1-0c3e5cb8a0e5a7787e4e2de58346ac2e)
 * opkg_install_cmd: Cannot install package iptables-mod-lua.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:111: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231：world] 错误 2
builder@Build-Server:/home/builder/lede_x86$

WYC-2020, 2023-01-04:
一看就是接到以前的编译的，在进行编译，删除linux 重新编译

qqsir-dev, 2023-01-04:
云编译的也出错：
make[2] -C package/system/opkg host-compile
 make[2] -C package/libs/libubox compile
 make[2] -C package/kernel/linux compile
    ERROR: package/kernel/linux failed to build.
make -r package/compile: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/runner/work/AutoLede/AutoLede/openwrt/include/toplevel.mk:231: package/compile] Error 1
。。。。。。。。。。。。。。。。。。。。
。。。。。。。。。。。。。。。。。。。。
touch /home/runner/work/AutoLede/AutoLede/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.i915-firmware_installed
echo "i915-firmware" >> /home/runner/work/AutoLede/AutoLede/openwrt/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
make[2]: Leaving directory '/home/runner/work/AutoLede/AutoLede/openwrt/package/firmware/linux-firmware'
time: package/firmware/linux-firmware/compile#26.79#4.51#31.03
make[2]: Entering directory '/home/runner/work/AutoLede/AutoLede/openwrt/package/kernel/linux'
Package kmod-drm-amdgpu is missing dependencies for the following libraries:
video.ko
make[2]: *** [modules/video.mk:346: /home/runner/work/AutoLede/AutoLede/openwrt/bin/targets/x86/64/packages/kmod-drm-amdgpu_6.1.2-1_x86_64.ipk] Error 1
make[2]: Leaving directory '/home/runner/work/AutoLede/AutoLede/openwrt/package/kernel/linux'
time: package/kernel/linux/compile#3.34#0.33#3.66
    ERROR: package/kernel/linux failed to build.
make[1]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[1]: Leaving directory '/home/runner/work/AutoLede/AutoLede/openwrt'
make: *** [/home/runner/work/AutoLede/AutoLede/openwrt/include/toplevel.mk:231: package/compile] Error 2
Error: Process completed with exit code 2.

qqsir-dev, 2023-01-04:
好像解决了！

closed 2023-01-04 by baodp