coolsnowwolf/lede #10727: 660x编译出错

286700234, 2023-01-04:
make[6]: *** [Makefile:473: xasprintf.o] Error 127
make[3]: Entering directory '/home/lyus/lede/tools/meson'
make[5]: *** [Makefile:527: all-recursive] Error 1
make[4]: *** [Makefile:443: all] Error 2
make[4]: Leaving directory '/home/lyus/lede/build_dir/host/elfutils-0.188'
make[3]: *** [Makefile:38: /home/lyus/lede/build_dir/host/elfutils-0.188/.built] Error 2
make[3]: Leaving directory '/home/lyus/lede/tools/elfutils'
time: tools/elfutils/compile#0.04#0.03#0.12
    ERROR: tools/elfutils failed to build.
make[2]: *** [tools/Makefile:207: tools/elfutils/compile] Error 1
make[2]: *** Waiting for unfinished jobs....
make[3]: Entering directory '/home/lyus/lede/tools/make-ext4fs'
make[3]: Leaving directory '/home/lyus/lede/tools/autoconf'
time: tools/autoconf/compile#0.05#0.02#0.11
make[3]: Leaving directory '/home/lyus/lede/tools/meson'
time: tools/meson/compile#0.03#0.02#0.06
make[3]: Leaving directory '/home/lyus/lede/tools/make-ext4fs'
time: tools/make-ext4fs/compile#0.04#0.02#0.07
make[2]: Leaving directory '/home/lyus/lede'
make[1]: *** [tools/Makefile:203: /home/lyus/lede/staging_dir/host/stamp/.tools_compile_nyyynyynnyyynynyyyyyyynyynyyyynynyyyyyyyyyyyyyyyyynynnyyyyyyy] Error 2
make[1]: Leaving directory '/home/lyus/lede'
make: *** [/home/lyus/lede/include/toplevel.mk:231：world] 错误 2

WYC-2020, 2023-01-05:
日志

coolsnowwolf, 2026-09-05:
当前 tools/elfutils 已更新到 0.192，并明确加入 `HOST_CFLAGS += -Wno-error -fPIC -std=gnu17`；旧 0.188 的宿主构建 Error 127 已不再对应当前工具包，关闭为 completed。

closed 2026-09-05 by coolsnowwolf