coolsnowwolf/lede #10731: 5.15.86同config同配置网络问题

sleel, 2023-01-05:
1.关于你要提交的问题
最新源码5.15.86同config同配置网络问题
Q：是否搜索了issue (使用 "x" 选择)

 没有类似的issue
2. 详细叙述
同config同配置文件，11月5.15.78内核的版本正常，最新源码5.15.86拨号后微信、懂球帝图片打不开或者很慢，客户端开VP恩正常，应该是dns解析有问题

(1) 具体问题
A：同config同配置文件，5.15.86和6.1内核都同样的问题。11月5.15.78正常。路由仅拨号，无passwall。换最新源码5.10.161内核正常。

(2) 路由器型号和固件版本
A：x86   内核5.15.86  PVE版本7.3-4

(3) 详细日志

Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.288569] ACPI: Enabled 2 GPEs in block 00 to 3F
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.291881] ACPI: PCI Root Bridge [PCI0] (domain 0000 [bus 00-ff])
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.292440] acpi PNP0A08:00: _OSC: OS supports [Segments MSI HPX-Type3]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.293988] acpi PNP0A08:00: _OSC: not requesting OS control; OS requires [ExtendedConfig ASPM ClockPM MSI]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.296401] PCI host bridge to bus 0000:00
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.296435] pci_bus 0000:00: root bus resource [io  0x0000-0x0cf7 window]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.297968] pci_bus 0000:00: root bus resource [io  0x0d00-0xffff window]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.299476] pci_bus 0000:00: root bus resource [mem 0x000a0000-0x000bffff window]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.300433] pci_bus 0000:00: root bus resource [mem 0x10000000-0xafffffff window]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.302207] pci_bus 0000:00: root bus resource [mem 0xc0000000-0xfebfffff window]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.303966] pci_bus 0000:00: root bus resource [mem 0x100000000-0x8ffffffff window]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.304434] pci_bus 0000:00: root bus resource [bus 00-ff]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.305790] pci 0000:00:00.0: [8086:29c0] type 00 class 0x060000
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.307889] pci 0000:00:01.0: [1234:1111] type 00 class 0x030000
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.312506] pci 0000:00:01.0: reg 0x10: [mem 0xfb000000-0xfbffffff pref]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.320335] pci 0000:00:01.0: reg 0x18: [mem 0xfea14000-0xfea14fff]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.332504] pci 0000:00:01.0: reg 0x30: [mem 0xfea00000-0xfea0ffff pref]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.334270] pci 0000:00:01.0: Video device with shadowed ROM at [mem 0x000c0000-0x000dffff]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.337202] pci 0000:00:1a.0: [8086:2937] type 00 class 0x0c0300
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.342891] pci 0000:00:1a.0: reg 0x20: [io  0x5040-0x505f]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.346062] pci 0000:00:1a.1: [8086:2938] type 00 class 0x0c0300
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.352439] pci 0000:00:1a.1: reg 0x20: [io  0x5060-0x507f]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.355711] pci 0000:00:1a.2: [8086:2939] type 00 class 0x0c0300
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.360439] pci 0000:00:1a.2: reg 0x20: [io  0x5080-0x509f]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.363648] pci 0000:00:1a.7: [8086:293c] type 00 class 0x0c0320
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.365404] pci 0000:00:1a.7: reg 0x10: [mem 0xfea15000-0xfea15fff]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.372797] pci 0000:00:1b.0: [8086:293e] type 00 class 0x040300
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.375124] pci 0000:00:1b.0: reg 0x10: [mem 0xfea10000-0xfea13fff]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.381774] pci 0000:00:1c.0: [1b36:000c] type 01 class 0x060400
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.385345] pci 0000:00:1c.0: reg 0x10: [mem 0xfea16000-0xfea16fff]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.392584] pci 0000:00:1c.1: [1b36:000c] type 01 class 0x060400
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.396441] pci 0000:00:1c.1: reg 0x10: [mem 0xfea17000-0xfea17fff]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.403648] pci 0000:00:1c.2: [1b36:000c] type 01 class 0x060400
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.406955] pci 0000:00:1c.2: reg 0x10: [mem 0xfea18000-0xfea18fff]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.413244] pci 0000:00:1c.3: [1b36:000c] type 01 class 0x060400
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.416440] pci 0000:00:1c.3: reg 0x10: [mem 0xfea19000-0xfea19fff]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.423664] pci 0000:00:1d.0: [8086:2934] type 00 class 0x0c0300
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.429709] pci 0000:00:1d.0: reg 0x20: [io  0x50a0-0x50bf]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.433201] pci 0000:00:1d.1: [8086:2935] type 00 class 0x0c0300
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.440922] pci 0000:00:1d.1: reg 0x20: [io  0x50c0-0x50df]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.446385] pci 0000:00:1d.2: [8086:2936] type 00 class 0x0c0300
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.452439] pci 0000:00:1d.2: reg 0x20: [io  0x50e0-0x50ff]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.455674] pci 0000:00:1d.7: [8086:293a] type 00 class 0x0c0320
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.457353] pci 0000:00:1d.7: reg 0x10: [mem 0xfea1a000-0xfea1afff]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.464728] pci 0000:00:1e.0: [8086:244e] type 01 class 0x060401
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.474388] pci 0000:00:1f.0: [8086:2918] type 00 class 0x060100
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.476493] pci 0000:00:1f.0: quirk: [io  0x0600-0x067f] claimed by ICH6 ACPI/GPIO/TCO
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.478692] pci 0000:00:1f.2: [8086:2922] type 00 class 0x010601
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.487375] pci 0000:00:1f.2: reg 0x20: [io  0x5100-0x511f]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.490803] pci 0000:00:1f.2: reg 0x24: [mem 0xfea1b000-0xfea1bfff]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.494635] pci 0000:00:1f.3: [8086:2930] type 00 class 0x0c0500
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.499889] pci 0000:00:1f.3: reg 0x20: [io  0x0700-0x073f]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.504316] pci 0000:00:1c.0: PCI bridge to [bus 01]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    0.504490] pci 0000:00:1c.0:   bridge window [mem 0xfe800000-0xfe9fffff]
[... 361 lines cut ...]
Thu Jan  5 15:41:19 2023 kern.notice kernel: [    1.502814] sd 6:0:0:0: [sda] 860160 512-byte logical blocks: (440 MB/420 MiB)
Thu Jan  5 15:41:19 2023 kern.notice kernel: [    1.504220] sd 6:0:0:0: [sda] Write Protect is off
Thu Jan  5 15:41:19 2023 kern.debug kernel: [    1.505249] sd 6:0:0:0: [sda] Mode Sense: 00 3a 00 00
Thu Jan  5 15:41:19 2023 kern.notice kernel: [    1.505272] sd 6:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
Thu Jan  5 15:41:19 2023 kern.warn kernel: [    1.511797] Alternate GPT is invalid, using primary GPT.
Thu Jan  5 15:41:19 2023 kern.info kernel: [    1.512960]  sda: sda1 sda2 sda128
Thu Jan  5 15:41:19 2023 kern.notice kernel: [    1.514943] sd 6:0:0:0: [sda] Attached SCSI disk
Thu Jan  5 15:41:19 2023 kern.info kernel: [    1.521296] VFS: Mounted root (squashfs filesystem) readonly on device 8:2.
Thu Jan  5 15:41:19 2023 kern.info kernel: [    1.523289] Freeing unused kernel image (initmem) memory: 1268K
Thu Jan  5 15:41:19 2023 kern.info kernel: [    1.528502] Write protecting the kernel read-only data: 18432k
Thu Jan  5 15:41:19 2023 kern.info kernel: [    1.531391] Freeing unused kernel image (text/rodata gap) memory: 2044K
Thu Jan  5 15:41:19 2023 kern.info kernel: [    1.533037] Freeing unused kernel image (rodata/data gap) memory: 1268K
Thu Jan  5 15:41:19 2023 kern.info kernel: [    1.534373] Run /sbin/init as init process
Thu Jan  5 15:41:19 2023 kern.debug kernel: [    1.535279]   with arguments:
Thu Jan  5 15:41:19 2023 kern.debug kernel: [    1.535281]     /sbin/init
Thu Jan  5 15:41:19 2023 kern.debug kernel: [    1.535282]   with environment:
Thu Jan  5 15:41:19 2023 kern.debug kernel: [    1.535283]     HOME=/
Thu Jan  5 15:41:19 2023 kern.debug kernel: [    1.535284]     TERM=linux
Thu Jan  5 15:41:19 2023 kern.debug kernel: [    1.535285]     BOOT_IMAGE=/boot/vmlinuz
[... cut ...]
Thu Jan  5 15:41:19 2023 kern.info kernel: [    1.742222] hid-generic 0003:0627:0001.0001: input,hidraw0: USB HID v0.01 Mouse [QEMU QEMU USB Tablet] on usb-0000:00:1d.7-1/input0
Thu Jan  5 15:41:19 2023 user.info kernel: [    1.750185] init: Console is alive
Thu Jan  5 15:41:19 2023 user.info kernel: [    2.550838] kmodloader: loading kernel modules from /etc/modules-boot.d/*
Thu Jan  5 15:41:19 2023 kern.info kernel: [    2.623436] pps_core: LinuxPPS API ver. 1 registered
Thu Jan  5 15:41:19 2023 kern.info kernel: [    2.624575] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>

sleel, 2023-01-05:
Thu Jan  5 15:41:19 2023 kern.info kernel: [    2.627687] PTP clock support registered
Thu Jan  5 15:41:19 2023 kern.info kernel: [    2.631322] Button Hotplug driver version 0.4.1
Thu Jan  5 15:41:19 2023 kern.info kernel: [    2.635334] igb: Intel(R) Gigabit Ethernet Network Driver
Thu Jan  5 15:41:19 2023 kern.info kernel: [    2.636565] igb: Copyright (c) 2007-2014 Intel Corporation.
Thu Jan  5 15:41:19 2023 kern.info kernel: [    2.681052] pps pps0: new PPS source ptp0
Thu Jan  5 15:41:19 2023 kern.info kernel: [    2.682057] igb 0000:06:10.0: added PHC on eth1
Thu Jan  5 15:41:19 2023 kern.info kernel: [    2.683071] igb 0000:06:10.0: Intel(R) Gigabit Ethernet Network Connection
Thu Jan  5 15:41:19 2023 kern.info kernel: [    2.684402] igb 0000:06:10.0: eth1: (PCIe:unknown:unknown) 00:e0:67:15:0c:a0
Thu Jan  5 15:41:19 2023 kern.info kernel: [    2.685755] igb 0000:06:10.0: eth1: PBA No: FFFFFF-0FF
Thu Jan  5 15:41:19 2023 kern.info kernel: [    2.686816] igb 0000:06:10.0: Using MSI-X interrupts. 1 rx queue(s), 1 tx queue(s)
Thu Jan  5 15:41:19 2023 kern.info kernel: [    2.688631] ACPI: \_SB_.GSIG: Enabled at IRQ 22
Thu Jan  5 15:41:19 2023 kern.info kernel: [    2.732850] pps pps1: new PPS source ptp1
Thu Jan  5 15:41:19 2023 kern.info kernel: [    2.733861] igb 0000:06:11.0: added PHC on eth2
Thu Jan  5 15:41:19 2023 kern.info kernel: [    2.734865] igb 0000:06:11.0: Intel(R) Gigabit Ethernet Network Connection
Thu Jan  5 15:41:19 2023 kern.info kernel: [    2.736159] igb 0000:06:11.0: eth2: (PCIe:unknown:unknown) 00:e0:67:15:0c:a1
Thu Jan  5 15:41:19 2023 kern.info kernel: [    2.737530] igb 0000:06:11.0: eth2: PBA No: FFFFFF-0FF
Thu Jan  5 15:41:19 2023 kern.info kernel: [    2.738601] igb 0000:06:11.0: Using MSI-X interrupts. 1 rx queue(s), 1 tx queue(s)
Thu Jan  5 15:41:19 2023 kern.info kernel: [    2.742122] sdhci-pltfm: SDHCI platform and OF driver helper
Thu Jan  5 15:41:19 2023 kern.info kernel: [    2.744722] usbcore: registered new interface driver uas
Thu Jan  5 15:41:19 2023 user.info kernel: [    2.746904] kmodloader: done loading kernel modules from /etc/modules-boot.d/*
Thu Jan  5 15:41:19 2023 user.info kernel: [    2.751571] init: - preinit -
Thu Jan  5 15:41:19 2023 kern.notice kernel: [    3.203701] random: jshn: uninitialized urandom read (4 bytes read)
Thu Jan  5 15:41:19 2023 kern.notice kernel: [    3.276648] random: jshn: uninitialized urandom read (4 bytes read)
Thu Jan  5 15:41:19 2023 kern.notice kernel: [    3.281100] random: jshn: uninitialized urandom read (4 bytes read)
Thu Jan  5 15:41:19 2023 kern.info kernel: [    3.492713] 8021q: adding VLAN 0 to HW filter on device eth0
Thu Jan  5 15:41:19 2023 kern.notice kernel: [    7.522787] random: jshn: uninitialized urandom read (4 bytes read)
Thu Jan  5 15:41:19 2023 user.info kernel: [    7.648159] mount_root: loading kmods from internal overlay
Thu Jan  5 15:41:19 2023 user.info kernel: [    7.657148] kmodloader: loading kernel modules from //etc/modules-boot.d/*
Thu Jan  5 15:41:19 2023 user.info kernel: [    7.661815] kmodloader: done loading kernel modules from //etc/modules-boot.d/*
Thu Jan  5 15:41:19 2023 kern.info kernel: [    7.776698] loop0: detected capacity change from 0 to 819200
Thu Jan  5 15:41:19 2023 kern.notice kernel: [    7.845054] random: procd: uninitialized urandom read (4 bytes read)
Thu Jan  5 15:41:19 2023 kern.info kernel: [    7.848526] loop0: detected capacity change from 819200 to 741888
Thu Jan  5 15:41:19 2023 kern.notice kernel: [    7.876358] F2FS-fs (loop0): Disable nat_bits due to incorrect cp_ver (10512000426748372003, 5758047348967564297)
Thu Jan  5 15:41:19 2023 kern.notice kernel: [    7.881229] F2FS-fs (loop0): Mounted with checkpoint version = 73176423
Thu Jan  5 15:41:19 2023 user.info kernel: [    7.883515] block: attempting to load /tmp/f2fs_cfg/upper/etc/config/fstab
Thu Jan  5 15:41:19 2023 user.info kernel: [    7.886167] block: extroot: device not present, retrying in 5 seconds
Thu Jan  5 15:41:19 2023 kern.warn kernel: [   12.904367] overlayfs: null uuid detected in lower fs '/', falling back to xino=off,index=off,nfs_export=off.
Thu Jan  5 15:41:19 2023 user.notice kernel: [   12.907490] mount_root: switched to extroot
Thu Jan  5 15:41:19 2023 user.warn kernel: [   13.096222] urandom-seed: Seeding with /etc/urandom.seed
Thu Jan  5 15:41:19 2023 user.info kernel: [   13.150210] procd: - early -
Thu Jan  5 15:41:19 2023 kern.notice kernel: [   13.363276] random: jshn: uninitialized urandom read (4 bytes read)
Thu Jan  5 15:41:19 2023 user.info kernel: [   13.770196] procd: - ubus -
Thu Jan  5 15:41:19 2023 kern.notice kernel: [   14.001605] random: ubusd: uninitialized urandom read (4 bytes read)
Thu Jan  5 15:41:19 2023 kern.notice kernel: [   14.005546] random: ubusd: uninitialized urandom read (4 bytes read)
Thu Jan  5 15:41:19 2023 user.info kernel: [   14.076017] procd: - init -
Thu Jan  5 15:41:19 2023 kern.notice kernel: [   14.186425] random: crng init done
Thu Jan  5 15:41:19 2023 kern.notice kernel: [   14.187320] random: 21 urandom warning(s) missed due to ratelimiting
Thu Jan  5 15:41:19 2023 user.info kernel: [   14.437084] urngd: v1.0.2 started.
Thu Jan  5 15:41:19 2023 user.info kernel: [   14.444940] kmodloader: loading kernel modules from /etc/modules.d/*
Thu Jan  5 15:41:19 2023 kern.info kernel: [   14.584228] NET: Registered PF_ALG protocol family
Thu Jan  5 15:41:19 2023 kern.info kernel: [   14.598198] ntfs3: Max link count 4000
[... cut ...]
Thu Jan  5 15:41:24 2023 daemon.notice netifd: Interface 'wan_6' has link connectivity
Thu Jan  5 15:41:24 2023 daemon.notice netifd: Interface 'wan_6' is setting up now
Thu Jan  5 15:41:24 2023 daemon.notice netifd: Interface 'wan_6' is now up
Thu Jan  5 15:41:24 2023 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Thu Jan  5 15:41:24 2023 user.notice firewall: Reloading firewall due to ifupdate of wan (pppoe-wan)
Thu Jan  5 15:41:25 2023 user.err : jail: can't read from child
Thu Jan  5 15:41:25 2023 user.err : jail: can't write to parent

Darks41, 2023-01-06:
我的R5S也是同样的问题，微信和网页都不正常，但是B站手机客户端还行

chiyu521, 2023-01-11:
看看 路由里 启动项 能不能点进去~

sleel, 2023-01-12:
> 看看 路由里 启动项 能不能点进去~

现在用的5.10    5.15.86测试的时候好像没有这个问题 因为我调整了启动项

chiyu521, 2023-01-13:
我也是用12月8日的一切正常  之后的版本 就出问题  特点就是点不进去启动项

Darks41, 2023-01-13:
现在两个问题
1.更改防火墙的相关设定可能会导致卡死在“正在应用设置”的转圈页面，但实际上已经更改成功了
2.路由器重启或者刷新固件后有些网页和app打不开或者打开很慢，要重启ipv6服务（亦或者是重新连接ipv6？我不太确定）之后才正常，我的办法是，在wan6口-高级设置-使用自定义ipv6前缀那里随便填点东西，保存应用，再把填的东西清空，保存应用

chiyu521, 2023-01-13:
我删掉IPV6 然后 就啥也开不了了

caiyd90, 2023-01-14:
关闭广告屏蔽大师 Plus+ 就正常了

sleel, 2023-01-14:
> 关闭广告屏蔽大师 Plus+ 就正常了

就没有集成这玩意

Darks41, 2023-01-14:
最新进展：更新代码后编译的最新一版固件貌似解决了上网的问题，把广告屏蔽大师关了以后转圈的问题也解决了，不过没有详细测试，不敢说100%OK

chiyu521, 2023-01-16:
等详细结果 现在过年期间家里人多不方便折腾

sleel, 2023-01-18:
在PVE 跑了一下  最新源码 6.1.6和5.15.88内核   懂球帝  微信 问题依旧 。没有集成任何去广告和passwall 仅拨号 
最新5.10.163 依然正常！

Darks41, 2023-01-21:
> 在PVE 跑了一下 最新源码 6.1.6和5.15.88内核 懂球帝 微信 问题依旧 。没有集成任何去广告和passwall 仅拨号 最新5.10.163 依然正常！

请教一下怎么编译不同版本内核的固件？

sleel, 2023-01-26:
> > 在PVE 跑了一下 最新源码 6.1.6和5.15.88内核 懂球帝 微信 问题依旧 。没有集成任何去广告和passwall 仅拨号 最新5.10.163 依然正常！
> 
> 请教一下怎么编译不同版本内核的固件？

修改/lede/target/linux/x86/Makefile   KERNEL_PATCHVER:=6.1    把默认6.1改成你要编译的版本

Darks41, 2023-01-27:
春节回来编译了5.15.9内核固件，故障又出现了，重连了几次IPV6才正常，现在试下编译5.10

Darks41, 2023-01-28:
今天打开testipv6.cn时提示IPv6 大数据包传输测试失败，会不会是这个问题？不过我打开其他的镜像站又没出现这个问题
“糟糕！IPv6 基本正常，但较大的数据包传输失败，这可能导致部分网站显示不全。询问你的运营商是不是 MTU 的问题，这可能与你的隧道有关。 检查你的防火墙，确保已放行 ICMPv6 消息（尤其是第 2 类，“数据包过大”）。”

wangzhuok, 2023-01-29:
我也遇到差不多的问题，也比较奇葩，固定的某个客户端不能联网，比如某个平板不能上网，打开网页可以，开APP不行，连手机热点一点问题没有，不知道什么原因。
固件没有集成adp和不可描述的插件。

sleel, 2023-03-14:
5.15.98 问题依旧

coolsnowwolf, 2026-09-05:
这是 2023 年 x86/PVE 环境下的 5.15/6.1 旧内核网络表现回归，评论中给出的 TCPMSS 按 PMTU 规则可缓解大包传输，issue 没有提供 MTU、IPv6 RA/PMTU 或当前源码的最小复现。当前无法归因于 LEDE 代码并形成通用修复。

closed 2026-09-05 by coolsnowwolf