coolsnowwolf/lede #10736: Package bind-libs is missing dependencies for the following libraries: libmaxminddb.so.0.0.7

GreatMichaelLee, 2023-01-06:
make[3]: Entering directory '/home/builder/lede_x86/feeds/packages/net/bind'
mkdir -p /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/builder/lede_x86/tmp/.root-copy.flock -c 'cp -fpR /home/builder/lede_x86/build_dir/target-x86_64_musl/bind-9.18.10/.pkgdir/bind-libs/. /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/'
touch /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp/.bind-libs_installed
mkdir -p /home/builder/lede_x86/bin/targets/x86/64/packages /home/builder/lede_x86/build_dir/target-x86_64_musl/bind-9.18.10/ipkg-x86_64/bind-libs/CONTROL /home/builder/lede_x86/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /home/builder/lede_x86/build_dir/target-x86_64_musl/bind-9.18.10/ipkg-x86_64/bind-libs/usr/lib
cp -fpR /home/builder/lede_x86/build_dir/target-x86_64_musl/bind-9.18.10/ipkg-install/usr/lib/*.so* /home/builder/lede_x86/build_dir/target-x86_64_musl/bind-9.18.10/ipkg-x86_64/bind-libs/usr/lib
find /home/builder/lede_x86/build_dir/target-x86_64_musl/bind-9.18.10/ipkg-x86_64/bind-libs -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package bind-libs is missing dependencies for the following libraries:
libmaxminddb.so.0.0.7
make[3]: *** [Makefile:290: /home/builder/lede_x86/bin/packages/x86_64/packages/bind-libs_9.18.10-2_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/net/bind'
time: package/feeds/packages/bind/compile#0.22#0.23#0.39
    ERROR: package/feeds/packages/bind failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/bind/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231：world] 错误 2
builder@Build-Server:/home/builder/lede_x86$
builder@Build-Server:/home/builder/lede_x86$

GreatMichaelLee, 2023-01-06:
有人能修下吗？刚make dirclean再编译后还是不行

KevinMX, 2023-01-07:
可能有关：https://github.com/coolsnowwolf/packages/pull/629

GreatMichaelLee, 2023-01-07:
> 可能有关：[coolsnowwolf/packages#629](https://github.com/coolsnowwolf/packages/pull/629)

不知道是不是这个改动引起的？麻烦大佬 @Beginner-Go 看看哈....

https://github.com/coolsnowwolf/packages/pull/629/commits/861a6f21da67238987f58633b1a0848fb6c48c7e

lxzmads, 2023-01-07:
加个依赖编译可以通过
```
diff --git a/net/bind/Makefile b/net/bind/Makefile
index 699f935..9962cf8 100644
--- a/net/bind/Makefile
+++ b/net/bind/Makefile
@@ -60,6 +60,7 @@ define Package/bind-libs
        +libpthread \
        +libatomic \
        +libuv \
+       +libmaxminddb \
        +BIND_ENABLE_DOH:libnghttp2 \
        +BIND_LIBXML2:libxml2 \
        +BIND_LIBJSON:libjson-c
```

GreatMichaelLee, 2023-01-07:
> 加个依赖编译可以通过
>
> ```
> diff --git a/net/bind/Makefile b/net/bind/Makefile
> index 699f935..9962cf8 100644
> --- a/net/bind/Makefile
> +++ b/net/bind/Makefile
> @@ -60,6 +60,7 @@ define Package/bind-libs [...]

感谢，这个work.

Beginner-Go, 2023-01-07:
fixed.

closed 2023-01-08 by aiamadeus