coolsnowwolf/lede #10739: LXC容器里面使用aliddns总是报错，请教下如何解决这些奇怪的错误。

rsd1987, 2023-01-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
在使用aliddns的时候，lxc容器总是不对，报了一堆错误，请问下是需要开什么特殊权限吗？
下面是ALIDDNS的更新日志信息
```
2023-01-06 22:08:24 WAN-IP: 113.206.5.74
2023-01-06 22:08:24 DOMAIN-IP: 119.8.15.190
2023-01-06 22:08:24 UPDATING...
2023-01-06 22:08:24 WAN-IP: 113.206.2.1
2023-01-06 22:08:24 DOMAIN-IP: 119.86.18.1
2023-01-06 22:08:24 UPDATING...
139633685359432:error:25066067:DSO support routines:dlfcn_load:could not load the shared library:crypto/dso/dso_dlfcn.c:118:filename(/usr/lib/engines-1.1/devcrypto.so): Error loading shared library /usr/lib/engines-1.1/devcrypto.so: No such file or directory
139633685359432:error:25070067:DSO support routines:DSO_load:could not load the shared library:crypto/dso/dso_lib.c:162:
139633685359432:error:260B6084:engine routines:dynamic_load:dso not found:crypto/engine/eng_dyn.c:434:
139633685359432:error:2606A074:engine routines:ENGINE_by_id:no such engine:crypto/engine/eng_list.c:421:id=devcrypto
139633685359432:error:260BC066:engine routines:int_engine_configure:engine configuration error:crypto/engine/eng_cnf.c:140:section=devcrypto, name=default_algorithms, value=ALL
139633685359432:error:0E07606D:configuration file routines:module_run:module initialization error:crypto/conf/conf_mod.c:177:module=engines, value=engines, retcode=-1      
139883617139528:error:25066067:DSO support routines:dlfcn_load:could not load the shared library:crypto/dso/dso_dlfcn.c:118:filename(/usr/lib/engines-1.1/devcrypto.so): Error loading shared library /usr/lib/engines-1.1/devcrypto.so: No such file or directory
139883617139528:error:25070067:DSO support routines:DSO_load:could not load the shared library:crypto/dso/dso_lib.c:162:
139883617139528:error:260B6084:engine routines:dynamic_load:dso not found:crypto/engine/eng_dyn.c:434:
139883617139528:error:2606A074:engine routines:ENGINE_by_id:no such engine:crypto/engine/eng_list.c:421:id=devcrypto
139883617139528:error:260BC066:engine routines:int_engine_configure:engine configuration error:crypto/engine/eng_cnf.c:140:section=devcrypto, name=default_algorithms, value=ALL
139883617139528:error:0E07606D:configuration file routines:module_run:module initialization error:crypto/conf/conf_mod.c:177:module=engines, value=engines, retcode=-1      
2023-01-06 22:08:24 ADD record 
2023-01-06 22:08:24 # ERROR, Please Check Config/Time
```

其中 在LXC的配置文件已经做了这些更改
```
arch: amd64
cores: 2
features: fuse=1,mknod=1,mount=nfs;cifs,nesting=1
hostname: OP
memory: 2048
net0: name=eth0,bridge=vmbr0,hwaddr=0A:32:D3:D4:9F:C0,type=veth
net1: name=eth1,bridge=vmbr0,hwaddr=92:D3:A1:E3:3B:29,tag=4,type=veth
onboot: 1
ostype: unmanaged
rootfs: local-lvm:vm-103-disk-0,size=4G
swap: 0
lxc.cgroup.devices.allow: a
lxc.cap.drop: 
lxc.cgroup2.devices.allow: c 108:0 rwm
lxc.mount.entry: /dev/ppp dev/ppp none bind,create=file
lxc.cgroup2.devices.allow: c 10:200 rwm
lxc.mount.entry: /dev/net/tun dev/net/tun none bind,create=file
```
如何能解决aliddns更新的问题，让他们不报错正常使用呢？

### (2) 路由器型号和固件版本
X86平台 运行在LXC容器下

### (3) 详细日志
见ALIDDNS更新日志

zoviyan, 2023-11-28:
解决了嘛，一样的问题，好像是少了加密库啥的

coolsnowwolf, 2026-09-05:
日志中的错误是 OpenSSL 尝试加载 `/usr/lib/engines-1.1/devcrypto.so` 失败，随后 aliddns 更新请求返回 `ADD record` 配置/时间错误。当前 LEDE 仓库并没有名为 aliddns 的包，只有 `ddns-scripts_aliyun`（直接调用阿里云 API）；该问题来自用户自定义 aliddns 程序与容器内 OpenSSL engine 配置，不是 LEDE 可修复代码。请移除/修正 `openssl.cnf` 中的 devcrypto engine，或安装匹配版本的 `libopenssl-devcrypto`，并检查容器时间与 DNS。缺少程序来源及完整 API 响应，关闭该历史问题。

closed 2026-09-05 by coolsnowwolf