coolsnowwolf/lede #10740: ATH79改了内核编译出错，

yuzhe1129, 2023-01-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

KERNEL_PATCHVER:=   改了 5.15  5.4 都试过了，都是出错，5.10编译可以成功， 请问如果想要5.15  5.4  怎么样才能成功

### (2) 路由器型号和固件版本

841V9  941V6编译固件

### (3) 详细日志
net/netfilter/xt_state.o
  CC [M]  net/netfilter/xt_time.o
  LD [M]  net/netfilter/nf_conntrack.o
  LD [M]  net/netfilter/nf_conntrack_h323.o
  LD [M]  net/netfilter/nf_nat.o
  LD [M]  net/netfilter/nf_flow_table.o
  AR      net/netfilter/built-in.a
  AR      net/built-in.a
make[5]: Leaving directory '/home/c/lede/build_dir/target-mips_24kc_musl/linux-ath79_tiny/linux-5.4.228'
make[4]: *** [Makefile:21: /home/c/lede/build_dir/target-mips_24kc_musl/linux-ath79_tiny/linux-5.4.228/.modules] Error 2
make[4]: Leaving directory '/home/c/lede/target/linux/ath79'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/c/lede/target/linux'
time: target/linux/compile#372.31#70.14#150.76
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/c/lede'
make[1]: *** [target/Makefile:23: /home/c/lede/staging_dir/target-mips_24kc_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/c/lede'
make: *** [/home/c/lede/include/toplevel.mk:231：world] 错误 2

et/bridge/br_mdb.o
  CC      net/bridge/br_multicast_eht.o
  CC      net/ipv6/ip6_offload.o
  CC      net/bridge/br_vlan.o
  CC      net/ipv6/tcpv6_offload.o
  CC      net/bridge/br_vlan_tunnel.o
  CC      net/bridge/br_vlan_options.o
  CC      net/ipv6/exthdrs_offload.o
  CC      net/bridge/br_switchdev.o
  CC      net/ipv6/inet6_hashtables.o
  CC      net/ipv6/mcast_snoop.o
  AR      net/bridge/built-in.a
  AR      net/ipv6/built-in.a
  AR      net/built-in.a
make[5]: Leaving directory '/home/c/lede/build_dir/target-mips_24kc_musl/linux-ath79_tiny/linux-5.15.86'
make[4]: *** [Makefile:21: /home/c/lede/build_dir/target-mips_24kc_musl/linux-ath79_tiny/linux-5.15.86/.modules] Error 2
make[4]: Leaving directory '/home/c/lede/target/linux/ath79'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/c/lede/target/linux'
time: target/linux/compile#456.48#93.88#217.14
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/c/lede'
make[1]: *** [target/Makefile:23: /home/c/lede/staging_dir/target-mips_24kc_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/c/lede'
make: *** [/home/c/lede/include/toplevel.mk:231：world] 错误 2

yuzhe1129, 2023-01-09:
又编译了一次发现如果勾上网易云的话就是出错，不勾的话就成功，请问这是网易云与什么冲突了？

coolsnowwolf, 2026-09-05:
日志没有实际失败行，所示 ath9k/mac80211 包均已成功打包，后续错误上下文缺失；无法确认 ATH79 841V9 的当前代码问题，关闭为 not planned。

closed 2026-09-05 by coolsnowwolf