coolsnowwolf/lede #10741: x86_64架构固件编译出的 openssh-server 无法正常连接

BROBIRD, 2023-01-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：x86_64架构固件编译出的 openssh-server 无法正常连接

### (2) 路由器型号和固件版本

A：x86_64 固件版本 73de9902ed3f7522bff55f032010040ee2d54829

### (3) 详细日志

A：
ssh 客户端提示：
```
Socket error Event: 32 Error: 10053.
Connection closing...Socket close.
Connection closed by foreign host.
```

openwrt 系统日志提示：
```
auth.err sshd[16674]: error: Unable to load host key: /etc/ssh/ssh_host_ecdsa_key
auth.info sshd[16674]: Accepted password for root from x.x.x.x port 52745 ssh2
auth.crit sshd[16674]: fatal: ssh_sandbox_violation: unexpected system call (arch:0xc000003e,syscall:20 @ 0x7fad35bd4c60) [preauth]
auth.crit sshd[16674]: fatal: privsep_preauth: preauth child exited with status 1
```
尝试更换 openssl 源码版本为 1.1.1q 并重新生成 `/etc/ssh/ssh_host_ecdsa_key` 无效

WYC-2020, 2023-01-30:
这个问题我以前说过，如果 后面没有升级openssl的话 直接在配置里面加一个什么limit就可以，如果升级的话 就需要打patch去掉CRIOGET 就可以了

BROBIRD, 2023-02-01:
> 这个问题我以前说过，如果 后面没有升级openssl的话 直接在配置里面加一个什么limit就可以，如果升级的话 就需要打patch去掉CRIOGET 就可以了

方便说得再详细一点吗？是否有相关的链接？谢谢！