coolsnowwolf/lede #10744: recipe for target 'package/kernel/ath10k-ct/compile' failed

zbluekuma, 2023-01-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
按照默认设置，重新git clone编译，有梯子全局，编译固件失败
### (1) 具体问题

A：recipe for target 'package/kernel/ath10k-ct/compile' failed

### (2) 路由器型号和固件版本

A：Arris sbr-AC1750路由器，固件为最新版本，内核为5.15

### (3) 详细日志

Applying ./patches/960-0011-ath10k-limit-pci-buffer-size.patch using plaintext: 
patching file ath10k-5.15/pci.c
touch /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/ath10k-ct-regular/ath10k-ct-2022-05-13-f808496f/.prepared_fbca216640ea0da906089ca0c8b980fa_6664517399ebbbc92a37c5bb081b5c53
make[3]: *** No rule to make target '/home/cnz/lede/staging_dir/target-mips_24kc_musl/usr/include/mac80211-backport/backport/autoconf.h', needed by '/home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/ath10k-ct-regular/ath10k-ct-2022-05-13-f808496f/.configured_68b329da9893e34099c7d8ad5cb9c940'.  Stop.
make[3]: Leaving directory '/home/cnz/lede/package/kernel/ath10k-ct'
time: package/kernel/ath10k-ct/regular/compile#0.54#0.38#0.85
    ERROR: package/kernel/ath10k-ct failed to build (build variant: regular).
package/Makefile:114: recipe for target 'package/kernel/ath10k-ct/compile' failed
make[2]: *** [package/kernel/ath10k-ct/compile] Error 1
make[2]: Leaving directory '/home/cnz/lede'
package/Makefile:110: recipe for target '/home/cnz/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/cnz/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/cnz/lede'
/home/cnz/lede/include/toplevel.mk:229: recipe for target 'world' failed
make: *** [world] Error 2

hzjnet, 2023-01-08:
还是build variant: regular错误，估计是没人修了，我前面提过直接被关issue

hzjnet, 2023-01-14:
> 还是build variant: regular错误，估计是没人修了，我前面提过,但是大佬直接被关issue

最后还是参考其它issue用openwrt官方22.03或者master版本的mac80211直接替换就正常了，仅仅测试了ipq401x和ipq806x

Kare-Udon, 2023-01-17:
遇到了同样的问题，log 如下：
```
make[3]: Entering directory '/workdir/openwrt/package/kernel/ath10k-ct'
make[3]: *** No rule to make target '/workdir/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211-backport/backport/autoconf.h', needed by '/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-regular/ath10k-ct-2022-05-13-f808496f/.configured_68b329da9893e34099c7d8ad5cb9c940'.  Stop.
make[3]: Leaving directory '/workdir/openwrt/package/kernel/ath10k-ct'
time: package/kernel/ath10k-ct/regular/compile#0.16#0.04#0.20
    ERROR: package/kernel/ath10k-ct failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/ath10k-ct/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.
```

hzjnet, 2023-01-19:
> 最后还是参考其它issue用openwrt官方22.03或者master版本的mac80211直接替换就正常了，仅仅测试了ipq401x和ipq806x, 修改前mesh完全不能用，修改后mesh正常，仅发现mesh子网dns不正常外nss没测试其它都正常！
> 
> 今早恢复出厂模式后重新设置mesh就一切正常了

最后测试还是用openwrt官方22.03的5.15.81版本mac80211稳定可用

zbluekuma, 2023-01-22:
@hzjnet 谢谢，有时间我来试试，22年上半年编译的固件也还算好用。

Luupw3d, 2023-04-07:
> > 最后还是参考其它issue用openwrt官方22.03或者master版本的mac80211直接替换就正常了，仅仅测试了ipq401x和ipq806x, 修改前mesh完全不能用，修改后mesh正常，仅发现mesh子网dns不正常外nss没测试其它都正常！
> > 今早恢复出厂模式后重新设置mesh就一切正常了
> 
> 最后测试还是用openwrt官方22.03的5.15.81版本mac80211稳定可用

我使用官方mac80211会导致WiFi无法正常开启，请问你的会不会

closed 2023-04-23 by aiamadeus