coolsnowwolf/lede #10748: 大佬们，这是什么错误？

mitsukileung, 2023-01-08:
K3
 
系统日志：

Sun Jan  8 23:39:09 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Sun Jan  8 23:39:19 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Sun Jan  8 23:39:21 2023 daemon.err uhttpd[3254]: sh: yes: unknown operand
Sun Jan  8 23:39:21 2023 daemon.err uhttpd[3254]: sh: yes: unknown operand
Sun Jan  8 23:39:21 2023 daemon.err uhttpd[3254]: sh: yes: unknown operand
Sun Jan  8 23:39:26 2023 daemon.err uhttpd[3254]: sh: yes: unknown operand
Sun Jan  8 23:39:26 2023 daemon.err uhttpd[3254]: sh: yes: unknown operand
Sun Jan  8 23:39:26 2023 daemon.err uhttpd[3254]: sh: yes: unknown operand
Sun Jan  8 23:39:29 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Sun Jan  8 23:39:39 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Sun Jan  8 23:39:49 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Sun Jan  8 23:39:59 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Sun Jan  8 23:40:09 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns

daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns  这个不断重复

lhuanyun, 2023-01-16:
我也有这个问题，你解决了吗

mitsukileung, 2023-01-17:
> 我也有这个问题，你解决了吗

没有

yanxin152133, 2023-02-20:
https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=3127867&extra=page%3D1&ordertype=1&page=2

xyz8899, 2023-04-28:
> https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=3127867&extra=page%3D1&ordertype=1&page=2

找不到解决方法

xuexijiaoben, 2024-03-28:
兄弟们，这个最终解决办法找到了没有呀

coolsnowwolf, 2026-09-05:
该 issue 仅提供了 dawn/`sh: yes: unknown operand` 运行日志，没有编译错误、调用脚本路径或配置内容，无法定位为 snort3 或其他包故障。`yes: unknown operand` 是 shell 条件表达式参数为空/格式错误的提示，应通过完整日志和触发页面定位；当前代码中也没有可对应的固定脚本。该旧咨询缺乏可复现信息，现关闭；如仍存在请附完整 uhttpd 日志、请求 URL 和相关 LuCI 应用。

closed 2026-09-05 by coolsnowwolf