coolsnowwolf/lede #10749: MTU显示问题请教

yidianyipie, 2023-01-08:
Hi Team,

最近发现一个问题，就是在luci页面设置的MTU值跟ifconfig 出来的不一致
[image]

[image]

经过多次的实验，网页端设置的值，比ifconfig 出来的值要大8

请问这个是为何，或者有什么办法来处理？

BR

a904055262, 2023-01-09:
pppoe拨号不得占用8个字节啊

gztingting, 2023-01-13:
> pppoe拨号不得占用8个字节啊

这么说luci页面里不用修改默认的MTU值囖，openwrt网络设置那里有个MSS CLAMPING选项，打√即可，对吧。

XinSSS, 2023-01-18:
想问下楼主你的是什么主题啊?

yidianyipie, 2023-01-18:
> > pppoe拨号不得占用8个字节啊
> 
> 这么说luci页面里不用修改默认的MTU值囖，openwrt网络设置那里有个MSS CLAMPING选项，打√即可，对吧。

我记得之前好像不是这样的，先这么着吧，主要是最近发现wan口总是掉线，所以才想到会不会是mtu的问题

BR
Robin

yidianyipie, 2023-01-18:
> 想问下楼主你的是什么主题啊?

引用了第三方库的一个
https://github.com/sypopo/luci-theme-argon-mc.git

closed 2023-01-18 by yidianyipie

XinSSS, 2023-01-18:
> 引用了第三方库的一个 https://github.com/sypopo/luci-theme-argon-mc.git

好的，谢谢