coolsnowwolf/lede #10752: 断电启动后，国内ip无法访问

xaotag, 2023-01-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：断电重启后，国内ipv4网站无法访问。 ipv6网站和国外正常，dns解析正常，关闭`fw3 和 openclash`均无效
[image]
openwrt 内部正常
[image]

### (2) 路由器型号和固件版本

A：x86 j4125 多口软路由 ，OpenWrt R22.12.1 / LuCI Master (git-22.354.59335-6a1c418)

xaotag, 2023-01-10:
wireshark 抓包 ，百度得知是数据包超时，但由于水平有限不知道原因
[image]

MageDelfador, 2023-01-14:
可能是NAT的问题
看看iptables -L -t nat

xaotag, 2023-01-14:
@MageDelfador 我已经重刷固件，当时的情况是。 国内ipv4 无法访问，但是 ipv6和墙外没问题。

xaotag, 2023-01-14:
老哥如果知道iptables 具体指令，我想在虚拟机中复现一下

closed 2023-01-14 by xaotag