coolsnowwolf/lede #10754: ERROR: tools/ucl failed to build.

zhangguanzhang, 2023-01-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

```
make[2]: Leaving directory '/workdir/openwrt/tools/zip'
time: tools/zip/compile#0.05#0.01#0.06
make[2]: *** tools/ucl: No such file or directory.  Stop.
make[2]: Entering directory '/workdir/openwrt'
make[2]: Leaving directory '/workdir/openwrt'
time: tools/ucl/compile#0.00#0.00#0.00
    ERROR: tools/ucl failed to build.
make[1]: *** [tools/Makefile:209: tools/ucl/compile] Error 1
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: tools/compile] Error 2
```

zhangguanzhang, 2023-01-11:
> 失败的程序要加依赖upx/host 看这一条 #10652

要加这行吗
```
echo CONFIG_PACKAGE_upx=m >> .config
```
还是不用管

lovehackintosh, 2023-01-11:
不用管
你是不是跑 `make tools/upx/compile` ?
如果是就git pull然后`make package/lean/upx/compile`

zhangguanzhang, 2023-01-11:
我试试

closed 2023-01-11 by zhangguanzhang