coolsnowwolf/lede #10759: 请教一个旁路由联网的问题

onlyJinx, 2023-01-11:
网络设备如图所示:

[image]

老毛子当无线中继去连接客厅的主路由, 所有网络设备都是**静态IP地址**, 如果手机连接老毛子wifiB的话(网关指向openwrt), 是可以正常使用网络, 包括翻墙, 这时候网络一切正常;
但是当我手机连接主路由wifiA的时候, 手机就无法上网, **只能访问一些内网资源**, 比如文件服务器的smb共享文件夹, 还有openwrt/主路由的后台界面, 无法访问其余包括国内外所有网站, 相当于流量没法从主路由出去;
我试着按照网上的一些说法, 在openwrt防火墙里添加一条`iptables -t nat -I POSTROUTING -o eth0 -j MASQUERADE` 还有取消lan口桥接, 但是并没有解决问题, 想请问下还有哪里需要做修改的嘛?

MageDelfador, 2023-01-14:
好像解密小游戏，我猜。。手机只在wifiB下设置了网关，连接wifiA的时候没设置网关？

onlyJinx, 2023-01-14:
> 好像解密小游戏，我猜。。手机只在wifiB下设置了网关，连接wifiA的时候没设置网关？

不是的，所有设备的ip & 网关全都是不变的，区别只有手机去连接wifiA和B，不管连接哪个，都不变更设置

MageDelfador, 2023-01-14:
> > 好像解密小游戏，我猜。。手机只在wifiB下设置了网关，连接wifiA的时候没设置网关？
> 
> 不是的，所有设备的ip & 网关全都是不变的，区别只有手机去连接wifiA和B，不管连接哪个，都不变更设置

一般手机里的ip设置是跟wifi绑定的，也就是说手机上的网关需要分别在两个wifi下都设置一次

onlyJinx, 2023-01-14:
@MageDelfador 这里的老毛子(wifiB)充当的是WiFi放大器的作用，所有的流量都是交给主路由处理的。
当连接主路由wifiA时，难道是流量出现兜圈子之类的？
此时手机流量是 `主路由(wifi连接)--老毛子--旁路由openwrt--老毛子--主路由--公网`

MageDelfador, 2023-01-14:
原来主路由才是真正网关，我猜事情是这样的
手机在wifiA下，发了一个数据包（目的IP地址：外网，源IP地址：192.168.1.10，源MAC地址：手机，目的MAC地址：Openwrt），数据包到了Openwrt，Openwrt向主路由并转发出去，然后数据包到达主路由，此时数据包是这样的（目的IP地址：外网，源IP地址：192.168.1.10，源MAC地址：Openwrt，目的MAC地址：主路由），主路由如果开启了反向路由验证功能，就会发现数据包的反向路由与入口不符（数据包入口是无线桥接，反向对192.168.1.10的路由是自己的wifi），于是丢弃数据包。
反向路由验证主要是为了防止伪造IP源地址。关闭这个功能的方法看主路由固件了。openwrt关闭的方法是`echo "0">/proc/sys/net/ipv4/conf/all/rp_filter`
在wifiB下，反向路由和数据包入口都是无线桥接，所以可以通过主路由的反向路由验证。

onlyJinx, 2023-01-14:
是的，主路由为所有设备的网络出口，我想确认下，这个关闭反向路由验证的操作是在主路由上操作还是在旁路由(openwrt)上操作？主路由的话是运营商给的，动不了。。

MageDelfador, 2023-01-14:
要在主路由操作。也可以给旁路由加NAT。在防火墙设置里，把LAN的“IP动态伪装”勾上试试。

onlyJinx, 2023-01-14:
勾选了还是不行, 只能访问一些内网的资源

[image]

MageDelfador, 2023-01-14:
我猜。。因为wan被删掉了，防火墙的配置出错导致启动失败？  命令行`fw3 restart`看看输出的结果正不正常

onlyJinx, 2023-01-14:
结果如下:

```
root@OpenWrt:~# fw3 restart
 * Flushing IPv4 filter table
 * Flushing IPv4 nat table
 * Flushing IPv4 mangle table
 * Flushing IPv6 filter table
 * Flushing IPv6 nat table
 * Flushing IPv6 mangle table
 * Flushing conntrack table ...
 * Populating IPv4 filter table
   * Redirect 'jellyfin'
   * Zone 'lan'
 * Populating IPv4 nat table
   * Redirect 'jellyfin'
   * Zone 'lan'
 * Populating IPv4 mangle table
   * Zone 'lan'
 * Populating IPv6 filter table
   * Zone 'lan'
 * Populating IPv6 nat table
Warning: fw3_ipt_rule_append(): Can't find target 'prerouting_lan_rule'
Warning: fw3_ipt_rule_append(): Can't find target 'postrouting_lan_rule'
Warning: fw3_ipt_rule_append(): Can't find target 'prerouting_rule'
Warning: fw3_ipt_rule_append(): Can't find target 'postrouting_rule'
   * Zone 'lan'
 * Populating IPv6 mangle table
   * Zone 'lan'
 * Set tcp_ecn to off
 * Set tcp_syncookies to on
 * Set tcp_window_scaling to on
 * Running script '/etc/firewall.user'
 * Running script '/var/etc/adbyby.include'
   ! Skipping due to path error: No such file or directory
 * Running script '/var/etc/openclash.include'
   ! Skipping due to path error: No such file or directory
 * Running script '/var/etc/passwall2.include'
   ! Skipping due to path error: No such file or directory
 * Running script '/var/etc/passwall2_server.include'
   ! Skipping due to path error: No such file or directory
 * Running script '/var/etc/shadowsocksr.include'
   ! Skipping due to path error: No such file or directory
 * Running script '/var/etc/v2ray_server.include'
   ! Skipping due to path error: No such file or directory
 * Running script '/var/etc/vssr.include'
   ! Skipping due to path error: No such file or directory
root@OpenWrt:~#
```

MageDelfador, 2023-01-14:
试试`iptables -t nat -I POSTROUTING -j MASQUERADE`
顺便再看下DNS服务器设置的对不对

onlyJinx, 2023-01-14:
刚做测试的时候发现了一个问题, 就是其实流量是出的去的, 但是只能打开国外的网页, 比如我的vps伪装域名, 我几台服务器web页面都可以打开, 但是国内的域名全部打不开; 为了排除干扰已经将旁路由的所有翻墙插件都关闭了.
DNS设置的话, 旁路由指向主路由(192.168.1.1), 网络正常;
手机的DNS和网关都是指向旁路由(192.168.1.200)
下面这条iptables规则刚测试了加不加都是一样的结果

```
root@OpenWrt:~# iptables -t nat -nvL POSTROUTING
Chain POSTROUTING (policy ACCEPT 23 packets, 1620 bytes)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 SNAT       tcp  --  *      *       0.0.0.0/0            192.168.1.203        tcp dpt:3389 to:192.168.1.200
    0     0 SNAT       udp  --  *      *       0.0.0.0/0            192.168.1.203        udp dpt:3389 to:192.168.1.200
  189 14832 MASQUERADE  all  --  *      eth0    0.0.0.0/0            0.0.0.0/0           
   23  1620 postrouting_rule  all  --  *      *       0.0.0.0/0            0.0.0.0/0            /* !fw3: Custom postrouting rule chain */
    0     0 zone_lan_postrouting  all  --  *      eth0    0.0.0.0/0            0.0.0.0/0            /* !fw3 */
root@OpenWrt:~# 
```

网卡为eth0

```
root@OpenWrt:~# ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether b2:12:0f:6e:32:2d brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.200/24 brd 192.168.1.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::b012:fff:fe6e:322d/64 scope link 
       valid_lft forever preferred_lft forever
3: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
root@OpenWrt:~# 
```

MageDelfador, 2023-01-14:
想了半天，没什么思路
看下路由表吧  然后旁路由上tcpdump抓个包看看
现在仍然是连主路由不正常  连老毛子完全正常吗

MageDelfador, 2023-01-14:
不用域名直接用ip访问国内网站试一下

onlyJinx, 2023-01-14:
> 想了半天，没什么思路 看下路由表吧 然后旁路由上tcpdump抓个包看看 现在仍然是连主路由不正常 连老毛子完全正常吗

是的, 现在的情况就是
--  手机连接连接老毛子wifiB(192.168.1.199)时, 可以正常上网以及使用旁路由的代理功能
--  手机连接主路由(192.168.1.1)时, 只能打开我的v2ray的伪装网站界面(确认不是浏览器缓存问题,服务端可以看到访问日志)
      但是手机打开v2rayN代理全局流量时网络就都正常了, 有那种只有访问我vps服务器的流量才能正常通行...
--  旁路由上已经关闭了ssrp
--  国内域名或者IP都无法打开
--  我稍后再全新安装一次openwrt, 在保持所有设置都是初始值的时候再测试下

**以上手机的ip & 网关 & dns设置都是指向旁路由192.168.1.200**

onlyJinx, 2023-01-14:
当前路由表 & 防火墙规则

[image]
[image]

onlyJinx, 2023-01-15:
我把电脑连接到主路由里做了路由跟踪,人都傻了.

首先重新装了openwrt(不保留任何配置), ip 192.168.1.111, 网关和主路由都指向主路由192.168.1.1
Transparent 是我的另一台debian

电脑(192.168.1.11)上网关&DNS都是指向新装的openwrt,对我的vps ip地址进行路由跟踪的时候, 发现他的流量会跑到我的debian上去, 然后就正常的跑出去了, 跟踪其余的ip都是全部超时, 到网关这一步都没有, 搞不明白为什么vps的跟踪路径会是经过debian的, 按照我的理解应该是
`电脑 --> openwrt(192.168.1.111) --> 主路由 --> 外网`
中间都不需要经过debian的

```
C:\Users\Na>tracert xxxxx.xyz

通过最多 30 个跃点跟踪
到 xxxxx.xyz [x.x.x.105] 的路由:

  1    59 ms    34 ms     4 ms  Transparent [192.168.1.250]
  2    31 ms     4 ms     4 ms  TianYi.Home [192.168.1.1]
  3   168 ms    17 ms     7 ms  119.131.128.1
  4    12 ms     7 ms     7 ms  x.x.36.169
  5    30 ms     8 ms     8 ms  x.x.34.45
  6     *        8 ms     *     202.97.94.142
  7    86 ms     8 ms     8 ms  202.97.94.114
  8    11 ms     7 ms     9 ms  59.43.244.141
  9   161 ms   161 ms   174 ms  59.43.182.77
 10   195 ms   168 ms   176 ms  218.30.48.146
 11   174 ms   173 ms     *     lax1-fatpipe-2.it7.net [x.x.70.235]
 12     *        *        *     请求超时。
 13   185 ms   163 ms   162 ms  xxxxx.xyz [x.x.x.105]

跟踪完成。

C:\Users\Na>tracert baidu.com

通过最多 30 个跃点跟踪
到 baidu.com [39.156.66.10] 的路由:

  1     *        *        *     请求超时。
  2     *        *        *     请求超时。
  3     *        *        *     请求超时。
  4     *        *        *     请求超时。
  5     *        *        *     请求超时。
  6     *        *        *     请求超时。
  7     *        *        *     请求超时。
  8     *        *        *     请求超时。
  9     *        *        *     请求超时。
 10     *        *        *     请求超时。
 11     *        *        *     请求超时。
 12     *        *        *     请求超时。
 13     *        *        *     请求超时。
 14     *        *        *     请求超时。
 15     *        *        *     请求超时。
 16     *        *        *     请求超时。
 17     *        *        *     请求超时。
 18     *        *        *     请求超时。
 19     *        *        *     请求超时。
 20     *        *        *     请求超时。
 21     *        *        *     请求超时。
 ```

**电脑的网络配置**

[image]

**openwrt的网络配置**

[image]

MageDelfador, 2023-01-15:
所以这个Transparent是接在哪里的？也是连接的老毛子吗
老毛子本身除了无线桥接有没有其他设置？
建议电脑、openwrt、老毛子都抓包看看

onlyJinx, 2023-01-15:
是的,Transparent也是放在老毛子下面的
老毛子开了ap模式之后, 是做不了设置的
[image]

老毛子流量是直接走主路由的
```
[JDC-1 /opt/home/admin]# traceroute baidu.com
traceroute to baidu.com (39.156.66.10), 30 hops max, 38 byte packets
 1  TianYi.Home (192.168.1.1)  2.300 ms  2.700 ms  2.220 ms
 2  119.131.128.1 (119.131.128.1)  4.280 ms  4.380 ms  3.700 ms
 3  14.147.5.114 (14.147.5.114)  6.280 ms  14.147.10.214 (14.147.10.214)  6.960 ms  14.147.5.114 (14.147.5.114)  14.540 ms
 4  14.147.11.145 (14.147.11.145)  5.320 ms  14.147.10.225 (14.147.10.225)  5.880 ms  14.147.4.113 (14.147.4.113)  5.120 ms
 5  202.97.44.161 (202.97.44.161)  35.920 ms  202.97.22.149 (202.97.22.149)  38.860 ms  *
 6  202.97.17.66 (202.97.17.66)  36.880 ms  202.97.17.62 (202.97.17.62)  38.620 ms  202.97.17.94 (202.97.17.94)  40.600 ms
 7  221.183.128.141 (221.183.128.141)  39.360 ms  36.800 ms  221.183.95.173 (221.183.95.173)  38.780 ms
 8  221.183.94.25 (221.183.94.25)  40.920 ms  39.060 ms  221.183.94.21 (221.183.94.21)  38.840 ms
 9  221.183.49.134 (221.183.49.134)  41.720 ms  221.183.49.138 (221.183.49.138)  38.080 ms  *
10  *  111.13.188.38 (111.13.188.38)  40.540 ms  111.13.0.174 (111.13.0.174)  40.240 ms
11  39.156.27.1 (39.156.27.1)  40.700 ms  *  41.420 ms
```

openwrt也是走主路由的
```
BusyBox v1.35.0 (2022-12-23 17:26:46 UTC) built-in shell (ash)

     _________
    /        /\      _    ___ ___  ___
   /  LE    /  \    | |  | __|   \| __|
  /    DE  /    \   | |__| _|| |) | _|
 /________/  LE  \  |____|___|___/|___|
 \        \   DE /
  \    LE  \    /  -------------------------------------------
   \  DE    \  /    OpenWrt SNAPSHOT, r5619-5acc56143
    \________\/    -------------------------------------------

root@OpenWrt:~# traceroute baidu.com
traceroute to baidu.com (39.156.66.10), 30 hops max, 46 byte packets
 1  192.168.1.1 (192.168.1.1)  3.229 ms  2.557 ms  2.522 ms
 2  119.131.128.1 (119.131.128.1)  23.449 ms  4.406 ms  2.902 ms
 3  14.147.9.130 (14.147.9.130)  5.479 ms  183.56.36.173 (183.56.36.173)  9.876 ms  12.431 ms
 4  113.98.75.165 (113.98.75.165)  32.583 ms  14.147.4.113 (14.147.4.113)  5.373 ms  6.068 ms
 5  202.97.100.9 (202.97.100.9)  44.781 ms  202.97.55.241 (202.97.55.241)  41.572 ms  202.97.109.190 (202.97.109.190)  37.416 ms
 6  202.97.17.74 (202.97.17.74)  39.399 ms  202.97.17.66 (202.97.17.66)  38.903 ms  202.97.17.94 (202.97.17.94)  43.025 ms
 7  221.183.86.45 (221.183.86.45)  39.128 ms  43.876 ms  *
 8  221.183.94.25 (221.183.94.25)  41.152 ms  39.309 ms  221.183.94.21 (221.183.94.21)  43.767 ms
 9  221.183.49.142 (221.183.49.142)  41.346 ms  221.183.49.138 (221.183.49.138)  38.516 ms  39.054 ms
10  111.13.188.38 (111.13.188.38)  42.110 ms  42.384 ms  39.156.27.1 (39.156.27.1)  41.567 ms
```

MageDelfador, 2023-01-15:
不抓包的话
我猜不出还有哪里有问题

onlyJinx, 2023-01-15:
抓包数据来了

下面我分为两种请将情景抓包:

第一种: 手机连接主路由wifiA, 然后把网关和DNS设置为旁路由192.168.1.200(此时网络故障)
第二种: 手机连接老毛子wifiB, 然后把网关和DNS设置为旁路由192.168.1.200(此时网络正常)

这是第一种:
下面为旁路由openwrt(192.168.1.200)的日志,不知道为什么, Transparent同样也会产生日志,这个时候是网络不通的
```
root@OpenWrt:~# tcpdump -s 0 -i eth0  port 443 or 80
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on eth0, link-type EN10MB (Ethernet), snapshot length 262144 bytes
20:07:26.050442 IP 192.168.1.10.48038 > 120.232.51.154.80: Flags [S], seq 72027684, win 65535, options [mss 1400,sackOK,TS val 4209981840 ecr 0,nop,wscale 10], length 0
20:07:26.347032 IP 192.168.1.10.51108 > szshort.weixin.qq.com.80: Flags [S], seq 2542776088, win 65535, options [mss 1400,sackOK,TS val 1891889376 ecr 0,nop,wscale 10], length 0
20:07:26.570372 IP 192.168.1.10.35144 > 183.60.8.218.80: Flags [S], seq 4141095502, win 65535, options [mss 1400,sackOK,TS val 219859846 ecr 0,nop,wscale 10], length 0
20:07:26.598120 IP 192.168.1.10.58704 > 183.60.8.69.80: Flags [S], seq 3394346461, win 65535, options [mss 1400,sackOK,TS val 308498730 ecr 0,nop,wscale 10], length 0
20:07:26.791387 IP 192.168.1.10.52704 > 36.103.232.140.80: Flags [S], seq 1583204484, win 65535, options [mss 1460,sackOK,TS val 3879612480 ecr 0,nop,wscale 10], length 0
20:07:26.791388 IP 192.168.1.10.52358 > connectivity.samsung.com.cn.443: Flags [S], seq 1734796193, win 65535, options [mss 1460,sackOK,TS val 1440929971 ecr 0,nop,wscale 10], length 0
20:07:26.854331 IP 192.168.1.10.56908 > 183.47.124.77.80: Flags [S], seq 3087601997, win 65535, options [mss 1400,sackOK,TS val 364758355 ecr 0,nop,wscale 10], length 0
20:07:27.081708 IP 192.168.1.10.48038 > 120.232.51.154.80: Flags [S], seq 72027684, win 65535, options [mss 1400,sackOK,TS val 4209982868 ecr 0,nop,wscale 10], length 0
20:07:27.367169 IP 192.168.1.10.51108 > szshort.weixin.qq.com.80: Flags [S], seq 2542776088, win 65535, options [mss 1400,sackOK,TS val 1891890401 ecr 0,nop,wscale 10], length 0
20:07:27.593608 IP 192.168.1.10.54934 > 110.76.8.204.443: Flags [S], seq 4250216857, win 65535, options [mss 1400,sackOK,TS val 2304553169 ecr 0,nop,wscale 10], length 0
20:07:27.811590 IP 192.168.1.10.36958 > 180.97.34.146.80: Flags [S], seq 1778919700, win 65535, options [mss 1460,sackOK,TS val 2189941262 ecr 0,nop,wscale 10], length 0
20:07:31.403958 IP 192.168.1.10.53718 > nms-m0-cn1.mdc-prd.cn.443: Flags [S], seq 3845682437, win 65535, options [mss 1460,sackOK,TS val 2497542902 ecr 0,nop,wscale 10], length 0
20:07:31.659314 IP 192.168.1.10.54934 > 110.76.8.204.443: Flags [S], seq 4250216857, win 65535, options [mss 1400,sackOK,TS val 2304557233 ecr 0,nop,wscale 10], length 0
20:07:31.883939 IP 192.168.1.10.41296 > 157.148.59.239.80: Flags [S], seq 698061538, win 65535, options [mss 1400,sackOK,TS val 3248493339 ecr 0,nop,wscale 10], length 0
20:07:33.197167 IP 192.168.1.10.48038 > 120.232.51.154.80: Flags [S], seq 72027684, win 65535, options [mss 1400,sackOK,TS val 4209988981 ecr 0,nop,wscale 10], length 0
20:07:33.451527 IP 192.168.1.10.51108 > szshort.weixin.qq.com.80: Flags [S], seq 2542776088, win 65535, options [mss 1400,sackOK,TS val 1891896481 ecr 0,nop,wscale 10], length 0
20:07:34.378919 IP 192.168.1.10.41738 > www.google.com.443: UDP, length 1250
20:07:34.418664 IP 192.168.1.10.54058 > safebrowsing.google.com.443: Flags [S], seq 1464341530, win 65535, options [mss 1460,sackOK,TS val 4097958962 ecr 0,nop,wscale 10], length 0
20:07:36.502738 IP 192.168.1.10.41738 > www.google.com.443: UDP, length 1250
20:07:36.682992 IP 192.168.1.10.51736 > www.google.com.443: Flags [S], seq 4214884307, win 65535, options [mss 1460,sackOK,TS val 204711180 ecr 0,nop,wscale 10], length 0
20:07:36.742714 IP 192.168.1.10.33120 > 121.40.206.159.443: Flags [S], seq 3499225937, win 65535, options [mss 1460,sackOK,TS val 265423285 ecr 0,nop,wscale 10], length 0
20:07:36.926270 IP 192.168.1.10.51740 > www.google.com.443: Flags [S], seq 2807327533, win 65535, options [mss 1460,sackOK,TS val 204711424 ecr 0,nop,wscale 10], length 0
20:07:36.978041 IP 192.168.1.10.51742 > www.google.com.443: Flags [S], seq 2594324269, win 65535, options [mss 1460,sackOK,TS val 204711476 ecr 0,nop,wscale 10], length 0
20:07:37.222426 IP 192.168.1.10.47322 > 101.227.132.126.443: Flags [S], seq 1704954101, win 65535, options [mss 1460,sackOK,TS val 1240055749 ecr 0,nop,wscale 10], length 0
20:07:37.283095 IP 192.168.1.10.51748 > www.google.com.443: Flags [S], seq 2120051268, win 65535, options [mss 1460,sackOK,TS val 204711781 ecr 0,nop,wscale 10], length 0
20:07:37.286000 IP 192.168.1.10.37424 > gear.sca.samsung.cn.443: Flags [S], seq 1646846100, win 65535, options [mss 1460,sackOK,TS val 673469362 ecr 0,nop,wscale 10], length 0
20:07:37.286001 IP 192.168.1.10.48758 > 203.209.230.36.443: Flags [S], seq 3566581355, win 65535, options [mss 1400,sackOK,TS val 198879192 ecr 0,nop,wscale 10], length 0
20:07:37.330881 IP 192.168.1.10.42540 > clientservices.googleapis.com.443: UDP, length 1250
20:07:37.330883 IP 192.168.1.10.49596 > clientservices.googleapis.com.443: Flags [S], seq 1620362195, win 65535, options [mss 1460,sackOK,TS val 3778719150 ecr 0,nop,wscale 10], length 0
20:07:37.445304 IP 192.168.1.10.54058 > safebrowsing.google.com.443: Flags [S], seq 1464341530, win 65535, options [mss 1460,sackOK,TS val 4097961989 ecr 0,nop,wscale 10], length 0
[... cut ...]
12:12:42.911813 IP 220.181.174.34.https > 192.168.1.10.49600: Flags [.], ack 1, win 501, options [nop,nop,TS val 2415413042 ecr 3778934191], length 0
12:12:43.423868 IP ruo.http > 192.168.1.10.41842: Flags [F.], seq 0, ack 1, win 510, options [nop,nop,TS val 536680364 ecr 2222627937], length 0
12:12:58.015799 IP 220.181.174.34.https > 192.168.1.10.49600: Flags [.], ack 1, win 501, options [nop,nop,TS val 2415428146 ecr 3778934191], length 0
^C
8 packets captured
8 packets received by filter
0 packets dropped by kernel
```

MageDelfador, 2023-01-15:
我猜。。 可能Transparent搞了事情，比如ARP欺骗什么的
tcpdump加-e看看MAC地址信息
最好电脑也抓个包
Transparent关机，openwrt把LAN的"IP动态伪装"打开，所有设备重启，或许能解决问题

onlyJinx, 2023-01-15:
Transparent已关机, 老毛子和openwrt重启了;
"IP动态伪装" 已经是勾选状态
电脑抓包的话, 我装个wireshark试试
这是连接主路由wifiA的日志(网络不通)
```
root@OpenWrt:~# tcpdump -e -s 0 -i eth0  port 443 or 80
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on eth0, link-type EN10MB (Ethernet), snapshot length 262144 bytes
21:50:57.930090 fe:b7:33:81:26:60 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 74: 192.168.1.10.45174 > beacons3.gvt2.com.80: Flags [S], seq 1034132078, win 65535, options [mss 1460,sackOK,TS val 1627675219 ecr 0,nop,wscale 10], length 0
21:50:57.930091 fe:b7:33:81:26:60 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 74: 192.168.1.10.44340 > 119.147.182.133.443: Flags [S], seq 39924836, win 65535, options [mss 1460,sackOK,TS val 4092336327 ecr 0,nop,wscale 10], length 0
21:50:57.930091 fe:b7:33:81:26:60 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 74: 192.168.1.10.36088 > 119.147.182.130.80: Flags [S], seq 487309509, win 65535, options [mss 1460,sackOK,TS val 129934590 ecr 0,nop,wscale 10], length 0
21:50:58.930683 fe:b7:33:81:26:60 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 74: 192.168.1.10.45174 > beacons3.gvt2.com.80: Flags [S], seq 1034132078, win 65535, options [mss 1460,sackOK,TS val 1627676227 ecr 0,nop,wscale 10], length 0
21:51:00.388782 fe:b7:33:81:26:60 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 74: 192.168.1.10.46482 > mail.sogservice.com.cn.443: Flags [S], seq 2130475276, win 65535, options [mss 1460,sackOK,TS val 1615486837 ecr 0,nop,wscale 10], length 0
21:51:00.908777 fe:b7:33:81:26:60 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 74: 192.168.1.10.45174 > beacons3.gvt2.com.80: Flags [S], seq 1034132078, win 65535, options [mss 1460,sackOK,TS val 1627678243 ecr 0,nop,wscale 10], length 0
21:51:01.981418 fe:b7:33:81:26:60 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 74: 192.168.1.10.36088 > 119.147.182.130.80: Flags [S], seq 487309509, win 65535, options [mss 1460,sackOK,TS val 129938590 ecr 0,nop,wscale 10], length 0
21:51:01.981419 fe:b7:33:81:26:60 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 74: 192.168.1.10.44340 > 119.147.182.133.443: Flags [S], seq 39924836, win 65535, options [mss 1460,sackOK,TS val 4092340327 ecr 0,nop,wscale 10], length 0
21:51:05.063087 fe:b7:33:81:26:60 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 74: 192.168.1.10.45174 > beacons3.gvt2.com.80: Flags [S], seq 1034132078, win 65535, options [mss 1460,sackOK,TS val 1627682339 ecr 0,nop,wscale 10], length 0
21:51:08.602054 fe:b7:33:81:26:60 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 74: 192.168.1.10.46482 > mail.sogservice.com.cn.443: Flags [S], seq 2130475276, win 65535, options [mss 1460,sackOK,TS val 1615495029 ecr 0,nop,wscale 10], length 0
21:51:12.088117 fe:b7:33:81:26:60 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 74: 192.168.1.10.35080 > 52.138.116.79.443: Flags [S], seq 2864412874, win 65535, options [mss 1460,sackOK,TS val 29281219 ecr 0,nop,wscale 10], length 0
21:51:13.084412 fe:b7:33:81:26:60 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 74: 192.168.1.10.35080 > 52.138.116.79.443: Flags [S], seq 2864412874, win 65535, options [mss 1460,sackOK,TS val 29282218 ecr 0,nop,wscale 10], length 0
21:51:15.086231 fe:b7:33:81:26:60 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 74: 192.168.1.10.35080 > 52.138.116.79.443: Flags [S], seq 2864412874, win 65535, options [mss 1460,sackOK,TS val 29284218 ecr 0,nop,wscale 10], length 0
21:51:15.801385 fe:b7:33:81:26:60 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 74: 192.168.1.10.51642 > tb.himg.baidu.com.80: Flags [S], seq 3083974147, win 65535, options [mss 1460,sackOK,TS val 3401483876 ecr 0,nop,wscale 10], length 0
21:51:15.849463 fe:b7:33:81:26:60 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 74: 192.168.1.10.51644 > tb.himg.baidu.com.80: Flags [S], seq 1410205670, win 65535, options [mss 1460,sackOK,TS val 3401483924 ecr 0,nop,wscale 10], length 0
21:51:15.875761 fe:b7:33:81:26:60 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 74: 192.168.1.10.51646 > tb.himg.baidu.com.80: Flags [S], seq 3688551480, win 65535, options [mss 1460,sackOK,TS val 3401483950 ecr 0,nop,wscale 10], length 0
21:51:15.910838 fe:b7:33:81:26:60 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 74: 192.168.1.10.51648 > tb.himg.baidu.com.80: Flags [S], seq 3698704992, win 65535, options [mss 1460,sackOK,TS val 3401483984 ecr 0,nop,wscale 10], length 0
21:51:15.934784 fe:b7:33:81:26:60 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 74: 192.168.1.10.60428 > 119.147.182.130.443: Flags [S], seq 152695495, win 65535, options [mss 1460,sackOK,TS val 129952573 ecr 0,nop,wscale 10], length 0
21:51:15.934785 fe:b7:33:81:26:60 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 74: 192.168.1.10.46894 > connectivity.samsung.com.cn.80: Flags [S], seq 2329988405, win 65535, options [mss 1460,sackOK,TS val 3555909504 ecr 0,nop,wscale 10], length 0
21:51:16.133886 fe:b7:33:81:26:60 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 74: 192.168.1.10.52850 > tiebapic.baidu.com.80: Flags [S], seq 662106343, win 65535, options [mss 1460,sackOK,TS val 352056622 ecr 0,nop,wscale 10], length 0
[... cut ...]
21:56:04.074634 b2:12:0f:6e:32:2d (oui Unknown) > 1a:b1:43:bf:56:5a (oui Unknown), ethertype IPv4 (0x0800), length 54: 119.100.50.48.80 > 192.168.1.10.41120: Flags [F.], seq 1, ack 1, win 556, length 0
21:56:04.079622 1a:b1:43:bf:56:5a (oui Unknown) > b2:12:0f:6e:32:2d (oui Unknown), ethertype IPv4 (0x0800), length 60: 192.168.1.10.41120 > 119.100.50.48.80: Flags [.], ack 2, win 71, length 0
21:56:04.079657 b2:12:0f:6e:32:2d (oui Unknown) > 28:93:7d:1d:83:46 (oui Unknown), ethertype IPv4 (0x0800), length 54: 192.168.1.200.41120 > 119.100.50.48.80: Flags [.], ack 2, win 71, length 0
^C
290 packets captured
290 packets received by filter
0 packets dropped by kernel
```

onlyJinx, 2023-01-15:
这是电脑(192.168.1.77)连接wifiA时候wireshark抓包日志(日志有点大,我中间部分剪掉了)

[image]

```
No.	Time	Source	Destination	Protocol	Length	Info										
1	0	192.168.1.77	142.251.42.234	TCP	66	12278  >  443 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 WS=256 SACK_PERM										
2	0.620539	192.168.1.77	220.181.174.162	TCP	66	12280  >  443 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 WS=256 SACK_PERM										
3	0.900995	192.168.1.77	220.181.174.98	TCP	66	12281  >  443 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 WS=256 SACK_PERM										
4	1.149419	192.168.1.77	220.181.174.98	TCP	66	12282  >  443 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 WS=256 SACK_PERM										
5	1.228388	192.168.1.77	220.181.174.98	TCP	66	12283  >  443 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 WS=256 SACK_PERM										
6	2.488632	192.168.1.77	23.33.32.40	TCP	66	12272  >  80 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 WS=256 SACK_PERM										
7	2.488632	192.168.1.77	23.33.32.40	TCP	66	12271  >  80 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 WS=256 SACK_PERM										
8	3.456024	AzureWav_54:9c:51	8a:88:2b:00:00:00	ARP	42	Who has 192.168.1.200? Tell 192.168.1.77										
9	3.467815	8a:88:2b:00:00:00	AzureWav_54:9c:51	ARP	60	192.168.1.200 is at 8a:88:2b:00:00:00										
10	4.635155	192.168.1.77	220.181.174.162	TCP	66	[TCP Retransmission] [TCP Port numbers reused] 12280  >  443 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 WS=256 SACK_PERM										
11	4.912574	192.168.1.77	220.181.174.98	TCP	66	[TCP Retransmission] [TCP Port numbers reused] 12281  >  443 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 WS=256 SACK_PERM										
12	5.160941	192.168.1.77	220.181.174.98	TCP	66	[TCP Retransmission] [TCP Port numbers reused] 12282  >  443 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 WS=256 SACK_PERM										
13	5.237488	192.168.1.77	220.181.174.98	TCP	66	[TCP Retransmission] [TCP Port numbers reused] 12283  >  443 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 WS=256 SACK_PERM										
14	5.301092	192.168.1.77	192.168.1.200	DNS	85	Standard query 0x0a35 A raw.githubusercontent.com										
15	5.30139	192.168.1.77	192.168.1.200	DNS	85	Standard query 0x4f77 HTTPS raw.githubusercontent.com										
16	5.307056	192.168.1.200	192.168.1.77	DNS	101	Standard query response 0x0a35 A raw.githubusercontent.com A 0.0.0.0										
17	5.318742	192.168.1.200	192.168.1.77	DNS	150	Standard query response 0x4f77 HTTPS raw.githubusercontent.com SOA dns1.p01.nsone.net										
18	5.624475	192.168.1.77	220.181.174.98	TCP	66	12285  >  443 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 WS=256 SACK_PERM										
19	6.181847	192.168.1.77	172.217.160.77	TCP	66	12286  >  443 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 WS=256 SACK_PERM										
619	57.714519	192.168.1.200	192.168.1.77	DNS	112	Standard query response 0x00dd HTTPS baidu.com SOA dns.baidu.com										
620	57.714519	192.168.1.200	192.168.1.77	DNS	359	Standard query response 0x2f73 A baidu.com A 39.156.66.10 A 110.242.68.66 NS ns3.baidu.com NS ns7.baidu.com NS dns.baidu.com NS ns4.baidu.com NS ns2.baidu.com A 180.76.76.92 A 110.242.68.134 A 111.45.3.226 A 14.215.178.80 A 220.181.33.31 A 112.80.248.64 A 36.152.45.193 AAAA 240e:940:603:4:0:ff:b01b:589a AAAA 240e:bf:b801:1002:0:ff:b024:26de										
621	57.715959	192.168.1.77	39.156.66.10	TCP	66	12334  >  443 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 WS=256 SACK_PERM										
622	57.717227	192.168.1.77	39.156.66.10	TCP	66	12335  >  443 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 WS=256 SACK_PERM										
623	57.729946	192.168.1.77	192.168.1.200	TCP	54	12293  >  22 [ACK] Seq=3647 Ack=51083 Win=130816 Len=0										
624	57.783961	192.168.1.200	192.168.1.77	SSHv2	566	Server: 										
625	57.824955	192.168.1.77	192.168.1.200	TCP	54	12293  >  22 [ACK] Seq=3647 Ack=51595 Win=130304 Len=0										
626	57.980514	192.168.1.77	39.156.66.10	TCP	66	12336  >  443 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 WS=256 SACK_PERM										
627	58.001204	192.168.1.200	192.168.1.77	SSHv2	342	Server: 										
628	58.056026	192.168.1.77	192.168.1.200	TCP	54	12293  >  22 [ACK] Seq=3647 Ack=51883 Win=130048 Len=0										
629	58.179995	AzureWav_54:9c:51	Broadcast	ARP	42	Who has 192.168.1.200? Tell 192.168.1.77										
630	58.227493	8a:88:2b:00:00:00	AzureWav_54:9c:51	ARP	60	192.168.1.200 is at 8a:88:2b:00:00:00										
631	58.227517	192.168.1.77	183.60.15.102	TCP	66	[TCP Retransmission] [TCP Port numbers reused] 12317  >  443 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 WS=256 SACK_PERM										
632	58.256211	192.168.1.77	10.75.162.10	TCP	66	[TCP Retransmission] [TCP Port numbers reused] 12323  >  443 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 WS=256 SACK_PERM										
633	58.256211	192.168.1.77	10.75.162.10	TCP	66	[TCP Retransmission] [TCP Port numbers reused] 12324  >  443 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 WS=256 SACK_PERM										
634	58.319081	AzureWav_54:9c:51	Broadcast	ARP	42	Who has 192.168.1.200? Tell 192.168.1.77										
635	58.32947	192.168.1.200	192.168.1.77	SSHv2	806	Server: 										
636	58.43242	8a:88:2b:00:00:00	AzureWav_54:9c:51	ARP	60	192.168.1.200 is at 8a:88:2b:00:00:00										
637	58.432451	192.168.1.77	192.168.1.200	TCP	54	12293  >  22 [ACK] Seq=3647 Ack=52635 Win=131328 Len=0										
638	58.458019	192.168.1.77	192.168.1.200	DNS	76	Standard query 0xe467 A dns.msftncsi.com										
639	58.46252	192.168.1.200	192.168.1.77	DNS	92	Standard query response 0xe467 A dns.msftncsi.com A 131.107.255.255										
640	58.472632	192.168.1.77	183.60.15.102	TCP	66	[TCP Retransmission] [TCP Port numbers reused] 12318  >  443 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 WS=256 SACK_PERM										
[... cut ...]
22:47:33.456081 b2:12:0f:6e:32:2d (oui Unknown) > 58:d5:0a:72:11:58 (oui Unknown), ethertype IPv4 (0x0800), length 54: catdot.dianping.com.443 > 192.168.1.10.33916: Flags [R], seq 3621544459, win 1000, length 0
22:47:33.721409 28:c2:dd:54:9c:51 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 66: 192.168.1.77.12345 > www.gstatic.com.80: Flags [S], seq 8570323, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
^C
276 packets captured
276 packets received by filter
0 packets dropped by kernel
```

MageDelfador, 2023-01-15:
原来答案是无线中继没开WDS
需要分别改主路由和老毛子的无线设置  改成WDS+AP模式，WDS主从也要设置好
不然数据包一过中继，目标MAC地址就没了

onlyJinx, 2023-01-16:
改WDS的话，我得把光猫改成桥接，用路由器拨号，我晚上回去再试试
~~运营商自带的那个光猫设置非常简陋，没有WDS，本着能用就好好用的心态，一直没去改桥接~~

onlyJinx, 2023-01-16:
WDS中继无法实现了, 我用TPlink的路由里面只有**WDS桥接** 然后连接后无法解决问题, 我想这个应该只是 **WISP**
想问下能通过软件层面解决嘛? 例如用iptables

然后我在老毛子抓包的时候发现一起有趣的问题

当我把其他设备关闭时,电脑连接wifiA, 此时老毛子日志里, 数据包是没有MAC目的地址, 像这样
```
23:26:14.068719 28:c2:dd:54:9c:51 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 66: 192.168.1.77.9401 > 204.79.197.239.https: Flags [S], seq 2000422779, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
23:26:14.069199 28:c2:dd:54:9c:51 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 66: 192.168.1.77.9400 > 204.79.197.239.https: Flags [S], seq 2198000620, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
23:26:14.780739 28:c2:dd:54:9c:51 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 66: 192.168.1.77.9407 > checkappexec.microsoft.com.https: Flags [S], seq 4161732518, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
23:26:15.202919 28:c2:dd:54:9c:51 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 66: 192.168.1.77.9409 > checkappexec.microsoft.com.https: Flags [S], seq 1773505195, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
23:26:15.239039 28:c2:dd:54:9c:51 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 66: 192.168.1.77.9388 > ec2-44-232-249-16.us-west-2.compute.amazonaws.com.https: Flags [S], seq 3639308617, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
23:26:15.410779 28:c2:dd:54:9c:51 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 66: 192.168.1.77.9410 > edge.microsoft.com.https: Flags [S], seq 2199282782, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
23:26:15.679279 28:c2:dd:54:9c:51 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 66: 192.168.1.77.9411 > edge.microsoft.com.https: Flags [S], seq 3724265376, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
23:26:16.202879 28:c2:dd:54:9c:51 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 66: 192.168.1.77.9409 > checkappexec.microsoft.com.https: Flags [S], seq 1773505195, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
23:26:16.424839 28:c2:dd:54:9c:51 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 66: 192.168.1.77.9410 > edge.microsoft.com.https: Flags [S], seq 2199282782, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
23:26:16.693359 28:c2:dd:54:9c:51 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 66: 192.168.1.77.9411 > edge.microsoft.com.https: Flags [S], seq 3724265376, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
23:26:17.305339 28:c2:dd:54:9c:51 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 66: 192.168.1.77.9389 > 204.79.197.239.https: Flags [S], seq 752667037, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
23:26:17.319859 28:c2:dd:54:9c:51 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 66: 192.168.1.77.9390 > deff.nelreports.net.https: Flags [S], seq 3693913803, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
23:26:17.537379 28:c2:dd:54:9c:51 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 66: 192.168.1.77.9391 > 204.79.197.239.https: Flags [S], seq 482442274, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
23:26:17.552819 28:c2:dd:54:9c:51 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 66: 192.168.1.77.9392 > deff.nelreports.net.https: Flags [S], seq 150265263, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
23:26:17.797919 28:c2:dd:54:9c:51 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 66: 192.168.1.77.9403 > edge-http.microsoft.com.www: Flags [S], seq 1426720080, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
23:26:18.052439 28:c2:dd:54:9c:51 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 66: 192.168.1.77.9404 > edge-http.microsoft.com.www: Flags [S], seq 112150629, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
23:26:18.208779 28:c2:dd:54:9c:51 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 66: 192.168.1.77.9409 > checkappexec.microsoft.com.https: Flags [S], seq 1773505195, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
23:26:18.425839 28:c2:dd:54:9c:51 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 66: 192.168.1.77.9410 > edge.microsoft.com.https: Flags [S], seq 2199282782, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
23:26:18.551879 28:c2:dd:54:9c:51 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 66: 192.168.1.77.9393 > ntp.msn.cn.https: Flags [S], seq 43340754, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
23:26:18.690839 28:c2:dd:54:9c:51 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 66: 192.168.1.77.9411 > edge.microsoft.com.https: Flags [S], seq 3724265376, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
23:26:18.786379 28:c2:dd:54:9c:51 (oui Unknown) > Broadcast, ethertype IPv4 (0x0800), length 66: 192.168.1.77.9407 > checkappexec.microsoft.com.https: Flags [S], seq 4161732518, wi
```

然后当我把我的充当smb文件服务器的debian开机了之后, 发现他把所有的数据包都往我的debian上面发送了

```

23:44:49.098460 e4:3a:6e:5a:f0:d6 (oui Unknown) > 28:93:7d:1d:83:46 (oui Unknown), ethertype IPv4 (0x0800), length 66: 192.168.1.202.40630 > 172.67.70.128.https: Flags [.], ack 270213, win 2480, options [nop,nop,TS val 3014616620 ecr 2918859192], length 0
[... cut ...]
23:44:49.278360 28:93:7d:1d:83:46 (oui Unknown) > e4:3a:6e:5a:f0:d6 (oui Unknown), ethertype IPv4 (0x0800), length 66: 172.67.70.128.https > 192.168.1.202.40630: Flags [.], ack 155450, win 142, options [nop,nop,TS val 2918859365 ecr 3014616620], length 0
23:44:49.435940 28:93:7d:1d:83:46 (oui Unknown) > e4:3a:6e:5a:f0:d6 (oui Unknown), ethertype IPv4 (0x0800), length 440: 172.67.70.128.https > 192.168.1.202.40630: Flags [P.], seq 271454:271828, ack 155450, win 142, options [nop,nop,TS val 2918859523 ecr 3014616639], length 374
```

为什么会出现这种情况? 之前他把数据都忘我的 Transparent 上丢是不是也是随机找一台主机当数据包的出口丢了? 
之前连接wifiA的时候只能打开我的vps ip地址, 可能是我的Transparent 有一些iptables规则的原因, Transparent是我用来当透明网关的(debian11).

MageDelfador, 2023-01-16:
这种情况恐怕最好桥接
要不连根网线？
或者再加个支持WDS的路由器？反正主路由的wifi也渣

iptables工作在网络层，无能为力

其他方法都会有些问题
试试把openwrt放在老毛子之前，让所有房间的流量都通过openwrt，然后自行判断哪些流量需要代理？这种方法又会引入很多关于主路由器设置的问题。。

老毛子向哪里发数据包的问题，这个因为前面来的数据包的目的MAC地址没有了，老毛子就自行判断该丢给谁，具体怎么判断就不知道了。
同网段但不桥接本身就是不正常的，应该尽可能避免。
也就是说要么弄好桥接，要么老毛子走NAT，后边换个网段比如192.168.2.x。然后再考虑openwrt的问题

onlyJinx, 2023-01-16:
网线的话接不了，租的房间，房东当时没布线，不然我不用无线桥接。
买个支持WDS的路由是个好建议，到时候活动蹲一个。
老毛子下的设备单独一个网断的话，会比较麻烦，因为下面有好几个机器，需要弄好几条iptables还要我去记转发的端口。

目前的连接方法影响到的也就是不能连接主路由的时候把网关设置为openwrt来翻墙，其余的内网资源访问是全部正常的，只是这个问题困扰了我很久所以就发issue请教一下。

最后想问下，为啥连接主路由wifiA然后设置网关为openwrt时，虽然不能访问外网，但是内网资源为什么可以访问？理论上不是应该也无法访问才对，数据包在从主路由发往老毛子的时候目的MAC地址都会丢失的。

MageDelfador, 2023-01-16:
老毛子把MAC地址改成广播，内网服务器收到后检查里面的目的ip地址，发现是自己的就接收了

MageDelfador, 2023-01-16:
也可能是老毛子根据目的ip地址，猜出MAC地址，具体抓包看一下就知道了

onlyJinx, 2023-01-16:
原来是这样。受教了，非常感谢耐心解答。

closed 2023-01-16 by onlyJinx