coolsnowwolf/lede #10762: libselinux编译失败

verages, 2023-01-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
Build error of libselinux for x86_64 target，能否提供能编译出的ubuntu具体版本
Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
搜索了存在但是都无解决方法
## 2. 详细叙述
在ubuntu18.04.6和ubuntu20.04上编译X86_64均失败，但编译arm64(r2s)能够编译出来
### (1) 具体问题

A：在编译X86b版本中会在libselinux-3.3失败，提示缺少libpcre.so.1，手动添加后，提示缺少libfts.so.0.该现象在r2s上没有出现

### (2) 路由器型号和固件版本

A：X86,当前最新仓库代码

### (3) 详细日志

A：
```
make[4]: Entering directory '/home/xdt/Public/wayne/openwrt/lede/build_dir/target-x86_64_musl/libselinux-3.3'
make[5]: Entering directory '/home/xdt/Public/wayne/openwrt/lede/build_dir/target-x86_64_musl/libselinux-3.3/include'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/xdt/Public/wayne/openwrt/lede/build_dir/target-x86_64_musl/libselinux-3.3/include'
make[5]: Entering directory '/home/xdt/Public/wayne/openwrt/lede/build_dir/target-x86_64_musl/libselinux-3.3/src'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/xdt/Public/wayne/openwrt/lede/build_dir/target-x86_64_musl/libselinux-3.3/src'
make[5]: Entering directory '/home/xdt/Public/wayne/openwrt/lede/build_dir/target-x86_64_musl/libselinux-3.3/utils'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/xdt/Public/wayne/openwrt/lede/build_dir/target-x86_64_musl/libselinux-3.3/utils'
make[5]: Entering directory '/home/xdt/Public/wayne/openwrt/lede/build_dir/target-x86_64_musl/libselinux-3.3/man'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/xdt/Public/wayne/openwrt/lede/build_dir/target-x86_64_musl/libselinux-3.3/man'
make[4]: Leaving directory '/home/xdt/Public/wayne/openwrt/lede/build_dir/target-x86_64_musl/libselinux-3.3'
CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/xdt/Public/wayne/openwrt/lede/build_dir/target-x86_64_musl/libselinux-3.3=libselinux-3.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/xdt/Public/wayne/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/xdt/Public/wayne/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/xdt/Public/wayne/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/xdt/Public/wayne/openwrt/lede/build_dir/target-x86_64_musl/libselinux-3.3=libselinux-3.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/xdt/Public/wayne/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/xdt/Public/wayne/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/xdt/Public/wayne/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include " LDFLAGS="-L/home/xdt/Public/wayne/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/xdt/Public/wayne/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro " make -C /home/xdt/Public/wayne/openwrt/lede/build_dir/target-x86_64_musl/libselinux-3.3/. AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/xdt/Public/wayne/openwrt/lede/build_dir/target-x86_64_musl/libselinux-3.3=libselinux-3.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" FTS_LDLIBS=-lfts SHLIBDIR=/usr/lib OS=Linux DESTDIR="/home/xdt/Public/wayne/openwrt/lede/build_dir/target-x86_64_musl/libselinux-3.3/ipkg-install"  install;
make[4]: Entering directory '/home/xdt/Public/wayne/openwrt/lede/build_dir/target-x86_64_musl/libselinux-3.3'
make[5]: Entering directory '/home/xdt/Public/wayne/openwrt/lede/build_dir/target-x86_64_musl/libselinux-3.3/include'
test -d /home/xdt/Public/wayne/openwrt/lede/build_dir/target-x86_64_musl/libselinux-3.3/ipkg-install/usr/include/selinux || install -m 755 -d /home/xdt/Public/wayne/openwrt/lede/build_dir/target-x86_64_musl/libselinux-3.3/ipkg-install/usr/include/selinux
install -m 644 selinux/get_context_list.h selinux/get_default_type.h selinux/avc.h selinux/context.h selinux/restorecon.h selinux/selinux.h selinux/label.h /home/xdt/Public/wayne/openwrt/lede/build_dir/target-x86_64_musl/libselinux-3.3/ipkg-install/usr/include/selinux
make[5]: Leaving directory '/home/xdt/Public/wayne/openwrt/lede/build_dir/target-x86_64_musl/libselinux-3.3/include'
make[5]: Entering directory '/home/xdt/Public/wayne/openwrt/lede/build_dir/target-x86_64_musl/libselinux-3.3/src'
test -d /home/xdt/Public/wayne/openwrt/lede/build_dir/target-x86_64_musl/libselinux-3.3/ipkg-install/usr/lib || install -m 755 -d /home/xdt/Public/wayne/openwrt/lede/build_dir/target-x86_64_musl/libselinux-3.3/ipkg-install/usr/lib
install: error while loading shared libraries: libpcre.so.1: cannot open shared object file: No such file or directory
make[5]: *** [Makefile:176: install] Error 127
make[5]: Leaving directory '/home/xdt/Public/wayne/openwrt/lede/build_dir/target-x86_64_musl/libselinux-3.3/src'
make[4]: *** [Makefile:48: install] Error 1
make[4]: Leaving directory '/home/xdt/Public/wayne/openwrt/lede/build_dir/target-x86_64_musl/libselinux-3.3'
make[3]: *** [Makefile:163: /home/xdt/Public/wayne/openwrt/lede/build_dir/target-x86_64_musl/libselinux-3.3/.built] Error 2
make[3]: Leaving directory '/home/xdt/Public/wayne/openwrt/lede/package/libs/libselinux'
time: package/libs/libselinux/compile#0.37#0.17#0.47
    ERROR: package/libs/libselinux failed to build.
make[2]: *** [package/Makefile:116: package/libs/libselinux/compile] Error 1
make[2]: Leaving directory '/home/xdt/Public/wayne/openwrt/lede'
make[1]: *** [package/Makefile:110: /home/xdt/Public/wayne/openwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/xdt/Public/wayne/openwrt/lede'
make: *** [/home/xdt/Public/wayne/openwrt/lede/include/toplevel.mk:231: world] Error 2
```

lovehackintosh, 2023-01-13:
如果未启用selinux,可以直接删掉package/libs/libselinux/Makefile

WYC-2020, 2023-01-16:
我这边x86 5.4内核一切正常，你是编译的什么内核

closed 2023-01-16 by verages

liwenjie119, 2023-09-17:
> 这个问题我遇到了，解决了。当初查询的时候，也搜到过这里，但没有找到答案。
> 
> 问题的表现是：install: error while loading shared libraries: libpcre.so.1: cannot open shared object file: No such file or directory
> 
> 造成问题的原因是，安装cuda后，系统中有LD_LIBRARY_PATH变量。共享库的默认搜索目录，会从LD_LIBRARY_PATH的赋值开始查找，于是出现找不到libpcre.so.1的错误。
> 
> 解决也很简单，在编译前，将LD_LIBRARY_PATH值设为空。 export LD_LIBRARY_PATH= 正常编译。编译完成后，重新赋值给LD_LIBRARY_PATH，以确保其他项目正常运行。其实Python项目通过venv或者conda env，可以单独设置LD_LIBRARY_PATH变量。维护的好，问题不大。
> 
> 我的方法放在这里，供后面的人参考，希望有帮助。

感谢分享，找了好久没发现啥原因，同是装了cuda