coolsnowwolf/lede #10769: 云编译kmod-mmc-spi报错

smallprogram, 2023-01-15:
### LOG
```log
make[3]: *** [modules/spi.mk:27: /workdir/openwrt/bin/targets/x86/64/packages/kmod-mmc-spi_6.1.6-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/workdir/openwrt/package/kernel/linux'
time: package/kernel/linux/compile#2.80#0.26#3.05
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.
```
### actionlink
https://github.com/smallprogram/OpenWrtAction/actions/runs/3920876384/jobs/6702755007

hgh188, 2023-01-15:
搞了3天 同样问题

smallprogram, 2023-01-15:
> 搞了3天 同样问题

临时解决方案，取消一下试试,好像是在kernel  SPI Support里面

ecrasy, 2023-01-15:
微软的Ubuntu镜像服务器down掉几天了
会不会和这个有关？
```
root@Ubuntu:~ # ping azure.archive.ubuntu.com -c 10
PING cloud-mirror-lb.westus.cloudapp.azure.com (40.81.13.82) 56(84) bytes of data.

--- cloud-mirror-lb.westus.cloudapp.azure.com ping statistics ---
10 packets transmitted, 0 received, 100% packet loss, time 9220ms
```
我是手动改了镜像才能编译：
sudo sed -i 's/azure.archive.ubuntu.com/us.archive.ubuntu.com/g' /etc/apt/sources.list

smallprogram, 2023-01-15:
> 微软的Ubuntu镜像服务器down掉几天了 会不会和这个有关？
>
> ```
> root@Ubuntu:~ # ping azure.archive.ubuntu.com -c 10
> PING cloud-mirror-lb.westus.cloudapp.azure.com (40.81.13.82) 56(84) bytes of data.
>
> --- [...]

跟这个没关系，我是Github Action

coolsnowwolf, 2023-01-16:
x64  不需要这个的

smallprogram, 2023-01-16:
> x64 不需要这个的

感谢大雕

closed 2023-01-16 by smallprogram

WYC-2020, 2023-01-16:
这个是那个模块限制有问题，模块的脚本限制5.15，所以自然你6.1编译不过并且5.15 6.1 已经去除了CONFIG_OF限制，参考这个提交https://lore.kernel.org/all/20210419112459.25241-6-andriy.shevchenko@linux.intel.com/，所以是否能把5.4 5.10的也去掉这个限制哈哈

commit fec4f5df (2023-01-27): kernel: test for kernel version greater 5.13 to fix mmc-spi

The of_mmc_spi.o resource is provider agnostic in kernels greater 5.13
and does not depend anymore on CONFIG_OF.

Fixes: #10769

Suggested-by: John Thomson <git@johnthomson.fastmail.com.au>