coolsnowwolf/lede #10770: 系统日志大量警告

binge8, 2023-01-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：系统日志大量警告

### (2) 路由器型号和固件版本

A：x86-64

### (3) 详细日志

A：
Sun Jan 15 21:18:04 2023 kern.info kernel: [15687.115716] TCP: too many orphaned sockets
Sun Jan 15 21:18:04 2023 kern.info kernel: [15687.116078] TCP: too many orphaned sockets
Sun Jan 15 21:18:04 2023 kern.info kernel: [15687.119902] TCP: too many orphaned sockets
Sun Jan 15 21:18:04 2023 kern.info kernel: [15687.123945] TCP: too many orphaned sockets
Sun Jan 15 21:18:04 2023 kern.info kernel: [15687.128039] TCP: too many orphaned sockets
Sun Jan 15 21:18:04 2023 kern.info kernel: [15687.132127] TCP: too many orphaned sockets
Sun Jan 15 21:18:04 2023 kern.info kernel: [15687.136333] TCP: too many orphaned sockets
Sun Jan 15 21:18:04 2023 kern.info kernel: [15687.140315] TCP: too many orphaned sockets
Sun Jan 15 21:18:04 2023 kern.info kernel: [15687.144451] TCP: too many orphaned sockets
Sun Jan 15 21:18:04 2023 kern.info kernel: [15687.148502] TCP: too many orphaned sockets

minlang112, 2023-01-15:
太多的孤儿socket……你下面有什么设备跑出了这么多TCP连接

binge8, 2023-01-17:
> net.ipv4.ip_local_port_range = 1024 65535 但还是建议自己查一下下游设备有什么东西跑出来怎么多TCP链接

明白了。感谢大佬指点

binge8, 2023-01-17:
> 调节一个内核tcp孤儿socket数量的参数，放到sysctl里，具体参数啥名自己谷歌吧

明白了。感谢大佬指点

binge8, 2023-01-17:
> 太多的孤儿socket……你下面有什么设备跑出了这么多TCP连接

明白了。感谢大佬指点

closed 2023-01-17 by binge8