coolsnowwolf/lede #10774: 自编译固件ext4-combined-efi.img，启动异常

seasun2k, 2023-01-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述

使用openwrt-x86-64-generic-ext4-combined-efi.img，虚拟机安装自编译的固件，启动后出现以下异常，但能正常使用。
[image]

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：esxi

### (3) 详细日志

A：

seasun2k, 2023-01-17:
应该和automount有关，去掉automount就不报错了，但是怎么才能编上automount？

coolsnowwolf, 2023-01-20:
看起来是

luoluomeng, 2023-01-30:
@seasun2k 请问你用的哪个内核版本？
我用6.x内核都没办法启动，5.15就可以

seasun2k, 2023-01-30:
> 

我用的6.1.6

Fyred, 2023-01-31:
6.1.8出现此问题，系统日志和内核日志均出现此报错

重编译5.15启动后仍然存在该问题

seasun2k, 2023-01-31:
> 

关掉automount

Fyred, 2023-01-31:
> > 
> 
> 关掉automount

关了是不是U盘什么的都不好用了

coolsnowwolf, 2026-09-05:
该 issue 只有“能正常使用”的截图，没有启动异常日志、镜像校验或 EFI/ESXi 配置，无法确认是当前镜像生成代码问题，关闭为 not planned。

closed 2026-09-05 by coolsnowwolf