coolsnowwolf/lede #10781: 6.1.6内核下AdGuardHome不定时导致OOM机制启用

moonerick, 2023-01-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
已经搜索，没有相关问题描述
## 2. 详细叙述

### (1) 具体问题

A：更新到6.1.6内核，出现了网页无法解析的问题，经查询，问题系AdguardHome出现了问题，目前的问题是在重新开机后的几小时至几十小时后，系统会认为AdguardHome有问题，从而使用oom机制，结束进程，但似乎又没能完全结束掉，从而导致后面不断启动oom机制，结束其它的进程，相关设置在5.15下面未曾出现过

### (2) 路由器型号和固件版本

A：x86 虚拟机

### (3) 详细日志

A：
Jan 17 21:18:55 OpenWrt kernel: [198903.886961] AdGuardHome invoked oom-killer: gfp_mask=0x140cca(GFP_HIGHUSER_MOVABLE|__GFP_COMP), order=0, oom_score_adj=0
Jan 17 21:18:55 OpenWrt kernel: [198903.908236] CPU: 2 PID: 26428 Comm: AdGuardHome Not tainted 6.1.6 #0
Jan 17 21:18:55 OpenWrt kernel: [198903.914226] Hardware name: QEMU Standard PC (Q35 + ICH9, 2009), BIOS 0.0.0 02/06/2015
Jan 17 21:18:55 OpenWrt kernel: [198903.951244] Call Trace:
Jan 17 21:18:55 OpenWrt kernel: [198903.956318]  <TASK>
Jan 17 21:18:55 OpenWrt kernel: [198903.958422]  dump_stack_lvl+0x4d/0x63
Jan 17 21:18:55 OpenWrt kernel: [198903.973653]  dump_stack+0x14/0x16
Jan 17 21:18:55 OpenWrt kernel: [198903.976373]  dump_header+0x4a/0x1a4
Jan 17 21:18:55 OpenWrt kernel: [198903.979226]  oom_kill_process.cold+0xb/0x10
Jan 17 21:18:55 OpenWrt kernel: [198903.997652]  out_of_memory+0x10b/0x2e0
Jan 17 21:18:55 OpenWrt kernel: [198904.000429]  __alloc_pages_slowpath.constprop.0+0xa9d/0xc60
Jan 17 21:18:55 OpenWrt kernel: [198904.006347]  __alloc_pages+0x25f/0x290
Jan 17 21:18:55 OpenWrt kernel: [198904.016452]  __folio_alloc+0x13/0x20
Jan 17 21:18:55 OpenWrt kernel: [198904.019012]  __filemap_get_folio+0x167/0x2d0
Jan 17 21:18:55 OpenWrt kernel: [198904.021077]  filemap_fault+0x3cf/0x7a0
Jan 17 21:18:55 OpenWrt kernel: [198904.036080]  f2fs_filemap_fault+0xd/0x10
Jan 17 21:18:55 OpenWrt kernel: [198904.039434]  __do_fault+0x31/0x90
Jan 17 21:18:55 OpenWrt kernel: [198904.041154]  __handle_mm_fault+0x9d0/0xdc0
Jan 17 21:18:55 OpenWrt kernel: [198904.043562]  handle_mm_fault+0x1b0/0x240
Jan 17 21:18:55 OpenWrt kernel: [198904.046331]  exc_page_fault+0x1aa/0x5a0
Jan 17 21:18:55 OpenWrt kernel: [198904.048847]  asm_exc_page_fault+0x2b/0x30
Jan 17 21:18:55 OpenWrt kernel: [198904.051225] RIP: 0033:0x46b900
Jan 17 21:18:55 OpenWrt kernel: [198904.053573] Code: Unable to access opcode bytes at 0x46b8d6.
Jan 17 21:18:55 OpenWrt kernel: [198904.056805] RSP: 002b:000000c000f31968 EFLAGS: 00010202
Jan 17 21:18:55 OpenWrt kernel: [198904.070271] RAX: 0000000000000000 RBX: 00000000001b30e0 RCX: 00000000001b30e0
Jan 17 21:18:55 OpenWrt kernel: [198904.073707] RDX: 000000c000f319c0 RSI: 000000c000f31af0 RDI: 0000000000000017
Jan 17 21:18:55 OpenWrt kernel: [198904.077801] RBP: 000000c000f319b0 R08: 000000000236c001 R09: 000000000236c040
Jan 17 21:18:55 OpenWrt kernel: [198904.081073] R10: 00000000001c7f90 R11: 0000000000014eb0 R12: 0000000000000000
Jan 17 21:18:55 OpenWrt kernel: [198904.083882] R13: 0000000000000000 R14: 000000c000b2c1a0 R15: 0000000000000000
Jan 17 21:18:55 OpenWrt kernel: [198904.086805]  </TASK>
Jan 17 21:18:55 OpenWrt kernel: [198904.088319] Mem-Info:
Jan 17 21:18:55 OpenWrt kernel: [198904.090207] active_anon:5852 inactive_anon:3723 isolated_anon:0
Jan 17 21:18:55 OpenWrt kernel: [198904.090207]  active_file:1042 inactive_file:268 isolated_file:0
Jan 17 21:18:55 OpenWrt kernel: [198904.090207]  unevictable:0 dirty:0 writeback:0
Jan 17 21:18:55 OpenWrt kernel: [198904.090207]  slab_reclaimable:1464 slab_unreclaimable:96217
Jan 17 21:18:55 OpenWrt kernel: [198904.090207]  mapped:4 shmem:189 pagetables:239
Jan 17 21:18:55 OpenWrt kernel: [198904.090207]  sec_pagetables:0 bounce:0
Jan 17 21:18:55 OpenWrt kernel: [198904.090207]  kernel_misc_reclaimable:0
Jan 17 21:18:55 OpenWrt kernel: [198904.090207]  free:4647 free_pcp:103 free_cma:0
Jan 17 21:18:55 OpenWrt kernel: [198904.132821] Node 0 active_anon:23408kB inactive_anon:14892kB active_file:144kB inactive_file:5804kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:244kB dirty:0kB writeback:0kB shmem:756kB writeback_tmp:0kB kernel_stack:2080kB pagetables:956kB sec_pagetables:0kB all_unreclaimable? no
Jan 17 21:18:55 OpenWrt kernel: [198904.164221] DMA free:2288kB boost:0kB min:476kB low:592kB high:708kB reserved_highatomic:0KB active_anon:80kB inactive_anon:836kB active_file:0kB inactive_file:304kB unevictable:0kB writepending:0kB present:15000kB managed:14364kB mlocked:0kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:0kB
Jan 17 21:18:55 OpenWrt kernel: [198904.190720] lowmem_reserve[]: 0 460 460 460
Jan 17 21:18:55 OpenWrt kernel: [198904.193392] DMA32 free:17460kB boost:0kB min:15904kB low:19880kB high:23856kB reserved_highatomic:0KB active_anon:14056kB inactive_anon:23328kB active_file:0kB inactive_file:2844kB unevictable:0kB writepending:0kB present:504000kB managed:476088kB mlocked:0kB bounce:0kB free_pcp:608kB local_pcp:0kB free_cma:0kB
Jan 17 21:18:55 OpenWrt kernel: [198904.212587] lowmem_reserve[]: 0 0 0 0
Jan 17 21:18:55 OpenWrt kernel: [198904.214568] DMA: 4*4kB (U) 4*8kB (U) 44*16kB (U) 31*32kB (UE) 9*64kB (UE) 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 2320kB
Jan 17 21:18:55 OpenWrt kernel: [198904.219339] DMA32: 98*4kB (UE) 169*8kB (UME) 153*16kB (UME) 68*32kB (UME) 33*64kB (UME) 20*128kB (UME) 11*256kB (UME) 3*512kB (U) 1*1024kB (U) 0*2048kB 0*4096kB = 16416kB
Jan 17 21:18:55 OpenWrt kernel: [198904.228216] 1773 total pagecache pages
Jan 17 21:18:55 OpenWrt kernel: [198904.233235] 0 pages in swap cache
Jan 17 21:18:55 OpenWrt kernel: [198904.245639] Free swap  = 0kB
Jan 17 21:18:55 OpenWrt kernel: [198904.247649] Total swap = 0kB
[... 266 lines cut ...]
Jan 17 22:31:06 OpenWrt kernel: [203235.504039] [    805]     0   805      283       17    45056        0             0 ash
Jan 17 22:31:06 OpenWrt kernel: [203235.519827] [   1920]   514  1920      248       37    36864        0             0 logd
Jan 17 22:31:06 OpenWrt kernel: [203235.522946] [   1921]     0  1921      255       26    36864        0             0 logread
Jan 17 22:31:06 OpenWrt kernel: [203235.527843] [   1973]     0  1973      280       30    36864        0             0 rpcd
Jan 17 22:31:06 OpenWrt kernel: [203235.531118] [   2331]     0  2331      332       42    36864        0             0 netifd
Jan 17 22:31:06 OpenWrt kernel: [203235.534312] [   2407]     0  2407      302       28    40960        0             0 odhcpd
Jan 17 22:31:06 OpenWrt kernel: [203235.558105] [   2697]     0  2697     1185      138    49152        0             0 uhttpd
Jan 17 22:31:06 OpenWrt kernel: [203235.561542] [   3342]     0  3342      193       13    40960        0             0 vlmcsd
Jan 17 22:31:06 OpenWrt kernel: [203235.566824] [   3782]     0  3782      234       14    45056        0             0 dropbear
Jan 17 22:31:06 OpenWrt kernel: [203235.576482] [   3896]     0  3896      281       14    45056        0             0 ntpd
Jan 17 22:31:06 OpenWrt kernel: [203235.579871] [   4173]     0  4173     1363      100    45056        0             0 ttyd
Jan 17 22:31:06 OpenWrt kernel: [203235.583294] [  23938]     0 23938      208       15    36864        0             0 odhcp6c
Jan 17 22:31:06 OpenWrt kernel: [203235.586372] [  19773]     0 19773      279       37    45056        0             0 pppd
Jan 17 22:31:06 OpenWrt kernel: [203235.589459] [  20225]     0 20225      335       31    40960        0             0 msd_lite
Jan 17 22:31:06 OpenWrt kernel: [203235.593209] [  27617]     0 27617      281       13    36864        0             0 crond
Jan 17 22:31:06 OpenWrt kernel: [203235.596294] [  27773]   453 27773      531      248    49152        0             0 dnsmasq
Jan 17 22:31:06 OpenWrt kernel: [203235.600352] [  19165]     0 19165   187207     6079   163840        0             0 AdGuardHome
Jan 17 22:31:06 OpenWrt kernel: [203235.613038] [  19228]     0 19228      234       19    36864        0             0 ubus
Jan 17 22:31:06 OpenWrt kernel: [203235.616110] [  19234]     0 19234      358       89    40960        0             0 dhcpv6.script
Jan 17 22:31:06 OpenWrt kernel: [203235.619404] [  19236]     0 19236      225       15    40960        0             0 jshn
Jan 17 22:31:06 OpenWrt kernel: [203235.622670] [  19239]     0 19239      306       32    45056        0             0 dhcpv6.script
Jan 17 22:31:06 OpenWrt kernel: [203235.626197] [  19240]     0 19240      208       16    36864        0             0 odhcp6c
Jan 17 22:31:06 OpenWrt kernel: [203235.629399] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/,task=AdGuardHome,pid=19165,uid=0
Jan 17 22:31:06 OpenWrt kernel: [203235.634285] Out of memory: Killed process 19165 (AdGuardHome) total-vm:748828kB, anon-rss:24316kB, file-rss:0kB, shmem-rss:0kB, UID:0 pgtables:160kB oom_score_adj:0
Jan 17 22:33:05 OpenWrt kernel: [203354.623419] jshn invoked oom-killer: gfp_mask=0x140cca(GFP_HIGHUSER_MOVABLE|__GFP_COMP), order=0, oom_score_adj=0

Cathgao, 2023-01-18:
日志过大

Travel2Here, 2023-01-18:
貌似遇到了相同问题，我是R4S。折腾了我两三天各种排查。烦死了~手贱。还好有6.1老版本的备份固件 那里面没装ADG。
我的初步排查发现adg或者passwall有问题。passwall2正常但访问控制不太正常。

moonerick, 2023-01-19:
> 貌似遇到了相同问题，我是R4S。折腾了我两三天各种排查。烦死了~手贱。还好有6.1老版本的备份固件 那里面没装ADG。 我的初步排查发现adg或者passwall有问题。passwall2正常但访问控制不太正常。 表现上它们跟dnsmasq或者firewall不太兼容。

最开始大雕启用6.0测试版的时候，另外一个大佬说adg在6.0上存在内存溢出，我更新到6.1.5运行了两三天貌似没有出现这个问题，更新到6.1.6后就又出现这个问题了，目前已经退回5.15.88了，估计需要adg来维护

Wilson-coder, 2023-01-19:
6.1内核好像小问题还是挺多的，我前几天编译了6.1.5，升级之后发现FTP服务器无法挂载路由器内部的EXT4分区，回退5.15就一切正常了。

Travel2Here, 2023-01-19:
> > 貌似遇到了相同问题，我是R4S。折腾了我两三天各种排查。烦死了~手贱。还好有6.1老版本的备份固件 那里面没装ADG。 我的初步排查发现adg或者passwall有问题。passwall2正常但访问控制不太正常。 表现上它们跟dnsmasq或者firewall不太兼容。
> 
> 最开始大雕启用6.0测试版的时候，另外一个大佬说adg在6.0上存在内存溢出，我更新到6.1.5运行了两三天貌似没有出现这个问题，更新到6.1.6后就又出现这个问题了，目前已经退回5.15.88了，估计需要adg来维护

兄弟5.15.88我也不行，你配了哪些app？我有adg、passwall、passwall2、一旦启动pw依然不能正常访问一部分国内站，被代理的都正常。手动重启network才行。日志里压根啥异样都没。
基础环境：lede的最新、kenzok8的small app库。

所以目前最大的怀疑还是passwall、dnsmasq、adguard这三个的问题。

moonerick, 2023-01-19:
> 

我adg, vssr(老竭力修改的passwall），msd_lite, kms, udpxy(未启用)，dnsmasq-full

closed 2023-01-23 by moonerick