coolsnowwolf/lede #10786: R66s 不定時reboot失敗 頻率比其他固件高出很多

kousyougi, 2023-01-19:
## 1.关于你要提交的问题

R66s 不定時reboot失敗 頻率比其他固件高出很多

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

一開始以為是TF卡品質不好，測試過不同品牌跟規格的TF卡都沒解決這個問題。
偶然拿其他的來源固件來測試發現跑起來很正常。

使用R68cn官網提供的 [Lean国际版固件]
[20220909_4.19内核_无Docker_openwrt-rockchip-R22.9.9-fastrhino_r66s-squashfs-sysupgrade.img.gz](https://drive.google.com/file/d/1gG2PGz-KgG5lQejyrVfIzzgkBNLpC7xz/view?usp=share_link)
跟自己從immortalwrt編譯出來的R66s固件，是正常的。定時五分鐘reboot，各跑兩天，reboot了大概五百次，都正常。

但自己編譯的LEDE 5.15、6.1 固件、ext4 或 squashfs，有用預設的.config，也有自己精簡很多的固件，定時reoot有很大概率會死機開開機失敗。也測試了這邊releases提供的 [openwrt-rockchip-R22.11.13-fastrhino_r66s-squashfs-sysupgrade.img.gz](https://github.com/coolsnowwolf/lede/releases/download/20220716/openwrt-rockchip-R22.11.13-fastrhino_r66s-squashfs-sysupgrade.img.gz)
一樣也是有這個問題。

測試方法是：crontab定時五分鐘reboot，刷完固件只簡單的設定就直接定時執行reboot，沒有跑其他程序，另一台設備每分鐘ping R66s，檢查什麼時候死機。有時候一個小時死機，有時大概半小時就死機了。沒有撐超過半天過。

有閒置R66s設備的人如果可以也請幫忙測一下，刷這邊releases的固件，測每五分鐘reoot，看一個小時後機器有沒有死掉。

AutoCONFIG, 2023-01-19:
我的r66s reboot关机后经常无法再次boot（不亮蓝灯），必须要手动拔插电源才行，习惯了，反正重启次数也不频繁。

问了客服，他们说是tf卡的锅，但是我换了几个tf卡还都是一样（包括买机子时候送的tf卡）。

我猜测可能是atf(-vendor)的问题，可以试试换一个版本的atf

kousyougi, 2023-01-19:
> 我的r66s reboot关机后经常无法再次boot（不亮蓝灯），必须要手动拔插电源才行，习惯了，反正重启次数也不频繁。
> 
> 问了客服，他们说是tf卡的锅，但是我换了几个tf卡还都是一样（包括买机子时候送的tf卡）。
> 
> 我猜测可能是atf(-vendor)的问题，可以试试换一个版本的atf

你要試試看上面固件嗎？
[20220909_4.19内核_无Docker_openwrt-rockchip-R22.9.9-fastrhino_r66s-squashfs-sysupgrade.img.gz](https://drive.google.com/file/d/1gG2PGz-KgG5lQejyrVfIzzgkBNLpC7xz/view?usp=share_link)
我測試正常。想要釐清問題。

AutoCONFIG, 2023-01-19:
> 你要試試看上面固件嗎？ [20220909_4.19内核_无Docker_openwrt-rockchip-R22.9.9-fastrhino_r66s-squashfs-sysupgrade.img.gz](https://drive.google.com/file/d/1gG2PGz-KgG5lQejyrVfIzzgkBNLpC7xz/view?usp=share_link) [...]

不折腾了，反正重启也不频繁，几乎就不重启的，影响不大。
确定immortalwrt没问题的话，你可以尝试用immortalwrt的替换lede的如下目录：
lede/target/linux/rockchip
lede/package/boot/arm-trusted-firmware-rockchip （删掉lede/package/boot/arm-trusted-firmware-rockchip-vendor，因为immortalwrt的arm-trusted-firmware-rockchip与lede的arm-trusted-firmware-rockchip-vendor是一样的）
lede/package/boot/uboot-rockchip
可能需要refresh patch，不过这都是小问题。
如果问题依旧的话，可能需要大佬来解答了

kousyougi, 2023-01-19:
> 

immortalwrt的reboot幾乎確定是沒問題的。這禮拜前前後後編譯不同版本的immortalwrt固件，測試reboot指令加起來超過上千次了。還沒有reboot失敗過。

immortalwrt現在最高版本又只到5.15，lede到6.1，複製過去也只能編譯5.15的固件。refresh patch 不會。
我做不來。放棄

coolsnowwolf, 2023-01-19:
也许是主线 uboot 的问题，待确认

kousyougi, 2023-05-05:
前几天编译的lede貌似没这个问题了。reboot了两百多次，都没死掉。

closed 2023-05-05 by kousyougi