coolsnowwolf/lede #10788: x86无线编译出错

zhg-em, 2023-01-20:
x86无线编译出错，不知大家有没有遇到这个问题，下面是错误信息：
                 from ./include/linux/if_ether.h:19,
                 from /home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/backports-5.15.33-1/backport-include/linux/if_ether.h:3,
                 from ./include/linux/etherdevice.h:20,
                 from /home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/backports-5.15.33-1/backport-include/linux/etherdevice.h:3,
                 from /home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/backports-5.15.33-1/drivers/net/wireless/intel/iwlwifi/mvm/mac-ctxt.c:7:
/home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/backports-5.15.33-1/include/linux/overflow.h:32: warning: "is_signed_type" redefined
   32 | #define is_signed_type(type)       (((type)(-1)) < (type)1)
      |
In file included from /home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/backports-5.15.33-1/backport-include/linux/compiler.h:3,
                 from ./include/linux/export.h:33,
                 from ./include/linux/linkage.h:7,
                 from ./include/linux/kernel.h:17,
                 from /home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/backports-5.15.33-1/backport-include/linux/kernel.h:3,
                 from ./include/linux/skbuff.h:13,
                 from /home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/backports-5.15.33-1/backport-include/linux/skbuff.h:3,
                 from ./include/linux/if_ether.h:19,
                 from /home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/backports-5.15.33-1/backport-include/linux/if_ether.h:3,
                 from ./include/linux/etherdevice.h:20,
                 from /home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/backports-5.15.33-1/backport-include/linux/etherdevice.h:3,
                 from /home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/backports-5.15.33-1/drivers/net/wireless/intel/iwlwifi/mvm/mac-ctxt.c:7:
./include/linux/compiler.h:240: note: this is the location of the previous definition
  240 | #define is_signed_type(type) (((type)(-1)) < (__force type)1)
      |
/home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/backports-5.15.33-1/drivers/net/wireless/intel/iwlwifi/mvm/mac-ctxt.c: In function 'iwl_mvm_mac_ctxt_cmd_fill_ap':
/home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/backports-5.15.33-1/drivers/net/wireless/intel/iwlwifi/mvm/mac-ctxt.c:1102:37: error: implicit declaration of function 'prandom_u32'; did you mean 'prandom_u32_max'? [-Werror=implicit-function-declaration]
 1102 |                         u32 rand = (prandom_u32() % (64 - 36)) + 36;
      |                                     ^~~~~~~~~~~
      |                                     prandom_u32_max
cc1: some warnings being treated as errors
make[12]: *** [scripts/Makefile.build:250: /home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/backports-5.15.33-1/drivers/net/wireless/intel/iwlwifi/mvm/mac-ctxt.o] Error 1
make[11]: *** [scripts/Makefile.build:500: /home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/backports-5.15.33-1/drivers/net/wireless/intel/iwlwifi/mvm] Error 2
make[10]: *** [scripts/Makefile.build:500: /home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/backports-5.15.33-1/drivers/net/wireless/intel/iwlwifi] Error 2
make[9]: *** [scripts/Makefile.build:500: /home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/backports-5.15.33-1/drivers/net/wireless/intel] Error 2
make[8]: *** [scripts/Makefile.build:500: /home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/backports-5.15.33-1/drivers/net/wireless] Error 2
make[7]: *** [Makefile:1992: /home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/backports-5.15.33-1] Error 2
make[6]: *** [Makefile.build:13: modules] Error 2
make[5]: *** [Makefile.real:93: modules] Error 2
make[4]: *** [Makefile:121: modules] Error 2
make[4]: Leaving directory '/home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/backports-5.15.33-1'
make[3]: *** [Makefile:397: /home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/backports-5.15.33-1/.built] Error 2
make[3]: Leaving directory '/home/xyz/lede/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#15.08#3.25#26.26
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/home/xyz/lede'
make[1]: *** [package/Makefile:110: /home/xyz/lede/staging_dir/target-i386_pentium4_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/xyz/lede'
make: *** [/home/xyz/lede/include/toplevel.mk:231：world] 错误 2

1582130940, 2023-01-21:
你不要勾CONFIG_PACKAGE_kmod-iwlwifi，你要跑iwlwifi的话用OP主线的mac80211替换掉lean的

zhg-em, 2023-01-22:
> 你不要勾CONFIG_PACKAGE_kmod-iwlwifi，你要跑iwlwifi的话用OP主线的mac80211替换掉lean的

谢谢回复，取消CONFIG_PACKAGE_kmod-iwlwifi后编译还是出错！
                 from ./include/linux/tracepoint.h:22,
                 from /home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/mt76-2022-12-22-5b509e80/usb_trace.h:9,
                 from /home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/mt76-2022-12-22-5b509e80/usb_trace.c:10:
/home/xyz/lede/staging_dir/target-i386_pentium4_musl/usr/include/mac80211/linux/overflow.h:32: warning: "is_signed_type" redefined
   32 | #define is_signed_type(type)       (((type)(-1)) < (type)1)
      |
In file included from /home/xyz/lede/staging_dir/target-i386_pentium4_musl/usr/include/mac80211-backport/linux/compiler.h:3,
                 from ./include/linux/build_bug.h:5,
                 from /home/xyz/lede/staging_dir/target-i386_pentium4_musl/usr/include/mac80211-backport/linux/build_bug.h:5,
                 from ./include/linux/container_of.h:5,
                 from ./include/linux/list.h:5,
                 from ./include/linux/module.h:12,
                 from /home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/mt76-2022-12-22-5b509e80/usb_trace.c:6:
./include/linux/compiler.h:240: note: this is the location of the previous definition
  240 | #define is_signed_type(type) (((type)(-1)) < (__force type)1)
      |
  LD [M]  /home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/mt76-2022-12-22-5b509e80/mt76-usb.o
  CC [M]  /home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/mt76-2022-12-22-5b509e80/mt7603/pci.o
In file included from ./include/linux/device.h:29,
                 from /home/xyz/lede/staging_dir/target-i386_pentium4_musl/usr/include/mac80211-backport/linux/device.h:3,
                 from ./include/linux/pci.h:37,
                 from /home/xyz/lede/staging_dir/target-i386_pentium4_musl/usr/include/mac80211-backport/linux/pci.h:3,
                 from /home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/mt76-2022-12-22-5b509e80/mt7603/pci.c:5:
/home/xyz/lede/staging_dir/target-i386_pentium4_musl/usr/include/mac80211/linux/overflow.h:32: error: "is_signed_type" redefined [-Werror]
   32 | #define is_signed_type(type)       (((type)(-1)) < (type)1)
      |
In file included from /home/xyz/lede/staging_dir/target-i386_pentium4_musl/usr/include/mac80211-backport/linux/compiler.h:3,
                 from ./include/linux/export.h:33,
                 from ./include/linux/linkage.h:7,
                 from ./include/linux/kernel.h:17,
                 from /home/xyz/lede/staging_dir/target-i386_pentium4_musl/usr/include/mac80211-backport/linux/kernel.h:3,
                 from /home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/mt76-2022-12-22-5b509e80/mt7603/pci.c:3:
./include/linux/compiler.h:240: note: this is the location of the previous definition
  240 | #define is_signed_type(type) (((type)(-1)) < (__force type)1)
      |
cc1: all warnings being treated as errors
make[6]: *** [scripts/Makefile.build:250: /home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/mt76-2022-12-22-5b509e80/mt7603/pci.o] Error 1
make[5]: *** [scripts/Makefile.build:500: /home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/mt76-2022-12-22-5b509e80/mt7603] Error 2
make[4]: *** [Makefile:1992: /home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/mt76-2022-12-22-5b509e80] Error 2
make[4]: Leaving directory '/home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-6.1.7'
make[3]: *** [Makefile:542: /home/xyz/lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/mt76-2022-12-22-5b509e80/.built] Error 2
make[3]: Leaving directory '/home/xyz/lede/package/kernel/mt76'
time: package/kernel/mt76/compile#23.15#4.24#31.05
    ERROR: package/kernel/mt76 failed to build.
make[2]: *** [package/Makefile:116: package/kernel/mt76/compile] Error 1
make[2]: Leaving directory '/home/xyz/lede'
make[1]: *** [package/Makefile:110: /home/xyz/lede/staging_dir/target-i386_pentium4_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/xyz/lede'
make: *** [/home/xyz/lede/include/toplevel.mk:231：world] 错误 2

1582130940, 2023-01-22:
你不会Wireless Drivers部分全勾了吧？
mt76能过编译的只建议勾选
CONFIG_PACKAGE_kmod-mt7601u=y
CONFIG_PACKAGE_kmod-mt7663-firmware-ap=y
CONFIG_PACKAGE_kmod-mt7663-firmware-sta=y
或者都不勾选，等你有联发科无线网卡需求再勾选，没有无线网卡需求不用选mt76部分

zhg-em, 2023-01-22:
> 你不会Wireless Drivers部分全勾了吧？ mt76能过编译的只建议勾选 CONFIG_PACKAGE_kmod-mt7601u=y CONFIG_PACKAGE_kmod-mt7663-firmware-ap=y CONFIG_PACKAGE_kmod-mt7663-firmware-sta=y 或者都不勾选，等你有联发科无线网卡需求再勾选，没有无线网卡需求不用选mt76部分

非常感谢你的解答！

我就是有一个MediaTek的无线网卡，应该是7612u的芯片，去年12月份的时候还是可以编译的。最近更新后就不能编译了！

zhg-em, 2023-02-06:
谢谢开发人员，问题已经解决。此issue可以关闭！

coolsnowwolf, 2026-09-05:
旧报告使用 mac80211 backports 5.15.33，因上游移除 `prandom_u32()` 导致 iwlwifi 编译失败。当前 mac80211/backports 已更新，代码不再使用该旧接口，关闭为 completed。

closed 2026-09-05 by coolsnowwolf