coolsnowwolf/lede #10796: X64编译失败，intel-microcode需更新版本

jliveusa, 2023-01-23:
[intel-microcode_3.20221108.1.tar.xz](http://ftp.debian.org/debian/pool/non-free/i/intel-microcode/intel-microcode_3.20221108.1.tar.xz)

lovehackintosh, 2023-01-23:
发下日志看看

jliveusa, 2023-01-23:
补一下：
2023-01-23T06:56:43.6253732Z mkdir -p /workdir/openwrt/dl
2023-01-23T06:56:43.6275263Z SHELL= flock /workdir/openwrt/tmp/.intel-microcode_3.20220809.1.tar.xz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "intel-microcode_3.20220809.1.tar.xz" "4cf6c3638bb52d9d45c1916af866fd0929628a6f459daac3edfd369149e9c665" "" "@DEBIAN/pool/non-free/i/intel-microcode/"    '
2023-01-23T06:56:43.6555429Z + curl -f --connect-timeout 20 --retry 5 --location https://mirrors.aliyun.com/debian/pool/non-free/i/intel-microcode/intel-microcode_3.20220809.1.tar.xz
2023-01-23T06:56:43.6615111Z   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
2023-01-23T06:56:43.6618230Z                                  Dload  Upload   Total   Spent    Left  Speed
2023-01-23T06:56:43.6618413Z 
2023-01-23T06:56:44.7059265Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2023-01-23T06:56:44.7061290Z   0  2318    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
2023-01-23T06:56:44.7063694Z curl: (22) The requested URL returned error: 404
2023-01-23T06:56:44.7092640Z Download failed.
2023-01-23T06:56:44.7094506Z + curl -f --connect-timeout 20 --retry 5 --location https://mirrors.tencent.com/debian/pool/non-free/i/intel-microcode/intel-microcode_3.20220809.1.tar.xz
2023-01-23T06:56:44.7157566Z   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
2023-01-23T06:56:44.7162183Z                                  Dload  Upload   Total   Spent    Left  Speed
2023-01-23T06:56:44.7163192Z 
2023-01-23T06:56:45.2387481Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2023-01-23T06:56:46.4307011Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2023-01-23T06:56:46.4311132Z   0   244    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
2023-01-23T06:56:46.4322683Z curl: (22) The requested URL returned error: 404
2023-01-23T06:56:46.4343486Z Download failed.
2023-01-23T06:56:46.4346514Z + curl -f --connect-timeout 20 --retry 5 --location https://ftp.debian.org/debian/pool/non-free/i/intel-microcode/intel-microcode_3.20220809.1.tar.xz
2023-01-23T06:56:46.4413045Z   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
2023-01-23T06:56:46.4416955Z                                  Dload  Upload   Total   Spent    Left  Speed
2023-01-23T06:56:46.4417124Z 
2023-01-23T06:56:46.8245453Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2023-01-23T06:56:46.8247027Z   0   260    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2023-01-23T06:56:46.8248022Z curl: (22) The requested URL returned error: 404
2023-01-23T06:56:46.8275757Z Download failed.
2023-01-23T06:56:46.8279269Z + curl -f --connect-timeout 20 --retry 5 --location https://mirror.leaseweb.com/debian/pool/non-free/i/intel-microcode/intel-microcode_3.20220809.1.tar.xz
2023-01-23T06:56:46.8353681Z   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
2023-01-23T06:56:46.8354611Z                                  Dload  Upload   Total   Spent    Left  Speed
2023-01-23T06:56:46.8354977Z 
2023-01-23T06:56:47.1314586Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2023-01-23T06:56:47.1316638Z   0 15657    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2023-01-23T06:56:47.1318656Z curl: (22) The requested URL returned error: 404
2023-01-23T06:56:47.1349322Z Download failed.
2023-01-23T06:56:47.1352932Z + curl -f --connect-timeout 20 --retry 5 --location https://mirror.netcologne.de/debian/pool/non-free/i/intel-microcode/intel-microcode_3.20220809.1.tar.xz
2023-01-23T06:56:47.1419151Z   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
2023-01-23T06:56:47.1420151Z                                  Dload  Upload   Total   Spent    Left  Speed
2023-01-23T06:56:47.1420789Z 
2023-01-23T06:56:47.2414599Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2023-01-23T06:56:47.8099502Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2023-01-23T06:56:47.8100849Z   0   153    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2023-01-23T06:56:47.8101673Z curl: (22) The requested URL returned error: 404
2023-01-23T06:56:47.8131688Z Download failed.
2023-01-23T06:56:47.8135167Z + curl -f --connect-timeout 20 --retry 5 --location https://sources.cdn.openwrt.org/intel-microcode_3.20220809.1.tar.xz
2023-01-23T06:56:47.8203919Z   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
2023-01-23T06:56:47.8204888Z                                  Dload  Upload   Total   Spent    Left  Speed
2023-01-23T06:56:47.8205091Z 
2023-01-23T06:56:48.7312890Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2023-01-23T06:56:48.7313437Z   0   467    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2023-01-23T06:56:48.7313778Z curl: (22) The requested URL returned error: 503
2023-01-23T06:56:48.7314168Z Warning: Problem : HTTP error. Will retry in 1 seconds. 5 retries left.
2023-01-23T06:56:49.7325414Z 
2023-01-23T06:56:50.3851377Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2023-01-23T06:56:50.3853880Z   0   467    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
[... cut ...]
2023-01-23T06:57:24.0503400Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2023-01-23T06:57:24.0507335Z   0   153    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2023-01-23T06:57:24.0511804Z curl: (22) The requested URL returned error: 404
2023-01-23T06:57:24.0531433Z Download failed.
2023-01-23T06:57:24.0532398Z No more mirrors to try - giving up.
2023-01-23T06:57:24.0548568Z make[3]: *** [Makefile:52: /workdir/openwrt/dl/intel-microcode_3.20220809.1.tar.xz] Error 2

lovehackintosh, 2023-01-23:
`wget https://github.com/coolsnowwolf/lede/pull/10797.patch && git am 10797.patch`试试

PR #10797 (2023-01-23): intel-microcode: update to 20221108
Q：你知道这是`pull request`吗？(使用 "x" 选择)
* [x] 我知道

Fixes: #10796
commit c9e74f86 (2023-01-23): intel-microcode: update to 20221108 (#10797)

Fixes: #10796
closed 2023-01-23 by aiamadeus