coolsnowwolf/lede #10806: 春节期间的源码编译时不显示详细日志

iKira, 2023-01-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [×] 没有类似的issue

## 2. 详细叙述

春节期间的源码编译时不显示详细日志了，只有下面的基础信息，请问如何开启之前的详细日志？

A：

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：

 make[1] world
 make[2] tools/compile
 make[2] package/cleanup
 make[3] -C tools/flock compile
 make[3] -C tools/xz compile
 make[3] -C tools/sed compile
 make[3] -C tools/tar compile
 make[3] -C tools/patch compile
 make[3] -C tools/m4 compile
 make[3] -C tools/autoconf-archive compile
 make[3] -C tools/expat compile
 make[3] -C tools/lzma compile
 make[3] -C tools/cpio compile
 make[3] -C tools/mtools compile
 make[3] -C tools/ninja compile
 make[3] -C tools/zlib compile
 make[3] -C tools/patch-image compile
 make[3] -C tools/sstrip compile
 make[3] -C tools/zip compile
 make[3] -C tools/make-ext4fs compile
 make[3] -C tools/squashfskit4 compile
 make[3] -C tools/meson compile
 make[3] -C tools/pkgconf compile
 make[3] -C tools/zstd compile
 make[3] -C tools/libressl compile
 make[3] -C tools/autoconf compile
 make[3] -C tools/elfutils compile
 make[3] -C tools/automake compile
 make[3] -C tools/missing-macros compile
 make[3] -C tools/libtool compile
 make[3] -C tools/dosfstools compile
 make[3] -C tools/cmake compile
 make[3] -C tools/flex compile
 make[3] -C tools/e2fsprogs compile
 make[3] -C tools/fakeroot compile
 make[3] -C tools/gengetopt compile
 make[3] -C tools/mklibs compile
 make[3] -C tools/patchelf compile
 make[3] -C tools/gmp compile
 make[3] -C tools/mtd-utils compile
 make[3] -C tools/bison compile
 make[3] -C tools/mpfr compile

lovehackintosh, 2023-01-25:
v字母要大写

iKira, 2023-01-25:
> v字母要大写

谢谢

closed 2023-01-25 by iKira