coolsnowwolf/lede #10811: php7编译过程仍然出错

Fly432, 2023-01-26:
反馈bug/问题模板，提建议请删除

## 1.php7编译过程仍然出错

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：php7编译仍然出错

### (2) 路由器型号和固件版本

A：X86，R231.1

### (3) 详细日志

A：In file included from /home/gyk/lede/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend.h:31:
/home/gyk/lede/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend_llist.h:31:80: warning: type defaults to 'int' in declaration of 'args' [-Wimplicit-int]
   31 | typedef void (*llist_apply_with_args_func_t)(void *data, int num_args, va_list args);
      |                                                                                ^~~~
In file included from /home/gyk/lede/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend.h:33:
/home/gyk/lede/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend_hash.h:149:69: warning: type defaults to 'int' in declaration of 'args' [-Wimplicit-int]
  149 | typedef int (*apply_func_args_t)(zval *pDest, int num_args, va_list args, zend_hash_key *hash_key);
      |                                                                     ^~~~
/home/gyk/lede/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend.h:189:127: warning: type defaults to 'int' in declaration of 'args' [-Wimplicit-int]
  189 |         void (*error_function)(int type, const char *error_filename, const uint32_t error_lineno, const char *format, va_list args) ZEND_ATTRIBUTE_PTR_FORMAT(printf, 4, 0);
      |                                                                                                                               ^~~~
/home/gyk/lede/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend.h:198:96: warning: type defaults to 'int' in declaration of 'ap' [-Wimplicit-int]
  198 |         void (*printf_to_smart_string_function)(smart_string *buf, const char *format, va_list ap);
      |                                                                                                ^~
/home/gyk/lede/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend.h:199:90: warning: type defaults to 'int' in declaration of 'ap' [-Wimplicit-int]
  199 |         void (*printf_to_smart_str_function)(smart_str *buf, const char *format, va_list ap);
      |                                                                                          ^~
/home/gyk/lede/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend.h:237:89: warning: type defaults to 'int' in declaration of 'ap' [-Wimplicit-int]
  237 | ZEND_API size_t zend_vspprintf(char **pbuf, size_t max_len, const char *format, va_list ap);
      |                                                                                         ^~
/home/gyk/lede/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend.h:239:84: warning: type defaults to 'int' in declaration of 'ap' [-Wimplicit-int]
  239 | ZEND_API zend_string *zend_vstrpprintf(size_t max_len, const char *format, va_list ap);
      |                                                                                    ^~
/home/gyk/lede/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend.h:283:134: warning: type defaults to 'int' in declaration of 'args' [-Wimplicit-int]
  283 | extern ZEND_API void (*zend_error_cb)(int type, const char *error_filename, const uint32_t error_lineno, const char *format, va_list args) ZEND_ATTRIBUTE_PTR_FORMAT(printf, 4, 0);
      |                                                                                                                                      ^~~~
/home/gyk/lede/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend.h:286:91: warning: type defaults to 'int' in declaration of 'ap' [-Wimplicit-int]
  286 | extern void (*zend_printf_to_smart_string)(smart_string *buf, const char *format, va_list ap);
      |                                                                                           ^~
/home/gyk/lede/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend.h:287:85: warning: type defaults to 'int' in declaration of 'ap' [-Wimplicit-int]
  287 | extern void (*zend_printf_to_smart_str)(smart_str *buf, const char *format, va_list ap);
      |                                                                                     ^~
In file included from /home/gyk/lede/build_dir/target-x86_64_musl/php-7.4.30/main/php.h:37:
/home/gyk/lede/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend_API.h:514:76: warning: type defaults to 'int' in declaration of 'argv' [-Wimplicit-int]
  514 | ZEND_API int zend_fcall_info_argv(zend_fcall_info *fci, int argc, va_list *argv);
      |                                                                            ^~~~
In file included from /home/gyk/lede/build_dir/target-x86_64_musl/php-7.4.30/main/php.h:219:
/home/gyk/lede/staging_dir/toolchain-x86_64_gcc-12.2.0_musl/include/fortify/unistd.h:43:29: error: redefinition of 'confstr'
   43 | _FORTIFY_FN(confstr) size_t confstr(int __n, char *__s, size_t __l)
      |                             ^~~~~~~
/usr/include/bits/unistd.h:240:8: note: previous definition of 'confstr' with type 'size_t(int,  char *, size_t)' {aka 'long unsigned int(int,  char *, long unsigned int)'}
  240 | __NTH (confstr (int __name, char *__buf, size_t __len))
      |        ^~~~~~~
/usr/include/sys/cdefs.h:57:62: note: in definition of macro '__NTH'
   57 | #  define __NTH(fct)    __attribute__ ((__nothrow__ __LEAF)) fct
      |                                                              ^~~
/home/gyk/lede/staging_dir/toolchain-x86_64_gcc-12.2.0_musl/include/fortify/unistd.h:53:27: error: redefinition of 'getcwd'
   53 | _FORTIFY_FN(getcwd) char *getcwd(char *__s, size_t __l)
      |                           ^~~~~~
/usr/include/bits/unistd.h:200:8: note: previous definition of 'getcwd' with type 'char *(char *, size_t)' {aka 'char *(char *, long unsigned int)'}
  200 | __NTH (getcwd (char *__buf, size_t __size))
[... 130 lines cut ...]
   47 |         va_list arg;
      |                 ^~~
/home/gyk/lede/build_dir/target-x86_64_musl/php-7.4.30/ext/sqlite3/sqlite3.c:50:18: warning: passing argument 1 of '__builtin_va_start' makes pointer from integer without a cast [-Wint-conversion]
   50 |         va_start(arg, format);
      |                  ^~~
      |                  |
      |                  int
/home/gyk/lede/build_dir/target-x86_64_musl/php-7.4.30/ext/sqlite3/sqlite3.c:50:18: note: expected '__va_list_tag *' but argument is of type 'int'
/home/gyk/lede/build_dir/target-x86_64_musl/php-7.4.30/ext/sqlite3/sqlite3.c:52:16: warning: passing argument 1 of '__builtin_va_end' makes pointer from integer without a cast [-Wint-conversion]
   52 |         va_end(arg);
      |                ^~~
      |                |
      |                int
/home/gyk/lede/build_dir/target-x86_64_musl/php-7.4.30/ext/sqlite3/sqlite3.c:52:16: note: expected '__va_list_tag *' but argument is of type 'int'
make[4]: *** [Makefile:568: ext/sqlite3/sqlite3.lo] Error 1
make[4]: Leaving directory '/home/gyk/lede/build_dir/target-x86_64_musl/php-7.4.30'
make[3]: *** [Makefile:621: /home/gyk/lede/build_dir/target-x86_64_musl/php-7.4.30/.built] Error 2
make[3]: Leaving directory '/home/gyk/lede/feeds/packages/lang/php7'
time: package/feeds/packages/php7/compile#0.38#0.19#0.66
    ERROR: package/feeds/packages/php7 failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/php7/compile] Error 1
make[2]: Leaving directory '/home/gyk/lede'
make[1]: *** [package/Makefile:110: /home/gyk/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/gyk/lede'
make: *** [/home/gyk/lede/include/toplevel.mk:231: world] Error 2

WYC-2020, 2023-01-30:
吧libiconv 编译上，然后就可以了，别问我为什么php8 不编译这个 可以编译通过，php7 configure检测不是很高级,原来是默认编译libiconv的，有一次提交吧这个改成默认不编译了

WYC-2020, 2023-01-30:
[image]
这样改下就可以了，增加依赖，同时把他提到curl前面，因为curl和sqlite这些库都会 根据iconv使用libiconv，你自己可以测试一下，我这边测试ok

Fly432, 2023-01-30:
> 吧libiconv 编译上，然后就可以了，别问我为什么php8 不编译这个 可以编译通过，php7 configure检测不是很高级,原来是默认编译libiconv的，有一次提交吧这个改成默认不编译了

WYC-2020大神，你好，R23.1.1里没有这个libiconv，但有一个libiconv-full, 如图,make menuconfig时是把它选上吧？
[image]

WYC-2020, 2023-01-31:
https://github.com/openwrt/packages/issues/14126 这个issue 已经给了答复，php8 为啥可以成功，因为他支持不依赖第三方单独编译php iconv 模块，至于功能情况视php作者而定，php7不行是在不依赖第三方iconv的情况下编译php模块不够完善导致，可能作者一开始想的就是php7 iconv模块必须在第三方支持下才能编译，到php8就可以不依赖而进行编译了，非要编译php7这样勾选
[image]
吧这个选上 @Fly432

Fly432, 2023-01-31:
清楚了！谢谢您WYC-2020大神！

WYC-2020, 2023-01-31:
[image]
这样改下吧 ，你就不用单独去哪里选择了，只要选择了iconv 他会自动依赖上，没选中就不会编译iconv @Fly432

coolsnowwolf, 2026-09-05:
旧 PHP 7.4.30 包已从当前 feeds 移除，当前提供 PHP 8.3；报告中的 musl fortify 与宿主 glibc 头文件冲突不再对应当前包，关闭为 completed。

closed 2026-09-05 by coolsnowwolf