coolsnowwolf/lede #10813: 更新源码后编译出错

mitsukileung, 2023-01-26:
ipq807x / AX3600

cc1: all warnings being treated as errors
make[6]: *** [Makefile:653: gzip.o] Error 1
make[5]: *** [Makefile:531: all-recursive] Error 1
make[4]: *** [Makefile:447: all] Error 2
make[4]: Leaving directory '/home/koysuke/lede/build_dir/target-aarch64_cortex-a53_musl/elfutils-0.188'
make[3]: *** [Makefile:117: /home/koysuke/lede/build_dir/target-aarch64_cortex-a53_musl/elfutils-0.188/.built] Error 2
make[3]: Leaving directory '/home/koysuke/lede/package/libs/elfutils'
time: package/libs/elfutils/compile#0.29#0.10#0.37
    ERROR: package/libs/elfutils failed to build.
make[2]: *** [package/Makefile:116: package/libs/elfutils/compile] Error 1
make[2]: Leaving directory '/home/koysuke/lede'
make[1]: *** [package/Makefile:110: /home/koysuke/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2

更新源码后编译出错

lovehackintosh, 2023-01-26:
日志没发全

lovehackintosh, 2023-01-26:
elfutils 的TARGET_CFLAGS需要加个判断, 如果是gcc12的话就加上`-Wno-error=use-after-free`

mitsukileung, 2023-01-26:
> elfutils 的TARGET_CFLAGS需要加个判断, 如果是gcc12的话就加上`-Wno-error=use-after-free`

加回这个可以通过了

mahakalaya, 2023-01-26:
感谢

> > elfutils 的TARGET_CFLAGS需要加个判断, 如果是gcc12的话就加上`-Wno-error=use-after-free`
> 
> 加回这个可以通过了

是的。感谢。

closed 2023-01-26 by coolsnowwolf