coolsnowwolf/lede #10814: 拉取最新源码后编译x86固件报错

bi4tms, 2023-01-26:
拉取最新源码后编译5.4/5/15/6/1内核的x86固件均报错，日志如下：
mv -f .deps/linux-core-attach.Tpo .deps/linux-core-attach.Po
x86_64-openwrt-linux-musl-gcc -D_GNU_SOURCE -DHAVE_CONFIG_H -DLOCALEDIR='"/usr/share/locale"' -I. -I..  -I. -I. -I../lib -I.. -I. -I./../libelf -I./../libebl -I./../libdw -I./../libdwelf -I./../debuginfod -I/home/x86.6.1/lede/staging_dir/toolchain-x86_64_gcc-12.2.0_musl/usr/include -I/home/x86.6.1/lede/staging_dir/toolchain-x86_64_gcc-12.2.0_musl/include/fortify -I/home/x86.6.1/lede/staging_dir/toolchain-x86_64_gcc-12.2.0_musl/include    -std=gnu99 -Wall -Wshadow -Wformat=2 -Wold-style-definition -Wstrict-prototypes -Wtrampolines -Wlogical-op -Wduplicated-cond -Wnull-dereference -Wimplicit-fallthrough=5 -Werror -Wunused -Wextra -Wstack-usage=262144   -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/x86.6.1/lede/build_dir/target-x86_64_musl/elfutils-0.188=elfutils-0.188 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   -D_GNU_SOURCE -Wno-unused-result -Wno-format-nonliteral  -MT dwfl_frame_regs.o -MD -MP -MF .deps/dwfl_frame_regs.Tpo -c -o dwfl_frame_regs.o dwfl_frame_regs.c
mv -f .deps/dwfl_frame_regs.Tpo .deps/dwfl_frame_regs.Po
x86_64-openwrt-linux-musl-gcc -D_GNU_SOURCE -DHAVE_CONFIG_H -DLOCALEDIR='"/usr/share/locale"' -I. -I..  -I. -I. -I../lib -I.. -I. -I./../libelf -I./../libebl -I./../libdw -I./../libdwelf -I./../debuginfod -I/home/x86.6.1/lede/staging_dir/toolchain-x86_64_gcc-12.2.0_musl/usr/include -I/home/x86.6.1/lede/staging_dir/toolchain-x86_64_gcc-12.2.0_musl/include/fortify -I/home/x86.6.1/lede/staging_dir/toolchain-x86_64_gcc-12.2.0_musl/include    -std=gnu99 -Wall -Wshadow -Wformat=2 -Wold-style-definition -Wstrict-prototypes -Wtrampolines -Wlogical-op -Wduplicated-cond -Wnull-dereference -Wimplicit-fallthrough=5 -Werror -Wunused -Wextra -Wstack-usage=262144   -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/x86.6.1/lede/build_dir/target-x86_64_musl/elfutils-0.188=elfutils-0.188 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   -D_GNU_SOURCE -Wno-unused-result -Wno-format-nonliteral  -MT gzip.o -MD -MP -MF .deps/gzip.Tpo -c -o gzip.o gzip.c
In function 'bigger_buffer',
    inlined from '__libdw_gunzip' at gzip.c:374:12:
gzip.c:96:9: error: pointer may be used after 'realloc' [-Werror=use-after-free]
   96 |     b = realloc (state->buffer, more -= 1024);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gzip.c:94:13: note: call to 'realloc' here
   94 |   char *b = realloc (state->buffer, more);
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[6]: *** [Makefile:648: gzip.o] Error 1
make[5]: *** [Makefile:526: all-recursive] Error 1
make[4]: *** [Makefile:442: all] Error 2
make[4]: Leaving directory '/home/x86.6.1/lede/build_dir/target-x86_64_musl/elfutils-0.188'
make[3]: *** [Makefile:117: /home/x86.6.1/lede/build_dir/target-x86_64_musl/elfutils-0.188/.built] Error 2
make[3]: Leaving directory '/home/x86.6.1/lede/package/libs/elfutils'
time: package/libs/elfutils/compile#87.79#76.00#169.13
    ERROR: package/libs/elfutils failed to build.
make[2]: *** [package/Makefile:116: package/libs/elfutils/compile] Error 1
make[2]: Leaving directory '/home/x86.6.1/lede'
make[1]: *** [package/Makefile:110: /home/x86.6.1/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/x86.6.1/lede'
make: *** [/home/x86.6.1/lede/include/toplevel.mk:231: world] Error 2

DHDAXCW, 2023-01-26:
https://github.com/coolsnowwolf/lede/commit/7df2cabf9acdab6b54626af095670b1f0b6f7ddf 把这个改回去

baodp, 2023-01-26:
我也出现一样的错误

closed 2023-01-26 by bi4tms