coolsnowwolf/lede #10816: MT6721编译k2p v2ray订阅丢弃

segadrive, 2023-01-26:
编译好时候v2ray不能订阅，v2ray内核都有，手动添加可以运行，日志：2023-01-26 22:56:20 丢弃无效节点: v2ray 节点, 英-2-Riven
2023-01-26 22:56:20 丢弃无效节点: v2ray 节点, 英-1-Olink
2023-01-26 22:56:20 丢弃无效节点: v2ray 节点, 台-1-HiNet
2023-01-26 22:56:20 丢弃无效节点: v2ray 节点, 港-18-ELITE
2023-01-26 22:56:20 丢弃无效节点: v2ray 节点, 港-17-ELITE
2023-01-26 22:56:20 丢弃无效节点: v2ray 节点, 港-16-ELITE
2023-01-26 22:56:20 丢弃无效节点: v2ray 节点, 港-15-DMIT
2023-01-26 22:56:20 丢弃无效节点: v2ray 节点, 港-14-DMIT
2023-01-26 22:56:20 丢弃无效节点: v2ray 节点, 港-13-DMIT
2023-01-26 22:56:20 丢弃无效节点: v2ray 节点, 港-12-DMIT
2023-01-26 22:56:20 丢弃无效节点: v2ray 节点, 港-11-DMIT
2023-01-26 22:56:20 丢弃无效节点: v2ray 节点, 港-10-DMIT
2023-01-26 22:56:20 丢弃无效节点: v2ray 节点, 港-9-DMIT
2023-01-26 22:56:20 丢弃无效节点: v2ray 节点, 港-8-DMIT
2023-01-26 22:56:20 丢弃无效节点: v2ray 节点, 港-7-DMIT
2023-01-26 22:56:20 丢弃无效节点: v2ray 节点, 港-6-DMIT
2023-01-26 22:56:20 丢弃无效节点: v2ray 节点, 港-5-DMIT
2023-01-26 22:56:20 丢弃无效节点: v2ray 节点, 港-4-DMIT

minlang112, 2023-01-26:
ssrplus 现已移除 alterId 配置项并强制启用 Vmess AEAD 认证
如果您发现无法连接到机场提供的 Vmess 节点，请联系客服并要求启用该认证（alterId = 0）

segadrive, 2023-01-27:
感谢

closed 2023-02-28 by aiamadeus