coolsnowwolf/lede #10817: 编译cjdns失败

Andylive5518, 2023-01-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译过程到了cjdns就报错，编译又失败

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：
/home/abc/1600/lede/build_dir/target-x86_64_musl/cjdns-cjdns-v21.1/node_build/builder.js:1050
                    throw err;
                    ^

Error: gcc -c -x cpp-output -o build_linux/net_SwitchPinger_admin_c.o -std=c99 -Wall -Wextra -Werror -Wno-pointer-sign -Wmissing-prototypes -pedantic -D linux=1 -Wno-unused-parameter -fomit-frame-pointer -D Log_DEBUG -g -D NumberCompress_TYPE=v3x5x8 -D Identity_CHECK=1 -D Allocator_USE_CANARIES=1 -D PARANOIA=1 -DHAS_ETH_INTERFACE=1 -fPIE -fdiagnostics-color=always -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/abc/1600/lede/build_dir/target-x86_64_musl/cjdns-cjdns-v21.1=cjdns-cjdns-v21.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -U_FORTIFY_SOURCE -Wno-error=array-bounds -Wno-error=stringop-overflow -Wno-error=stringop-overread -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/abc/1600/lede/build_dir/target-x86_64_musl/cjdns-cjdns-v21.1=cjdns-cjdns-v21.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -U_FORTIFY_SOURCE -Wno-error=array-bounds -Wno-error=stringop-overflow -Wno-error=stringop-overread -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/abc/1600/lede/build_dir/target-x86_64_musl/cjdns-cjdns-v21.1=cjdns-cjdns-v21.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -U_FORTIFY_SOURCE -Wno-error=array-bounds -Wno-error=stringop-overflow -Wno-error=stringop-overread -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/abc/1600/lede/build_dir/target-x86_64_musl/cjdns-cjdns-v21.1=cjdns-cjdns-v21.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -U_FORTIFY_SOURCE -Wno-error=array-bounds -Wno-error=stringop-overflow -Wno-error=stringop-overread -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/abc/1600/lede/build_dir/target-x86_64_musl/cjdns-cjdns-v21.1=cjdns-cjdns-v21.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -U_FORTIFY_SOURCE -Wno-error=array-bounds -Wno-error=stringop-overflow -Wno-error=stringop-overread -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/abc/1600/lede/build_dir/target-x86_64_musl/cjdns-cjdns-v21.1=cjdns-cjdns-v21.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -U_FORTIFY_SOURCE -Wno-error=array-bounds -Wno-error=stringop-overflow -Wno-error=stringop-overread -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/abc/1600/lede/build_dir/target-x86_64_musl/cjdns-cjdns-v21.1=cjdns-cjdns-v21.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -U_FORTIFY_SOURCE -Wno-error=array-bounds -Wno-error=stringop-overflow -Wno-error=stringop-overread -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/abc/1600/lede/build_dir/target-x86_64_musl/cjdns-cjdns-v21.1=cjdns-cjdns-v21.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -U_FORTIFY_SOURCE -Wno-error=array-bounds -Wno-error=stringop-overflow -Wno-error=stringop-overread -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/abc/1600/lede/build_dir/target-x86_64_musl/cjdns-cjdns-v21.1=cjdns-cjdns-v21.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -U_FORTIFY_SOURCE -Wno-error=array-bounds -Wno-error=stringop-overflow -Wno-error=stringop-overread -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/abc/1600/lede/build_dir/target-x86_64_musl/cjdns-cjdns-v21.1=cjdns-cjdns-v21.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -U_FORTIFY_SOURCE -Wno-error=array-bounds -Wno-error=stringop-overflow -Wno-error=stringop-overread -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/abc/1600/lede/build_dir/target-x86_64_musl/cjdns-cjdns-v21.1=cjdns-cjdns-v21.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -U_FORTIFY_SOURCE -Wno-error=array-bounds -Wno-error=stringop-overflow -Wno-error=stringop-overread -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/abc/1600/lede/build_dir/target-x86_64_musl/cjdns-cjdns-v21.1=cjdns-cjdns-v21.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -U_FORTIFY_SOURCE -Wno-error=array-bounds -Wno-error=stringop-overflow -Wno-error=stringop-overread -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/abc/1600/lede/build_dir/target-x86_64_musl/cjdns-cjdns-v21.1=cjdns-cjdns-v21.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -U_FORTIFY_SOURCE -Wno-error=array-bounds -Wno-error=stringop-overflow -Wno-error=stringop-overread -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/abc/1600/lede/build_dir/target-x86_64_musl/cjdns-cjdns-v21.1=cjdns-cjdns-v21.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -U_FORTIFY_SOURCE -Wno-error=array-bounds -Wno-error=stringop-overflow -Wno-error=stringop-overread -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/abc/1600/lede/build_dir/target-x86_64_musl/cjdns-cjdns-v21.1=cjdns-cjdns-v21.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -U_FORTIFY_SOURCE -Wno-error=array-bounds -Wno-error=stringop-overflow -Wno-error=stringop-overread -Os -D_FORTIFY_SOURCE=2 -fno-stack-protector -fstack-protector-all -Wstack-protector -D CJD_PACKAGE_VERSION="20221001-566-g0c40df89d-dirty" build_linux/net_SwitchPinger_admin_c.i

cc1: note: someone does not honour COPTS correctly, passed 15 times
net/SwitchPinger_admin.c: In function 'adminPing':
net/SwitchPinger_admin.c:103:133: error: dangling pointer 'err' to an unnamed temporary may be used [-Werror=dangling-pointer=]
  103 |         Dict d = Dict_CONST(String_CONST("error"), String_OBJ(err), NULL);
      |                                                                                                                                     ^
net/SwitchPinger_admin.c:82:26: note: unnamed temporary defined here
   82 |         err = String_CONST("path was not parsable.");
      |                          ^
cc1: all warnings being treated as errors

    at error (/home/abc/1600/lede/build_dir/target-x86_64_musl/cjdns-cjdns-v21.1/node_build/builder.js:104:15)
    at /home/abc/1600/lede/build_dir/target-x86_64_musl/cjdns-cjdns-v21.1/node_build/builder.js:185:29
    at /home/abc/1600/lede/build_dir/target-x86_64_musl/cjdns-cjdns-v21.1/node_build/builder.js:150:17
    at ChildProcess.<anonymous> (/home/abc/1600/lede/build_dir/target-x86_64_musl/cjdns-cjdns-v21.1/node_modules/saferphore/index.js:22:34)
    at ChildProcess.emit (events.js:198:13)
    at maybeClose (internal/child_process.js:982:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
make[3]: *** [Makefile:149: /home/abc/1600/lede/build_dir/target-x86_64_musl/cjdns-cjdns-v21.1/.built] Error 1
make[3]: Leaving directory '/home/abc/1600/lede/feeds/routing/cjdns'
time: package/feeds/routing/cjdns/compile#29.76#13.99#36.35
    ERROR: package/feeds/routing/cjdns failed to build.
make[2]: *** [package/Makefile:116: package/feeds/routing/cjdns/compile] Error 1
make[2]: Leaving directory '/home/abc/1600/lede'
make[1]: *** [package/Makefile:110: /home/abc/1600/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/abc/1600/lede'
make: *** [/home/abc/1600/lede/include/toplevel.mk:231：world] 错误 2

WYC-2020, 2023-01-29:
[image]
 这样修复，这是上游的仓库，只能去上游反馈修复，如果没办法 你就吧这个库移动到lede这里，然后不要带上-f就行，或者你移动到你feed 比routing之前的仓库也可以 @Andylive5518

coolsnowwolf, 2026-09-05:
已针对报告的 GCC `-Werror=dangling-pointer` 失败提交并 push：routing 仓库 `75994b736e2f`（`cjdns: avoid failing on dangling pointer warnings`），仅增加 `-Wno-error=dangling-pointer`。本地包目标构建已启动，但被既有 x86 内核包阶段缺失 `pppox.ko` 阻断，未能进入 cjdns 源码编译；该阻断与本修复无关。

closed 2026-09-05 by coolsnowwolf