coolsnowwolf/lede #10818: x86最新源碼編譯出錯

starxu1971, 2023-01-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本
X86
A：

### (3) 详细日志
make[2]: Entering directory '/Lede_source/openwrt'
make[3]: Entering directory '/Lede_source/openwrt/package/libs/libjson-c'
. /Lede_source/openwrt/include/shell.sh; /Lede_source/openwrt/staging_dir/host/bin/libdeflate-gzip -dc /Lede_source/openwrt/dl/json-c-0.16-nodoc.tar.gz | tar -C /Lede_source/openwrt/build_dir/hostpkg/json-c-0.16/.. -xf -
bash: line 1: /Lede_source/openwrt/staging_dir/host/bin/libdeflate-gzip: No such file or directory
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors
make[3]: Leaving directory '/Lede_source/openwrt/package/libs/libjson-c'
make[3]: *** [Makefile:63: /Lede_source/openwrt/build_dir/hostpkg/json-c-0.16/.prepared79ee385269811c01bc490ac1bebb7b41_6664517399ebbbc92a37c5bb081b5c53] Error 2
time: package/libs/libjson-c/host-compile#0.33#0.08#0.39
    ERROR: package/libs/libjson-c [host] failed to build.
make[2]: *** [package/Makefile:114: package/libs/libjson-c/host/compile] Error 1
make[2]: Leaving directory '/Lede_source/openwrt'
make[1]: *** [package/Makefile:110: /Lede_source/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/Lede_source/openwrt'
make: *** [/Lede_source/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.
A：

lovehackintosh, 2023-01-27:
make clean

starxu1971, 2023-01-27:
在線編譯的不存在make clean

lovehackintosh, 2023-01-27:
你是一步一步编译的吗，需要先`make tools/install -j$(nproc)`才会生成libdeflate-gzip

starxu1971, 2023-01-27:
就這裡在線編譯腳本沒有這一步

starxu1971, 2023-01-27:
https://github.com/starxu1971/build-actions 你看看

crecywang, 2023-01-28:
本地编译同样的报错提示，make clean后解决了。

closed 2023-01-29 by coolsnowwolf