coolsnowwolf/lede #10819: uqmi编译失败

codesnas, 2023-01-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：ERROR: package/network/utils/uqmi failed to build.

### (2) 路由器型号和固件版本

A：QEMU ARMv8 Virtual Machine(cortex-a53)

### (3) 详细日志

A：

```
2023-01-27T07:03:18.4516987Z -- The C compiler identification is GNU 11.3.0
2023-01-27T07:03:18.4627431Z -- Detecting C compiler ABI info
2023-01-27T07:03:18.5015518Z -- Detecting C compiler ABI info - failed
2023-01-27T07:03:18.5016516Z -- Check for working C compiler: /workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/bin/aarch64-openwrt-linux-musl-gcc
2023-01-27T07:03:18.5368455Z -- Check for working C compiler: /workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/bin/aarch64-openwrt-linux-musl-gcc - broken
2023-01-27T07:03:18.5369643Z CMake Error at /workdir/openwrt/staging_dir/host/share/cmake-3.25/Modules/CMakeTestCCompiler.cmake:70 (message):
2023-01-27T07:03:18.5370461Z   The C compiler
2023-01-27T07:03:18.5371436Z 
2023-01-27T07:03:18.5371713Z -- Configuring incomplete, errors occurred!
2023-01-27T07:03:18.5372997Z     "/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/bin/aarch64-openwrt-linux-musl-gcc"
2023-01-27T07:03:18.5373724Z See also "/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/uqmi-2020-11-22-0a19b5b7/CMakeFiles/CMakeOutput.log".
2023-01-27T07:03:18.5374665Z 
2023-01-27T07:03:18.5375094Z See also "/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/uqmi-2020-11-22-0a19b5b7/CMakeFiles/CMakeError.log".
2023-01-27T07:03:18.5375697Z   is not able to compile a simple test program.
2023-01-27T07:03:18.5375980Z 
2023-01-27T07:03:18.5376182Z   It fails with the following output:
2023-01-27T07:03:18.5376442Z 
2023-01-27T07:03:18.5376984Z     Change Dir: /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/uqmi-2020-11-22-0a19b5b7/CMakeFiles/CMakeScratch/TryCompile-wbevh3
2023-01-27T07:03:18.5377473Z     
2023-01-27T07:03:18.5377968Z     Run Build Command(s):/workdir/openwrt/staging_dir/host/bin/ninja cmTC_c9a14 && [1/2] Building C object CMakeFiles/cmTC_c9a14.dir/testCCompiler.c.o
2023-01-27T07:03:18.5378525Z     FAILED: CMakeFiles/cmTC_c9a14.dir/testCCompiler.c.o 
2023-01-27T07:03:18.5380769Z     /workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/bin/aarch64-openwrt-linux-musl-gcc   -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/uqmi-2020-11-22-0a19b5b7=uqmi-2020-11-22-0a19b5b7 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include -ffunction-sections -fdata-sections -Wno-error=dangling-pointer -Wno-error=maybe-uninitialized -o CMakeFiles/cmTC_c9a14.dir/testCCompiler.c.o -c /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/uqmi-2020-11-22-0a19b5b7/CMakeFiles/CMakeScratch/TryCompile-wbevh3/testCCompiler.c
2023-01-27T07:03:18.5382520Z     cc1: error: '-Wno-error=dangling-pointer': no option '-Wdangling-pointer'; did you mean '-Wdangling-else'?
2023-01-27T07:03:18.5383607Z     ninja: build stopped: subcommand failed.
2023-01-27T07:03:18.5383946Z     
2023-01-27T07:03:18.5384223Z     
2023-01-27T07:03:18.5384424Z 
2023-01-27T07:03:18.5384575Z   
2023-01-27T07:03:18.5384776Z 
2023-01-27T07:03:18.5385021Z   CMake will not be able to correctly generate this project.
2023-01-27T07:03:18.5385430Z Call Stack (most recent call first):
2023-01-27T07:03:18.5385785Z   CMakeLists.txt:3 (PROJECT)
2023-01-27T07:03:18.5386029Z 
2023-01-27T07:03:18.5386107Z 
2023-01-27T07:03:18.5429108Z make[3]: *** [Makefile:57: /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/uqmi-2020-11-22-0a19b5b7/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
2023-01-27T07:03:18.5429772Z make[3]: Leaving directory '/workdir/openwrt/package/network/utils/uqmi'
2023-01-27T07:03:18.5430387Z time: package/network/utils/uqmi/compile#0.44#0.12#0.54
2023-01-27T07:03:18.5438729Z     ERROR: package/network/utils/uqmi failed to build.
2023-01-27T07:03:18.5442165Z make[2]: *** [package/Makefile:116: package/network/utils/uqmi/compile] Error 1
2023-01-27T07:03:18.5460876Z make[2]: Leaving directory '/workdir/openwrt'
2023-01-27T07:03:18.5489730Z make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
2023-01-27T07:03:18.5498152Z make[1]: Leaving directory '/workdir/openwrt'
2023-01-27T07:03:18.5519969Z make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
2023-01-27T07:03:18.5970748Z ##[error]Process completed with exit code 2.
```

shuenpin, 2023-01-27:
我也遇到一樣的 issue，是不是跟近期的 gcc12 換成 gcc11 有關？

lovehackintosh, 2023-01-27:
`git revert ccc69944452c28efbffe5c847e17113e37579cba` 
不是mips用户都建议使用gcc12

mahakalaya, 2023-01-29:
更新源码后出现错误
CMake Error: Could not find CMAKE_ROOT !!!
CMake has most likely not been installed correctly.
Modules directory not found in
/home/simon/lede/staging_dir/host/share/cmake-3.25
CMake Error: Error executing cmake::LoadCache(). Aborting.

make[2]: *** [Makefile:57: /home/simon/lede/build_dir/target-aarch64_cortex-a53_musl/uqmi-2022-05-04-56cb2d40/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[2]: Leaving directory '/home/simon/lede/package/network/utils/uqmi'
time: package/network/utils/uqmi/compile#0.10#0.03#0.13
    ERROR: package/network/utils/uqmi failed to build.
make[1]: *** [package/Makefile:116: package/network/utils/uqmi/compile] Error 1
make[1]: Leaving directory '/home/simon/lede'
make: *** [/home/simon/lede/include/toplevel.mk:231：package/uqmi/compile] 错误 2

closed 2023-01-29 by coolsnowwolf

shuenpin, 2023-01-30:
我的 uqmi 也是編譯失敗

FAILED: CMakeFiles/uqmi.dir/dev.c.o
/home/sbhsu/Downloads/lede/staging_dir/toolchain-aarch64_generic_gcc-12.2.0_musl/bin/aarch64-openwrt-linux-musl-gcc -DDEBUG  -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/sbhsu/Downloads/lede/build_dir/target-aarch64_generic_musl/uqmi-2022-05-04-56cb2d40=uqmi-2022-05-04-56cb2d40 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/sbhsu/Downloads/lede/staging_dir/target-aarch64_generic_musl/usr/include -ffunction-sections -fdata-sections -DNDEBUG   -Os -ggdb -Wall -Werror --std=gnu99 -Wmissing-declarations -Wno-enum-conversion -g3 -MD -MT CMakeFiles/uqmi.dir/dev.c.o -MF CMakeFiles/uqmi.dir/dev.c.o.d -o CMakeFiles/uqmi.dir/dev.c.o -c /home/sbhsu/Downloads/lede/build_dir/target-aarch64_generic_musl/uqmi-2022-05-04-56cb2d40/dev.c
/home/sbhsu/Downloads/lede/build_dir/target-aarch64_generic_musl/uqmi-2022-05-04-56cb2d40/dev.c: In function 'qmi_request_wait':
/home/sbhsu/Downloads/lede/build_dir/target-aarch64_generic_musl/uqmi-2022-05-04-56cb2d40/dev.c:217:23: error: storing the address of local variable 'complete' in '*req.complete' [-Werror=dangling-pointer=]
  217 |         req->complete = &complete;
      |         ~~~~~~~~~~~~~~^~~~~~~~~~~
/home/sbhsu/Downloads/lede/build_dir/target-aarch64_generic_musl/uqmi-2022-05-04-56cb2d40/dev.c:208:14: note: 'complete' declared here
  208 |         bool complete = false;
      |              ^~~~~~~~
/home/sbhsu/Downloads/lede/build_dir/target-aarch64_generic_musl/uqmi-2022-05-04-56cb2d40/dev.c:208:14: note: 'req' declared here
cc1: all warnings being treated as errors
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:51: /home/sbhsu/Downloads/lede/build_dir/target-aarch64_generic_musl/uqmi-2022-05-04-56cb2d40/.built] Error 1
make[3]: Leaving directory '/home/sbhsu/Downloads/lede/package/network/utils/uqmi'
time: package/network/utils/uqmi/compile#0.41#0.25#0.67
    ERROR: package/network/utils/uqmi failed to build.
make[2]: *** [package/Makefile:116: package/network/utils/uqmi/compile] Error 1

lovehackintosh, 2023-02-07:
`wget https://github.com/coolsnowwolf/lede/pull/10825.patch && git am 10825.patch`再试试

xuexiao-weizi, 2023-02-08:
> `wget https://github.com/coolsnowwolf/lede/pull/10825.patch && git am 10825.patch`再试试

谢谢了，我去试试

hzjnet, 2024-01-21:
> git am 10825.patch

尝试了 不行
 git am 10825.patch
fatal: previous rebase directory .git/rebase-apply still exists but mbox given.

还是一样的错误