coolsnowwolf/lede #10821: 6.1.8内核惊现错误/dev/sda1: Can't open blockdev

Fyred, 2023-01-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：内核错误

### (2) 路由器型号和固件版本

A：x86_64 OpenWrt R23.1.1 / LuCI Master (git-23.025.27564-bfa4697)

### (3) 详细日志

A：
[   72.731156] squashfs: Unknown parameter 'discard'
[   72.736403] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x89e45589)
[   72.744050] F2FS-fs (sda128): Can't find valid F2FS filesystem in 1th superblock
[   72.751686] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x2e48937b)
[   72.759338] F2FS-fs (sda128): Can't find valid F2FS filesystem in 2th superblock
[   72.767101] exFAT-fs (sda128): invalid boot record signature
[   72.772825] exFAT-fs (sda128): failed to read boot sector
[   72.778310] exFAT-fs (sda128): failed to recognize exfat type
[  162.960648] /dev/sda1: Can't open blockdev
[  162.965025] /dev/sda1: Can't open blockdev
[  162.969378] /dev/sda1: Can't open blockdev
[  162.973729] squashfs: Unknown parameter 'discard'
[  162.979140] /dev/sda1: Can't open blockdev

从squashfs: Unknown parameter 'discard'到/dev/sda1: Can't open blockdev感觉都很不正常呀，之前5.15内核日志都是很清爽的

WYC-2020, 2023-01-29:
我这边都很干净哈哈，刚编译试了一下，因为我用我这边的克隆仓库编译的，但是又不完全一样所以也不知道 是哪里问题，我放一个 我这边编译的 你不嫌弃麻烦试试看，由于阿里云不能分享压缩包，所以我这边直接是创建自解压的，你下载后直接运行 解压出来 然后刷机就可以测试了
https://www.aliyundrive.com/s/EkJjVWqCpfz

yidianyipie, 2023-01-30:
> 我这边都很干净哈哈，刚编译试了一下，因为我用我这边的克隆仓库编译的，但是又不完全一样所以也不知道 是哪里问题，我放一个 我这边编译的 你不嫌弃麻烦试试看，由于阿里云不能分享压缩包，所以我这边直接是创建自解压的，你下载后直接运行 解压出来 然后刷机就可以测试了 https://www.aliyundrive.com/s/EkJjVWqCpfz

我这边也有这个问题，我记得之前有个说是把automount去掉就行了

WYC-2020, 2023-01-30:
> > 我这边都很干净哈哈，刚编译试了一下，因为我用我这边的克隆仓库编译的，但是又不完全一样所以也不知道 是哪里问题，我放一个 我这边编译的 你不嫌弃麻烦试试看，由于阿里云不能分享压缩包，所以我这边直接是创建自解压的，你下载后直接运行 解压出来 然后刷机就可以测试了 https://www.aliyundrive.com/s/EkJjVWqCpfz
> 
> 我这边也有这个问题，我记得之前有个说是把automount去掉就行了

我的没去掉

Fyred, 2023-01-30:
怎么最近automount和autosamba总是不服帖，automount就整一堆报错出来，autosamba就自动挂几个你不想的共享

WYC-2020, 2023-01-30:
> 怎么最近automount和autosamba总是不服帖，automount就整一堆报错出来，autosamba就自动挂几个你不想的共享

那就打到服位置，其实samba加一个开关就能控制，我那个固件就加了

Fyred, 2023-01-30:
> > 怎么最近automount和autosamba总是不服帖，automount就整一堆报错出来，autosamba就自动挂几个你不想的共享
> 
> 那就打到服位置，其实samba加一个开关就能控制，我那个固件就加了

嗯，谢谢啦！

closed 2023-01-30 by Fyred

sclsyin, 2024-04-18:
> > 怎么最近automount和autosamba总是不服帖，automount就整堆报错出来，autosamba就自动挂几个你不想的共享
> 
> 然后打到服务器位置，其实samba加一个开关就可以了，我那个硬件就加了

请问下这个是从哪设置的？