coolsnowwolf/lede #10823: 编译失败

chensbo, 2023-01-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
编译失败
### (1) 具体问题

A：

### (2) 路由器型号和固件版本
x86
A：

### (3) 详细日志

cc1: note: someone does not honour COPTS correctly, passed 0 times
x86_64-openwrt-linux-musl-gcc -std=c99 -Wall -Wextra -O2 -pthread -D __MUSL__ -c ipt2socks.c -o ipt2socks.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
ipt2socks.c:35:28: error: unknown type name '__off64_t'; did you mean 'off64_t'?
   35 | ssize_t splice(int __fdin, __off64_t *__offin, int __fdout,
      |                            ^~~~~~~~~
      |                            off64_t
ipt2socks.c:36:9: error: unknown type name '__off64_t'; did you mean 'off64_t'?
   36 |         __off64_t *__offout, size_t __len, unsigned int __flags) {
      |         ^~~~~~~~~
      |         off64_t
ipt2socks.c: In function 'tcp_stream_payload_forward_cb':
ipt2socks.c:759:25: warning: implicit declaration of function 'splice' [-Wimplicit-function-declaration]
  759 |         ssize_t nrecv = splice(self_watcher->fd, NULL, self_pipefd[1], NULL, TCP_SPLICE_MAXLEN, SPLICE_F_MOVE | SPLICE_F_NONBLOCK);
      |                         ^~~~~~
make[4]: *** [Makefile:28: ipt2socks.o] Error 1
make[4]: Leaving directory '/lede/build_dir/target-x86_64_musl/ipt2socks/ipt2socks-1.1.2-cfbc2189356aba7fcafb0bc961a95419f313d8a7'
make[3]: *** [Makefile:42: /lede/build_dir/target-x86_64_musl/ipt2socks/ipt2socks-1.1.2-cfbc2189356aba7fcafb0bc961a95419f313d8a7/.built] Error 2
make[3]: Leaving directory '/lede/feeds/small/ipt2socks'
time: package/feeds/small/ipt2socks/compile#2.15#0.37#3.73
    ERROR: package/feeds/small/ipt2socks failed to build.
make[2]: *** [package/Makefile:116: package/feeds/small/ipt2socks/compile] Error 1
make[2]: Leaving directory '/lede'
make[1]: *** [package/Makefile:110: /lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/lede'
make: *** [/lede/include/toplevel.mk:231: world] Error 2

WYC-2020, 2023-01-29:
我这边 5.4内核 x86  ipt2socks 1.1.3版本正常

vinwinchen, 2023-01-30:
我编译6内核也是这个问题

WYC-2020, 2023-02-01:
> 我编译6内核也是这个问题
> 
> make[1]: *** [package/Makefile:110: /home/make/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2 make[1]: Leaving directory '/home/make/lede' make: *** [/home/make/lede/include/toplevel.mk:231: world] Error 2

你这个贴了个寂寞

coolsnowwolf, 2026-09-05:
当前 `feeds/helloworld/ipt2socks` 已更新到 1.1.4，旧 issue 针对的 1.1.2 源码中的 `__off64_t` 编译错误已不再对应当前包；关闭为 completed。

closed 2026-09-05 by coolsnowwolf