coolsnowwolf/lede #10828: [建议]关于intel x550 自动协商2.5G/5G

shenlijun, 2023-01-29:
抛开x710的那价格，x550算是目前比较完善的5速网卡，aqc那几个兄弟好像默认还没带驱动吧

曾经看到有人说5.4内核可以协商，后来的就不可以，其实应该是后续新驱动的问题，和内核无关：

[这里](https://hualaishi.us/intel-x550%E7%BD%91%E5%8D%A1%E5%BC%80%E5%90%AF2-5g%E3%80%815g%E8%87%AA%E5%8A%A8%E5%8D%8F%E5%95%86/)用ethtool添加advertise的方法亲测可行：
ethtool -s <ethX> advertise 0x1800000001028

虽然通过init.d或者hotplug.d之类的方法可以解决，但总觉得如果能在编译时加入自动判断和操作的脚本会更好一些，要不每次更新固件都要再搞一次.

主要是自己了解不深，不知道哪位大佬能搞一波😁

shenlijun, 2023-01-29:
#9707

shenlijun, 2023-01-29:
顺带一说，试过很多个x710-t2l和x710-t4l，只有蓝标那种很贵的卡才能5速，其它比如dell的oem卡(价格蓝标减半)不可以2.5G/5G，尽管看上去有一样的芯片和一样的pcb

WYC-2020, 2023-01-29:
这玩意直接给驱动打一个补丁，让其自动协商呗，刚找了一个patch 不知道是否有效http://patches.dpdk.org/project/dpdk/patch/1423742468-30404-31-git-send-email-changchun.ouyang@intel.com/

coolsnowwolf, 2023-01-29:
6.1.8 内核已经支持了啊

WYC-2020, 2023-01-29:
这个问题openwrt官方都有issue，并且也给了答复，链接在这里
https://github.com/openwrt/openwrt/issues/10459
从这个issue中可以追踪到可能是linux内核这个部分的修改
https://github.com/torvalds/linux/commit/a296d665eae1e8ec6445683bfb999c884058426a
可以打个补丁还原了然后在测试一下，但是没设备不好搞
@shenlijun

shenlijun, 2023-01-30:
> 6.1.8 内核已经支持了啊

昨天中午试了下，不行的，默认协商还是1G

> 这个问题openwrt官方都有issue，并且也给了答复，链接在这里 [openwrt/openwrt#10459](https://github.com/openwrt/openwrt/issues/10459) 从这个issue中可以追踪到可能是linux内核这个部分的修改 [torvalds/linux@a296d66](https://github.com/torvalds/linux/commit/a296d665eae1e8ec6445683bfb999c884058426a) 可以打个补丁还原了然后在测试一下，但是没设备没法测试，只能你自己搞下 @shenlijun

想了想做patch的话，在内核或者驱动更新的时候总是需要维护一下；
不如直接做个包，init.d或者hotplug.d之类的地方插个脚本，判断网口型号，然后加advertise，维护成本低一些
改天我搞一个试试

WYC-2020, 2023-01-30:
> 昨天中午试了下，不行的，默认协商还是1G
>
>
> 想了想做patch的话，在内核或者驱动更新的时候总是需要维护一下； 不如直接做个包，init.d或者hotplug.d之类的地方插个脚本，判断网口型号，然后加advertise，维护成本低一些 改天我搞一个试试

我更倾向于内核，一劳永逸，并且不用在意那个无线名称

shenlijun, 2023-01-30:
> 我更倾向于内核，一劳永逸，并且不用在意那个无线名称

开机运行一次这个应该就可以，也不用专门维护

`#!/bin/sh

all_pcis=`lspci | grep -i 'eth' | grep -i 'x550' | cut -d ' ' -f 1`
all_ifs=`cat /proc/net/dev | grep -i 'eth' | cut -d :  -f 1 | sed 's/^[ \t]*//g'`

for ifname in ${all_ifs}
do
    if_pci=`ethtool -i ${ifname} | grep -i 'bus-info' | cut -b 16-`
    if [[ "$all_pcis" =~ "$if_pci" ]]
    then
        ethtool -s ${ifname} advertise 0x1800000001028
    fi
done`

shenlijun, 2023-01-31:
以前没有弄过，瞎jer复制粘贴搞了一个，暂时测试没问题

https://github.com/shenlijun/openwrt-x550-nbase-t

这样就不用再管内核和驱动的更新了

shenlijun, 2023-01-31:
有需要的话可以加到源里来 @coolsnowwolf

shenlijun, 2023-02-23:
> 顺带一说，试过很多个x710-t2l和x710-t4l，只有蓝标那种很贵的卡才能5速，其它比如dell的oem卡(价格蓝标减半)不可以2.5G/5G，尽管看上去有一样的芯片和一样的pcb

现在有人能刷固件，把dell的刷成intel的，可以5速了

closed 2023-07-26 by coolsnowwolf

wu0lei, 2023-09-04:
> > 顺带一说，试过很多个x710-t2l和x710-t4l，只有蓝标那种很贵的卡才能5速，其它比如dell的oem卡(价格蓝标减半)不可以2.5G/5G，尽管看上去有一样的芯片和一样的pcb
> 
> 现在有人能刷固件，把dell的刷成intel的，可以5速了

老哥能传授一下吗

shenlijun, 2023-09-06:
> > > 顺带一说，试过很多个x710-t2l和x710-t4l，只有蓝标那种很贵的卡才能5速，其它比如dell的oem卡(价格蓝标减半)不可以2.5G/5G，尽管看上去有一样的芯片和一样的pcb
> > 
> > 
> > 现在有人能刷固件，把dell的刷成intel的，可以5速了
> 
> 老哥能传授一下吗

咸鱼上有人刷，100块