coolsnowwolf/lede #10829: NEW3和RedmiAC2100 默认没有启动wifi

liangzhanggb, 2023-01-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
自己编译了NEW3和RedmiAC2100路由器固件，发现系统重启后，wifi总是关闭的，即使打开了wifi，重启系统后，wifi又变成关闭的了

## 2. 详细叙述

### (1) 具体问题

A：基于master分支，发现new3路由器、redmi AC2100路由器默认无线都是关闭的，即使打开了，系统重启后wifi仍然是关闭
B：基于master分支，编译固件有做过如下必要的改动
Target System (MediaTek Ralink MIPS) 
Subtarget (MT7621 based boards)
Target Profile (Newifi D2)

### (2) 路由器型号和固件版本

A：new3路由器、redmi AC2100路由器都存在重启后，默认关闭wifi
B：使用的是2023年1.29日早上的版本呢，commmitid：fec4f5df67b2794020d5e622f02ce9186f985acf

### (3) 详细日志

A：内核日志
[    0.000000] Linux version 5.15.36 (sd@sdu) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 8.4.0 r4489-ce2fb409e) 8.4.0, GNU ld (GNU Binutils) 2.34) #0 SMP Tue May 3 22:42:20 2022
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=aec3da8a-b14d-eab8-dfe9-9b0039427902 rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Enabled xstate features 0x3, context size is 576 bytes, using 'compacted' format.
[    0.000000] signal: max sigframe size: 1104
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007ffddfff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ffde000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feffc000-0x00000000feffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: QEMU Standard PC (i440FX + PIIX, 1996), BIOS rel-1.14.0-0-g155821a1990b-prebuilt.qemu.org 04/01/2014
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000000] kvm-clock: cpu 0, msr 2483001, primary cpu clock
[    0.000001] kvm-clock: using sched offset of 4751132605 cycles
[    0.000007] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000014] tsc: Detected 1996.800 MHz processor
[    0.000649] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000652] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000656] last_pfn = 0x7ffde max_arch_pfn = 0x400000000
[    0.000697] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.004225] found SMP MP-table at [mem 0x000f5a80-0x000f5a8f]
[    0.004406] ACPI: Early table checksum verification disabled
[    0.004415] ACPI: RSDP 0x00000000000F58A0 000014 (v00 BOCHS )
[    0.004422] ACPI: RSDT 0x000000007FFE1E72 000038 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.004431] ACPI: FACP 0x000000007FFE1C54 000074 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.004436] ACPI: DSDT 0x000000007FFDF040 002C14 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.004440] ACPI: FACS 0x000000007FFDF000 000040
[    0.004443] ACPI: APIC 0x000000007FFE1CC8 000080 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.004446] ACPI: SSDT 0x000000007FFE1D48 0000CA (v01 BOCHS  VMGENID  00000001 BXPC 00000001)
[    0.004448] ACPI: HPET 0x000000007FFE1E12 000038 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.004451] ACPI: WAET 0x000000007FFE1E4A 000028 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.004453] ACPI: Reserving FACP table memory at [mem 0x7ffe1c54-0x7ffe1cc7]
[    0.004455] ACPI: Reserving DSDT table memory at [mem 0x7ffdf040-0x7ffe1c53]
[    0.004456] ACPI: Reserving FACS table memory at [mem 0x7ffdf000-0x7ffdf03f]
[    0.004458] ACPI: Reserving APIC table memory at [mem 0x7ffe1cc8-0x7ffe1d47]
[    0.004459] ACPI: Reserving SSDT table memory at [mem 0x7ffe1d48-0x7ffe1e11]
[    0.004460] ACPI: Reserving HPET table memory at [mem 0x7ffe1e12-0x7ffe1e49]
[    0.004461] ACPI: Reserving WAET table memory at [mem 0x7ffe1e4a-0x7ffe1e71]
[    0.004498] Zone ranges:
[    0.004500]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[... 532 lines cut ...]
[    7.106266] PPP generic driver version 2.4.2
[    7.108255] PPP MPPE Compression module registered
[    7.110026] NET: Registered PF_PPPOX protocol family
[    7.112162] wireguard: WireGuard 1.0.0 loaded. See www.wireguard.com for information.
[    7.113345] wireguard: Copyright (C) 2015-2019 Jason A. Donenfeld <Jason@zx2c4.com>. All Rights Reserved.
[    7.117130] kmodloader: done loading kernel modules from /etc/modules.d/*
[    9.674987] random: crng init done
[   10.252608] 8021q: adding VLAN 0 to HW filter on device eth0
[   10.255115] br-lan: port 1(eth0) entered blocking state
[   10.256169] br-lan: port 1(eth0) entered disabled state
[   10.257236] device eth0 entered promiscuous mode
[   10.259586] br-lan: port 1(eth0) entered blocking state
[   10.260473] br-lan: port 1(eth0) entered forwarding state
[   10.263915] 8021q: adding VLAN 0 to HW filter on device eth1
[   11.271207] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   36.628275] squashfs: Unknown parameter 'discard'
[   36.630532] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0xd44d89e4)
[   36.631587] F2FS-fs (sda128): Can't find valid F2FS filesystem in 1th superblock
[   36.633422] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x80a345bd)
[   36.634441] F2FS-fs (sda128): Can't find valid F2FS filesystem in 2th superblock
[   36.637107] exFAT-fs (sda128): invalid boot record signature
[   36.638046] exFAT-fs (sda128): failed to read boot sector
[   36.638910] exFAT-fs (sda128): failed to recognize exfat type
[   66.723242] squashfs: Unknown parameter 'discard'
[   66.725086] /dev/sda1: Can't open blockdev

WYC-2020, 2023-01-29:
你稍微搜索一下就知道有类似的issue ，并且有答复
https://github.com/coolsnowwolf/lede/issues/10662

Cathgao, 2023-01-29:
日经问题，提之前能不能搜索下已有issue？真当贴吧了
lan接口增加自定义接口ra0rai0

liangzhanggb, 2023-01-29:
@WYC-2020 @Cathgao 
多谢二位，按照#10662 脚本方法可以，还有个小的疑问，请问这些为什么LEDE默认不开启相关feature？

补充：尝试搜索了，搜索的方法不对，抱歉了。

closed 2023-01-29 by liangzhanggb

miaoermua, 2023-02-19:
应该添加个 `/sbin/mtkwifi up` 的计划任务或者本地启动脚本也可以

Cathgao, 2023-02-27:
> 

代码的问题，wifi的两个接口RA0、RAI0没有默认添加给LAN，所以开机自启LAN口的时候没有自动把这两个口打开。至于为什么会这样，可能是初始化脚本有问题