coolsnowwolf/lede #10830: 编译报错,大佬帮忙看一下 谢谢

lmxslpc, 2023-01-29:
路由器型号和固件版本
x86_64 平台
云编译

### (3) 详细日志
2023-01-29T10:00:44.1604317Z sed -i "s/Installed-Time: .*/Installed-Time: 1674966004/" /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/root-x86/usr/lib/opkg/status
2023-01-29T10:00:44.2181672Z rm -rf /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/root-x86/boot /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/root-x86/tmp/* /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/root-x86/usr/lib/opkg/info/*.postinst* /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/root-x86/usr/lib/opkg/lists/* /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/root-x86/var/lock/*.lock
2023-01-29T10:00:44.2342707Z find /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/root-x86/ -mindepth 1 -execdir touch -hcd "@1674966004" "{}" +
2023-01-29T10:00:45.0077260Z make[2]: Leaving directory '/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt'
2023-01-29T10:00:45.1877153Z make[2]: Entering directory '/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt'
2023-01-29T10:00:47.4404167Z make[3]: Entering directory '/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/target/linux'
2023-01-29T10:00:47.4548815Z make[4]: Entering directory '/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/target/linux/x86'
2023-01-29T10:00:47.5325867Z rm -f /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.90/localversion
2023-01-29T10:00:47.5349140Z /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/scripts/kconfig.pl  + + /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/target/linux/generic/config-5.15 /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/target/linux/x86/config-5.15 /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/target/linux/x86/64/config-5.15 > /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.90/.config.target
2023-01-29T10:00:47.5849496Z awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/.config >> /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.90/.config.target
2023-01-29T10:00:47.5922867Z echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.90/.config.target
2023-01-29T10:00:47.5940870Z echo "# CONFIG_KALLSYMS_ALL is not set" >> /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.90/.config.target
2023-01-29T10:00:47.5957185Z echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.90/.config.target
2023-01-29T10:00:47.5976506Z /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/scripts/package-metadata.pl kconfig /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/tmp/.packageinfo /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/.config 5.15 > /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.90/.config.override
2023-01-29T10:00:47.6935452Z /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/scripts/kconfig.pl 'm+' '+' /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.90/.config.target /dev/null /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.90/.config.override > /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.90/.config.set
2023-01-29T10:00:47.7492349Z mv /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.90/.config.set /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.90/.config.old
2023-01-29T10:00:47.7532613Z grep -v INITRAMFS /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.90/.config.old > /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.90/.config.set
2023-01-29T10:00:47.7572935Z echo 'CONFIG_INITRAMFS_SOURCE=""' >> /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.90/.config.set
2023-01-29T10:00:47.7590663Z echo '# CONFIG_INITRAMFS_FORCE is not set' >> /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.90/.config.set
2023-01-29T10:00:47.7608709Z rm -rf /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/modules
2023-01-29T10:00:47.7633391Z cmp -s /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.90/.config.set /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.90/.config.prev || { cp /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.90/.config.set /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.90/.config; cp /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.90/.config.set /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.90/.config.prev; }
2023-01-29T10:00:47.7677745Z export MAKEFLAGS= ; [ -d /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.90/user_headers ] || make -C /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.90 KCFLAGS="-fmacro-prefix-map=/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sun Jan 29 04:20:04 2023" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.15.90 CC="x86_64-openwrt-linux-musl-gcc"  INSTALL_HDR_PATH=/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.90/user_headers headers_install
2023-01-29T10:00:47.7694159Z grep '=[ym]' /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.90/.config.set | LC_ALL=C sort | /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/staging_dir/host/bin/mkhash md5 > /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.90/.vermagic
2023-01-29T10:00:47.7735757Z touch /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.90/.configured
2023-01-29T10:00:47.7758232Z rm -f /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/root-x86/init
2023-01-29T10:00:47.7783313Z make -C /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.90 KCFLAGS="-fmacro-prefix-map=/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPI [... cut ...]
2023-01-29T10:04:38.4726052Z make: *** [/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/include/toplevel.mk:231: world] Error 2
2023-01-29T10:04:38.5149854Z ##[error]Process completed with exit code 2.

完整日志:
https://pipelines.actions.githubusercontent.com/serviceHosts/43ff6c3d-dc0d-4a41-9f60-02687d34b530/_apis/pipelines/1/runs/13/signedlogcontent/2?urlExpires=2023-01-29T11%3A01%3A49.7891857Z&urlSigningMethod=HMACV1&urlSignature=oiXk5bDe8cEoNR8SHRdQ7k%2F3%2FVZIAh%2BT2bElkMCFf0I%3D

DHDAXCW, 2023-01-29:
rootfs空间调大点

closed 2023-01-29 by coolsnowwolf

lmxslpc, 2023-01-29:
> rootfs空间调大点

谢大佬 我试试