coolsnowwolf/lede #10834: v2ray-plugin编译报错 大佬帮看一下 谢谢

lmxslpc, 2023-01-30:
x86_64 5.4内核

**Building targets
github.com/v2fly/v2ray-core/v5/common/protocol/quic
github.com/v2fly/v2ray-core/v5/common/protocol/quic

../../../../../dl/go-mod-cache/github.com/v2fly/v2ray-core/v5@v5.2.0/common/protocol/quic/sniff.go:39:18: undefined: CipherSuiteTLS13
../../../../../dl/go-mod-cache/github.com/v2fly/v2ray-core/v5@v5.2.0/common/protocol/quic/sniff.go:155:12: undefined: AEADAESGCMTLS13
note: module requires Go 1.19
golang.org/x/exp/constraints
golang.org/x/exp/constraints

../../../../../dl/go-mod-cache/golang.org/x/[exp@v0.0.0-20220722155223-a9213eeb770e](mailto:exp@v0.0.0-20220722155223-a9213eeb770e)/constraints/constraints.go:13:2: syntax error: unexpected ~, expecting method or interface name
../../../../../dl/go-mod-cache/golang.org/x/[exp@v0.0.0-20220722155223-a9213eeb770e](mailto:exp@v0.0.0-20220722155223-a9213eeb770e)/constraints/constraints.go:20:2: syntax error: unexpected ~, expecting method or interface name
../../../../../dl/go-mod-cache/golang.org/x/[exp@v0.0.0-20220722155223-a9213eeb770e](mailto:exp@v0.0.0-20220722155223-a9213eeb770e)/constraints/constraints.go:27:9: syntax error: unexpected |, expecting semicolon or newline or }
../../../../../dl/go-mod-cache/golang.org/x/[exp@v0.0.0-20220722155223-a9213eeb770e](mailto:exp@v0.0.0-20220722155223-a9213eeb770e)/constraints/constraints.go:34:2: syntax error: unexpected ~, expecting method or interface name
../../../../../dl/go-mod-cache/golang.org/x/[exp@v0.0.0-20220722155223-a9213eeb770e](mailto:exp@v0.0.0-20220722155223-a9213eeb770e)/constraints/constraints.go:41:2: syntax error: unexpected ~, expecting method or interface name
../../../../../dl/go-mod-cache/golang.org/x/[exp@v0.0.0-20220722155223-a9213eeb770e](mailto:exp@v0.0.0-20220722155223-a9213eeb770e)/constraints/constraints.go:49:10: syntax error: unexpected |, expecting semicolon or newline or }
note: module requires Go 1.18
github.com/lucas-clemente/quic-go/internal/qtls
github.com/lucas-clemente/quic-go/internal/utils/linkedlist
github.com/lucas-clemente/quic-go/internal/utils/linkedlist

../../../../../dl/go-mod-cache/github.com/lucas-clemente/quic-go@v0.31.1/internal/utils/linkedlist/linkedlist.go:15:16: syntax error: unexpected any, expecting ]
../../../../../dl/go-mod-cache/github.com/lucas-clemente/quic-go@v0.31.1/internal/utils/linkedlist/linkedlist.go:31:17: syntax error: unexpected [, expecting comma or )
../../../../../dl/go-mod-cache/github.com/lucas-clemente/quic-go@v0.31.1/internal/utils/linkedlist/linkedlist.go:39:17: syntax error: unexpected [, expecting comma or )
../../../../../dl/go-mod-cache/github.com/lucas-clemente/quic-go@v0.31.1/internal/utils/linkedlist/linkedlist.go:48:13: syntax error: unexpected any, expecting ]
../../../../../dl/go-mod-cache/github.com/lucas-clemente/quic-go@v0.31.1/internal/utils/linkedlist/linkedlist.go:54:14: syntax error: unexpected [, expecting comma or )
../../../../../dl/go-mod-cache/github.com/lucas-clemente/quic-go@v0.31.1/internal/utils/linkedlist/linkedlist.go:62:9: syntax error: unexpected [, expecting (
../../../../../dl/go-mod-cache/github.com/lucas-clemente/quic-go@v0.31.1/internal/utils/linkedlist/linkedlist.go:66:14: syntax error: unexpected [, expecting comma or )
../../../../../dl/go-mod-cache/github.com/lucas-clemente/quic-go@v0.31.1/internal/utils/linkedlist/linkedlist.go:69:14: syntax error: unexpected [, expecting comma or )
../../../../../dl/go-mod-cache/github.com/lucas-clemente/quic-go@v0.31.1/internal/utils/linkedlist/linkedlist.go:77:14: syntax error: unexpected [, expecting comma or )
../../../../../dl/go-mod-cache/github.com/lucas-clemente/quic-go@v0.31.1/internal/utils/linkedlist/linkedlist.go:85:14: syntax error: unexpected [, expecting comma or )
../../../../../dl/go-mod-cache/github.com/lucas-clemente/quic-go@v0.31.1/internal/utils/linkedlist/linkedlist.go:85:14: too many errors
note: module requires Go 1.18
github.com/lucas-clemente/quic-go/internal/qtls

../../../../../dl/go-mod-cache/github.com/lucas-clemente/quic-go@v0.31.1/internal/qtls/go_oldversion.go:5:5: cannot use "The version of quic-go you're using can't be built using outdated Go... (type untyped string) as type int in assignment
note: module requires Go 1.18

make[3]: *** [Makefile:43: /home/linpc/istoreos/build_dir/target-x86_64_musl/v2ray-plugin-5.2.0/.built] Error 2
make[3]: Leaving directory '/home/linpc/istoreos/feeds/helloworld/v2ray-plugin'
time: package/feeds/helloworld/v2ray-plugin/compile#0.89#1.12#2.50
ERROR: package/feeds/helloworld/v2ray-plugin failed to build.
make[2]: *** [package/Makefile:116: package/feeds/helloworld/v2ray-plugin/compile] Error 1
make[2]: Leaving directory '/home/linpc/istoreos'
make[1]: *** [package/Makefile:110: /home/linpc/istoreos/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/linpc/istoreos'
make: *** [/home/linpc/istoreos/include/toplevel.mk:234：world] 错误 2**

yanxin152133, 2023-01-30:
https://github.com/xiaorouji/openwrt-passwall/issues/2091
试试这个

lmxslpc, 2023-01-30:
> [xiaorouji/openwrt-passwall#2091](https://github.com/xiaorouji/openwrt-passwall/issues/2091) 试试这个

好的 谢谢大佬 我试试

coolsnowwolf, 2026-09-05:
当前 v2ray-plugin 已更新至 5.49.0，并使用仓库 Go 构建框架；issue 中针对旧 v2ray-core 5.2.0/过旧 Go 编译器的错误已被版本更新覆盖，关闭为 completed。

closed 2026-09-05 by coolsnowwolf