coolsnowwolf/lede #10838: x86云编译错误

Jonekadsen, 2023-01-31:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
云编译一直失败，改了好久还是失败

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译error2错误

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：40960+0 records in
2023-01-30T18:52:32.8040494Z 40960+0 records out
2023-01-30T18:52:32.8065004Z 20971520 bytes (21 MB, 20 MiB) copied, 0.114687 s, 183 MB/s
2023-01-30T18:52:32.8067128Z + rm -f /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.kernel
2023-01-30T18:52:32.8171923Z rm -fR /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.grub2
2023-01-30T18:52:32.8249431Z install -d -m0755 /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.grub2
2023-01-30T18:52:32.8332328Z cp -fpR /workdir/openwrt/staging_dir/target-x86_64_musl/image/grub2/boot.img /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.grub2/
2023-01-30T18:52:32.8411387Z cp -fpR /workdir/openwrt/staging_dir/target-x86_64_musl/image/grub2/gpt-core.img /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.grub2/core.img
2023-01-30T18:52:32.8500121Z echo '(hd0) /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk' > /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.grub2/device.map
2023-01-30T18:52:32.8563398Z /workdir/openwrt/staging_dir/host/bin/grub-bios-setup -m "/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.grub2/device.map" -d "/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.grub2" -r "hd0,gpt1" /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk
2023-01-30T18:52:32.8704245Z /workdir/openwrt/staging_dir/host/bin/grub-bios-setup: warning: Your BIOS Boot Partition is under 1 MiB, please increase its size..
2023-01-30T18:52:32.9153722Z 40960+0 records in
2023-01-30T18:52:32.9154057Z 40960+0 records out
2023-01-30T18:52:32.9154458Z 20971520 bytes (21 MB, 20 MiB) copied, 0.14043 s, 149 MB/s
2023-01-30T18:52:32.9157402Z + rm -f /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.kernel
2023-01-30T18:52:32.9186880Z rm -fR /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.grub2
2023-01-30T18:52:32.9217165Z install -d -m0755 /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.grub2
2023-01-30T18:52:32.9248019Z cp -fpR /workdir/openwrt/staging_dir/target-x86_64_musl/image/grub2/boot.img /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.grub2/
2023-01-30T18:52:32.9279838Z cp -fpR /workdir/openwrt/staging_dir/target-x86_64_musl/image/grub2/gpt-core.img /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.grub2/core.img
2023-01-30T18:52:32.9313187Z echo '(hd0) /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz' > /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.grub2/device.map
2023-01-30T18:52:32.9333027Z /workdir/openwrt/staging_dir/host/bin/grub-bios-setup -m "/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.grub2/device.map" -d "/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.grub2" -r "hd0,gpt1" /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz
2023-01-30T18:52:32.9377666Z /workdir/openwrt/staging_dir/host/bin/grub-bios-setup: warning: Your BIOS Boot Partition is under 1 MiB, please increase its size..
2023-01-30T18:52:33.0342138Z if command -v qemu-img; then qemu-img convert -f raw -O vmdk /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.new; mv /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.new /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk; else echo "WARNING: Install qemu-img to create VDI/VMDK images" >&2; exit 1; fi
2023-01-30T18:52:33.0363714Z WARNING: Install qemu-img to create VDI/VMDK images
2023-01-30T18:52:33.0367836Z make[5]: *** [Makefile:164: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk] Error 1
2023-01-30T18:52:33.0368435Z make[5]: *** Waiting for unfinished jobs....
2023-01-30T18:52:33.0858881Z /workdir/openwrt/staging_dir/host/bin/gzip -f -9n -c /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz  > /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.new
2023-01-30T18:52:38.6526966Z sha256sum "/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz" | cut -d" " -f1 > "/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.sha256sum"
2023-01-30T18:52:39.9436425Z [ ! -s "/workdir/openwrt/key-build" -o ! -s "/workdir/openwrt/key-build.ucert" -o ! -s "/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz" ] || { cp "/workdir/openwrt/key-build.ucert" "/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.ucert" ; usign -S -m "/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz" -s "/workdir/openwrt/key-build" -x "/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.sig" ; ucert -A -c "/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.ucert" -x "/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.sig" ; fwtool -S "/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.ucert" "/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz" ; }
2023-01-30T18:52:41.4385597Z make[5]: Leaving directory '/workdir/openwrt/target/linux/x86/image'
2023-01-30T18:52:41.4396555Z make[4]: *** [Makefile:27: install] Error 2
2023-01-30T18:52:41.4397138Z make[4]: Leaving directory '/workdir/openwrt/target/linux/x86'
2023-01-30T18:52:41.4397977Z make[3]: *** [Makefile:11: install] Error 2
2023-01-30T18:52:41.4398607Z make[3]: Leaving directory '/workdir/openwrt/target/linux'
2023-01-30T18:52:41.4398946Z time: target/linux/install#243.92#8.31#134.39
2023-01-30T18:52:41.4401529Z     ERROR: target/linux failed to build.
2023-01-30T18:52:41.4434899Z make[2]: *** [target/Makefile:30: target/linux/install] Error 1
2023-01-30T18:52:41.4435703Z make[2]: Leaving directory '/workdir/openwrt'
2023-01-30T18:52:41.4444016Z make[1]: *** [target/Makefile:24: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.target_install] Error 2
2023-01-30T18:52:42.2488396Z make[1]: Leaving directory '/workdir/openwrt'
2023-01-30T18:52:42.2605748Z make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
2023-01-30T18:52:42.2816879Z ##[error]Process completed with exit code 2.

[完整日志](https://github.com/Jonekadsen/OpenWrt/suites/10662409502/logs?attempt=1)

WYC-2020, 2023-02-01:
明显不是bios 那个问题，这个问题 几万年了 l没修复而已，改下对齐就可以修复这玩意，不知道从哪里搬运的云编译脚本
[image]
@Jonekadsen

Jonekadsen, 2023-02-01:
> 明显不是bios 那个问题，这个问题 几万年了 l没修复而已，改下对齐就可以修复这玩意，不知道从哪里搬运的云编译脚本 [image] @Jonekadsen

P3的云编译呀，之前都没事的，仅半年才这样

Jonekadsen, 2023-02-01:
> 明显不是bios 那个问题，这个问题 几万年了 l没修复而已，改下对齐就可以修复这玩意，不知道从哪里搬运的云编译脚本 [image] @Jonekadsen

我是怎么操作对齐？config里开启qemu-img吗？
刚刚ubuntu上编译了，我看config里也没开启这个啊，也编译成功了，搜issues里有说编不出来vmdk的，我看我bin/targets里也编译出vmdk了

同样的config，actions就是编不出，不是太懂，还请赐教：)

[编译脚本](https://github.com/Jonekadsen/OpenWrt)

WYC-2020, 2023-02-02:
ubunt 安装qemu-img 啊

WYC-2020, 2023-02-02:
我仓库的的依赖 命令，拿去直接跑下 

sudo apt-get update

sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3 python2.7 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib antlr3 gperf wget curl swig rsync aria2 ca-certificates python3-pyelftools python-pyelftools yasm

closed 2023-06-22 by Jonekadsen