coolsnowwolf/lede #10843: 为啥又把6.1改为testing kernel了？？？

GreatMichaelLee, 2023-02-01:
commit fb0c3a7fdefbbb1c517840b4b875fccd88c48563

Zaphkito, 2023-02-01:
有点难绷

iKira, 2023-02-02:
可能是N5105这些11代U在虚拟机上运行6.1崩得厉害。反正N5105用ESXi虚拟机运行6.1内核版本的话必崩，几乎1天1崩（软重启），换5.15内核好一点，两三天崩一次。实在受不了换回勾4125了，稳得一批。

GreatMichaelLee, 2023-02-02:
> 可能是N5105这些11代U在虚拟机上运行6.1崩得厉害。反正N5105用ESXi虚拟机运行6.1内核版本的话必崩，几乎1天1崩（软重启），换5.15内核好一点，两三天崩一次。实在受不了换回勾4125了，稳得一批。

我的从来不绷啊？？啥情况呢？这个还是三天前升级了一下重启的，最长的我跑了28天(掉电重启了)
[image]

GreatMichaelLee, 2023-02-02:
改回去也没关系，就是make menuconfig改成testing kernel各取所需的事儿，但建议不要把这种改动藏到一个不相关的提交里，我是持续自动build和自动upgrade的，突然一下某天发现内核降到5.15了还查了半天。单独做个commit会更友好一点。

zzy19868, 2023-02-03:
我的机器上，6.1内核 passwall 速度很慢，换5.15后正常

Pacio, 2023-02-03:
不清楚，我这边6.1内核已经稳了很久了

kenwestern, 2023-02-08:
我的J6413，6.1下firewall经常保存卡死，i226网卡平均几小时软重启，换回5.15正常了，无法理解

HollyMadCow, 2023-02-08:
最近几个版本的编译出来后，N5105物理机都在迅雷下载并且内网大流量的情况下会重启，以前怎么整都不会，迅雷开启下载，有个20M以上下载速度，内网260M/S往NAS传输文件，很容易OPENWRT就重启了

closed 2023-02-08 by aiamadeus

KevinMX, 2023-02-18:
> > 可能是N5105这些11代U在虚拟机上运行6.1崩得厉害。反正N5105用ESXi虚拟机运行6.1内核版本的话必崩，几乎1天1崩（软重启），换5.15内核好一点，两三天崩一次。实在受不了换回勾4125了，稳得一批。
> 
> 我的从来不绷啊？？啥情况呢？这个还是三天前升级了一下重启的，最长的我跑了28天(掉电重启了) [image]

因为你是 ESXi。这玩意不是 Linux 内核的。直接跑 Linux 就不一定了。