coolsnowwolf/lede #10844: 同步最新源码后出错

mitsukileung, 2023-02-01:
ipq807x / Ax6 / 5.15

Applying ./patches/0010-QSDK-config-Avoid-Werror-heroics.patch using plaintext: 
patching file config
Hunk #1 FAILED at 132.
1 out of 1 hunk FAILED -- saving rejects to file config.rej
Patch failed!  Please fix ./patches/0010-QSDK-config-Avoid-Werror-heroics.patch!
make[3]: *** [Makefile:131: /home/koysuke/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-nohnat/qca-ssdk-2022-02-09-0d6d4106/.prepared_4671c9ded4029a28ced9302fc619fd6c_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/koysuke/lede/package/qca/qca-ssdk'
time: package/qca/qca-ssdk/nohnat/compile#0.12#0.19#0.26
    ERROR: package/qca/qca-ssdk failed to build (build variant: nohnat).
make[2]: *** [package/Makefile:116: package/qca/qca-ssdk/compile] Error 1
make[2]: Leaving directory '/home/koysuke/lede'
make[1]: *** [package/Makefile:110: /home/koysuke/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/koysuke/lede'
make: *** [/home/koysuke/lede/include/toplevel.mk:231：world] 错误 2

mitsukileung, 2023-02-02:
大佬，同步PR可以跳过之前的错误但还是编译失败

cc1: fatal error: /home/koysuke/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.91/arch/arm64/boot/dts/qcom/ipq8070-tl-er2260t.dts: No such file or directory
compilation terminated.
make[5]: *** [Makefile:18: /home/koysuke/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/image-ipq8070-tl-er2260t.dtb] Error 1
make[5]: Leaving directory '/home/koysuke/lede/target/linux/ipq807x/image'
make[4]: *** [Makefile:26: install] Error 2
make[4]: Leaving directory '/home/koysuke/lede/target/linux/ipq807x'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/home/koysuke/lede/target/linux'
time: target/linux/install#26.42#21.78#44.61
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory '/home/koysuke/lede'
make[1]: *** [target/Makefile:24: /home/koysuke/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/koysuke/lede'
make: *** [/home/koysuke/lede/include/toplevel.mk:231：world] 错误 2

closed 2023-02-02 by coolsnowwolf

mitsukileung, 2023-02-02:
5.15内核编译还是出错，5.10内核编译成功

1054009064, 2023-02-02:
5.15 不好用，我沒有編譯過，等 Lean 吧。

mitsukileung, 2023-02-02:
> 5.15 不好用，我沒有編譯過，等 Lean 吧。

好的，谢谢大佬

PR #10845 (2023-02-02): qca-ssdk: refresh patches
Fixed patch application failure.

Fixes: 89200af ("bump qca-ssdk nss-dp and ssdk-shell, Add tplink-tl-er2260t basic support (#10777)")
Fixes: #10844

Q：你知道这是`pull request`吗？(使用 "x" 选择)
* [x] 我知道
commit f6d0acb9 (2023-02-02): qca-ssdk: refresh patches (#10845)

Fixed patch application failure.

Fixes: 89200af ("bump qca-ssdk nss-dp and ssdk-shell, Add tplink-tl-er2260t basic support (#10777)")
Fixes: #10844