coolsnowwolf/lede #10847: rockchip 编译 qmi_wwan_q-3.0失败

zhouzhiwen2000, 2023-02-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：rockchip 编译 qmi_wwan_q-3.0失败,  implicit declaration of function 'netif_rx_ni'.

### (2) 路由器型号和固件版本

A：photonicat

### (3) 详细日志

A：/home/ubuntu/lede_photonicat/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/qmi_wwan_q-3.0/qmi_wwan_q.c:322:13: error: implicit declaration of function 'netif_rx_ni'; did you mean 'netif_rx'? [-Werror=implicit-function-declaration]
  322 |             netif_rx_ni(reply);
      |             ^~~~~~~~~~~
      |             netif_rx

应该和这个类似
https://github.com/coolsnowwolf/lede/issues/9803

coolsnowwolf, 2023-02-02:
6.1 内核么？

zhouzhiwen2000, 2023-02-02:
> 6.1 内核么？

是的

shenlijun, 2023-02-02:
兄弟是怎么做emmc固件的？

zhouzhiwen2000, 2023-02-02:
package/wwan/driver/quectel_cm_5G/Makefile里的路径（22）行还有问题。

ntbowen, 2023-02-05:
> > 6.1 内核么？
> 
> 是的

我很奇怪我的6.1内核就是无法引导，换成5.15.91，同样的配置编译出来的固件就能引导

zhouzhiwen2000, 2023-02-05:
> > > 6.1 内核么？
> > 
> > 
> > 是的
> 
> 我很奇怪我的6.1内核就是无法引导，换成5.15.91，同样的配置编译出来的固件就能引导

我6.1内核引导正常啊。你是不是编译时加了啥驱动?

ntbowen, 2023-02-05:
> > > > 6.1 内核么？
> > > 
> > > 
> > > 是的
> > 
> > 
> > 我很奇怪我的6.1内核就是无法引导，换成5.15.91，同样的配置编译出来的固件就能引导
> 
> 我6.1内核引导正常啊。你是不是编译时加了啥驱动?

是的，增加了qca6390、wcn6855、mt7921、mt7922的驱动，我手上有这个模块，想在光影猫上试试能不能正常工作。编译都没什么问题，就是无法引导。

zhouzhiwen2000, 2023-02-05:
> > > > > 6.1 内核么？
> > > > 
> > > > 
> > > > 是的
> > > 
> > > 
> > > 我很奇怪我的6.1内核就是无法引导，换成5.15.91，同样的配置编译出来的固件就能引导
> > 
> > 
> > 我6.1内核引导正常啊。你是不是编译时加了啥驱动?
> 
> 是的，增加了qca6390、wcn6855、mt7921、mt7922的驱动，我手上有这个模块，想在光影猫上试试能不能正常工作。编译都没什么问题，就是无法引导。

接TTL看一下是不是内核panic了。

closed 2023-03-01 by zhouzhiwen2000