coolsnowwolf/lede #10849: 拉取最新源码后编译报错

bi4tms, 2023-02-02:
拉取最新源码后编译报错，日志如下：
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/x86.5.4/lede/tmp/go-build" GO_MOD_CACHE_DIR="/home/x86.5.4/lede/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/xtls/xray-core" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/xtls/xray-core/main" GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/x86.5.4/lede/build_dir/target-x86_64_musl/Xray-core-1.7.3" GO_BUILD_DIR="/home/x86.5.4/lede/build_dir/target-x86_64_musl/Xray-core-1.7.3/.go_work/build" GO_BUILD_BIN_DIR="/home/x86.5.4/lede/build_dir/target-x86_64_musl/Xray-core-1.7.3/.go_work/build/bin" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/x86.5.4/lede/staging_dir/target-x86_64_musl/usr/share/gocode/src" GOOS="linux" GOARCH="amd64" GO386="" GOAMD64="v1" GOARM="" GOMIPS="" GOMIPS64="" GOPPC64="" CGO_ENABLED=1 CC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/x86.5.4/lede/build_dir/target-x86_64_musl/Xray-core-1.7.3=Xray-core-1.7.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/x86.5.4/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/x86.5.4/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/x86.5.4/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/x86.5.4/lede/build_dir/target-x86_64_musl/Xray-core-1.7.3=Xray-core-1.7.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/x86.5.4/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/x86.5.4/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro" GOPATH="/home/x86.5.4/lede/build_dir/target-x86_64_musl/Xray-core-1.7.3/.go_work/build" GOCACHE="/home/x86.5.4/lede/tmp/go-build" GOMODCACHE="/home/x86.5.4/lede/dl/go-mod-cache" GOENV=off /usr/bin/env bash /home/x86.5.4/lede/feeds/packages/lang/golang//golang-build.sh build -v -buildvcs=false -trimpath -ldflags "all=-buildid '1675338619' -linkmode external -extldflags '-L/home/x86.5.4/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/x86.5.4/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -Wl,-z,now -Wl,-z,relro'"        -ldflags " -X github.com/xtls/xray-core/core.build=OpenWrt -X github.com/xtls/xray-core/core.version=1.7.3 -buildid '1675338619' -linkmode external -extldflags '-L/home/x86.5.4/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/x86.5.4/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -Wl,-z,now -Wl,-z,relro'"
Finding targets

Building targets
../../../../../dl/go-mod-cache/github.com/quic-go/qtls-go1-20@v0.1.0/handshake_client.go:11:2: package crypto/ecdh is not in GOROOT (/home/x86.5.4/lede/staging_dir/hostpkg/lib/go-cross/src/crypto/ecdh)

make[3]: *** [Makefile:90: /home/x86.5.4/lede/build_dir/target-x86_64_musl/Xray-core-1.7.3/.built] Error 1
make[3]: Leaving directory '/home/x86.5.4/lede/feeds/passwall_packages/xray-core'
time: package/feeds/passwall_packages/xray-core/compile#1.03#6.26#6.83
    ERROR: package/feeds/passwall_packages/xray-core failed to build.
make[2]: *** [package/Makefile:116: package/feeds/passwall_packages/xray-core/compile] Error 1
make[2]: Leaving directory '/home/x86.5.4/lede'
make[1]: *** [package/Makefile:110: /home/x86.5.4/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/x86.5.4/lede'
make: *** [/home/x86.5.4/lede/include/toplevel.mk:231: world] Error 2

若需改至passwall项目下提issue烦请告知

xinaz, 2023-02-03:
碰到了同样的问题
Building targets
../../../../../dl/go-mod-cache/github.com/quic-go/qtls-go1-20@v0.1.0/handshake_client.go:11:2: package crypto/ecdh is not in GOROOT (/home/xinaz/project/lede/staging_dir/hostpkg/lib/go-cross/src/crypto/ecdh)

make[3]: *** [Makefile:52: /home/xinaz/project/lede/build_dir/target-arm_cortex-a9_musl_eabi/Xray-core-1.7.3/.built] Error 1
make[3]: Leaving directory '/home/xinaz/project/lede/feeds/small/xray-core'
time: package/feeds/small/xray-core/compile#1.54#0.54#0.60
    ERROR: package/feeds/small/xray-core failed to build.
make[2]: *** [package/Makefile:116: package/feeds/small/xray-core/compile] Error 1
make[2]: Leaving directory '/home/xinaz/project/lede'
make[1]: *** [package/Makefile:110: /home/xinaz/project/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/xinaz/project/lede'
make: *** [/home/xinaz/project/lede/include/toplevel.mk:231: world] Error 2

xinaz, 2023-02-03:
make clean后重编译，问题不能复现了

Geelens, 2023-02-03:
同样问题

bootli, 2023-02-03:
同样问题

EternalAries, 2023-02-04:
安装golang 1.2就好了

hzjnet, 2023-02-08:
> 安装golang 1.2就好了

请教在github关掉svn后如何获得golang 1.20版本openwrt的源码？

xinaz, 2023-02-09:
下载1.9x，手工加进去后编译成功。

hzjnet, 2023-06-26:
> 下载1.9x，手工加进去后编译成功。

谢谢 ，不过不是想要的目标