coolsnowwolf/lede #10851: 建议把官方源码的bootstrap主题适配过来

ecrasy, 2023-02-03:
建议把官方源码的bootstrap主题适配过来

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：L大源码的bootstrap主题太简陋了  
对比官方源码master的bootstrap主题，  
L大源码的bootstrap主题遗漏了很多信息显示和设置页面
如果不选默认主题，就只能使用argon主题了  
bootstrap作为默认主题，是否应该提供和argon主题差不多的信息显示？

### (2) 路由器型号和固件版本

A：No

### (3) 详细日志

A：No

WYC-2020, 2023-02-03:
想太多，l肯定是不会适配这个，专心搞功能性的他，看别人有人适配没 ，反正感觉现在主题非常多了。没那么纠结

ecrasy, 2023-02-03:
这不是纠结的问题
这是一个默认主题
应该要提供相应的功能
否则应该把默认去掉
使用其它所谓"很多主题"中的一个来作为默认

kkkevin07, 2023-02-03:
可以通过修改配置来实现
```
cd ./feeds/luci/themes
rm -rf luci-theme-argon  
git clone -b 18.06 https://github.com/jerrykuku/luci-theme-argon.git  
cd ../../../
vim ./feeds/luci/collections/luci/Makefile
```
修改bootstrap为argon即可修改默认选择的主题为argon主题

minlang112, 2023-02-03:
自己fork一下，参照楼上改一下配置

lxl6125, 2023-02-04:
信息和设置界面关主题什么事😅

ufk119, 2023-02-07:
[ecrasy](https://github.com/ecrasy)是肯定知道怎么适配默认主题的。他的意思是，既然你提供bootstrap作为默认主题，就应该提供完整的功能，否则就不要作为默认主题。

closed 2023-02-08 by aiamadeus