coolsnowwolf/lede #10852: miniupnpd 一直报错

uosX, 2023-02-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
miniupnpd 一直报错
Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
没有
## 2. 详细叙述
### (1) 具体问题
未知

### (2) 路由器型号和固件版本
倍控G31 X86

### (3) 详细日志
Fri Feb  3 09:16:02 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:02 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:02 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:02 2023 daemon.err miniupnpd[27649]: addnatrule() : chain MINIUPNPD not found
Fri Feb  3 09:16:02 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:02 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:02 2023 daemon.err miniupnpd[27649]: addnatrule() : chain MINIUPNPD not found
Fri Feb  3 09:16:02 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:02 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:02 2023 daemon.err miniupnpd[27649]: addnatrule() : chain MINIUPNPD not found
Fri Feb  3 09:16:03 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:03 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:03 2023 daemon.err miniupnpd[27649]: addnatrule() : chain MINIUPNPD not found
Fri Feb  3 09:16:03 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:03 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:03 2023 daemon.err miniupnpd[27649]: addnatrule() : chain MINIUPNPD not found
Fri Feb  3 09:16:03 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:03 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:03 2023 daemon.err miniupnpd[27649]: addnatrule() : chain MINIUPNPD not found
Fri Feb  3 09:16:03 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:03 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:03 2023 daemon.err miniupnpd[27649]: addnatrule() : chain MINIUPNPD not found
Fri Feb  3 09:16:15 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:15 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:15 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:15 2023 daemon.err miniupnpd[27649]: addnatrule() : chain MINIUPNPD not found
Fri Feb  3 09:16:15 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:15 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:15 2023 daemon.err miniupnpd[27649]: addnatrule() : chain MINIUPNPD not found
Fri Feb  3 09:16:15 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:15 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:15 2023 daemon.err miniupnpd[27649]: addnatrule() : chain MINIUPNPD not found
Fri Feb  3 09:16:15 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:15 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:15 2023 daemon.err miniupnpd[27649]: addnatrule() : chain MINIUPNPD not found
Fri Feb  3 09:16:15 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:15 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:15 2023 daemon.err miniupnpd[27649]: addnatrule() : chain MINIUPNPD not found
Fri Feb  3 09:16:15 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:15 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:15 2023 daemon.err miniupnpd[27649]: addnatrule() : chain MINIUPNPD not found
Fri Feb  3 09:16:15 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:15 2023 daemon.err miniupnpd[27649]: chain MINIUPNPD not found
Fri Feb  3 09:16:15 2023 daemon.err miniupnpd[27649]: addnatrule() : chain MINIUPNPD not found

WYC-2020, 2023-02-03:
防火墙缺少miniupnpd ，init里面有加入
[image]
 自己重启看下还有没，如果不行就看下 是否打印了upnp daemon" "external interface not found, not starting

uosX, 2023-02-06:
Mon Feb  6 16:06:04 2023 daemon.err miniupnpd[32347]: addmasqueraderule() : chain MINIUPNPD-POSTROUTING not found
Mon Feb  6 16:06:04 2023 daemon.notice miniupnpd[32347]: add_redirect_rule2(): addmasqueraderule returned -1
Mon Feb  6 16:06:04 2023 daemon.err miniupnpd[32347]: addmasqueraderule() : chain MINIUPNPD-POSTROUTING not found
Mon Feb  6 16:06:04 2023 daemon.notice miniupnpd[32347]: add_redirect_rule2(): addmasqueraderule returned -1
Mon Feb  6 16:06:04 2023 daemon.err miniupnpd[32347]: addmasqueraderule() : chain MINIUPNPD-POSTROUTING not found
Mon Feb  6 16:06:04 2023 daemon.notice miniupnpd[32347]: add_redirect_rule2(): addmasqueraderule returned -1

uosX, 2023-02-06:
同样有这个问题，
Mon Feb  6 16:06:04 2023 daemon.err miniupnpd[32347]: addmasqueraderule() : chain MINIUPNPD-POSTROUTING not found
Mon Feb  6 16:06:04 2023 daemon.notice miniupnpd[32347]: add_redirect_rule2(): addmasqueraderule returned -1
Mon Feb  6 16:06:04 2023 daemon.err miniupnpd[32347]: addmasqueraderule() : chain MINIUPNPD-POSTROUTING not found
Mon Feb  6 16:06:04 2023 daemon.notice miniupnpd[32347]: add_redirect_rule2(): addmasqueraderule returned -1
Mon Feb  6 16:06:04 2023 daemon.err miniupnpd[32347]: addmasqueraderule() : chain MINIUPNPD-POSTROUTING not found
Mon Feb  6 16:06:04 2023 daemon.notice miniupnpd[32347]: add_redirect_rule2(): addmasqueraderule returned -1

suisuyy, 2023-07-31:
有公网ip吗，upnp是吧内网端口映射到公网的，好像现在upnp会检测是否公网才能运行，因为没公网跑起来也没用

closed 2026-09-05 by coolsnowwolf