coolsnowwolf/lede #10854: Docker守护进程无法启动

kkkevin07, 2023-02-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
自编译添加 luci-app-dockerman ，openwrt首次启动docker正常，重启openwrt后无法启动
A：

### (2) 路由器型号和固件版本
型号：X86_64
固件版本：OpenWrt R23.2.14 / LuCI Master (git-23.025.27564-bfa4697)
A：

### (3) 详细日志
重启后系统异常日志，docker 相关的系统日志：
`Fri Feb  3 22:41:11 2023 daemon.info samba4-server: io_uring module found, enabling VFS io_uring. (also needs Kernel 5.4+ Support)
Fri Feb  3 22:41:11 2023 daemon.err nmbd[3258]: [2023/02/03 22:41:11.995059,  0] ../../lib/util/become_daemon.c:135(daemon_ready)
Fri Feb  3 22:41:11 2023 daemon.err nmbd[3258]:   daemon_ready: daemon 'nmbd' finished starting up and ready to serve connections
Fri Feb  3 22:41:11 2023 daemon.err smbd[3257]: [2023/02/03 22:41:11.999809,  0] ../../source3/passdb/secrets.c:66(secrets_init_path)
Fri Feb  3 22:41:11 2023 daemon.err smbd[3257]:   Failed to open /etc/samba/secrets.tdb
Fri Feb  3 22:41:11 2023 daemon.err smbd[3257]: [2023/02/03 22:41:11.999927,  0] ../../source3/passdb/secrets.c:66(secrets_init_path)
Fri Feb  3 22:41:11 2023 daemon.err smbd[3257]:   Failed to open /etc/samba/secrets.tdb
Fri Feb  3 22:41:11 2023 daemon.err smbd[3257]: [2023/02/03 22:41:11.999960,  0] ../../lib/util/become_daemon.c:121(exit_daemon)
Fri Feb  3 22:41:11 2023 daemon.err smbd[3257]:   exit_daemon: daemon failed to start: smbd can not open secrets.tdb, error code 13
Fri Feb  3 22:41:12 2023 daemon.notice procd: /etc/rc.d/S99autocore: uci: I/O error
Fri Feb  3 22:41:12 2023 user.notice dockerd-init: Bridge device 'docker0' already defined in network config
Fri Feb  3 22:41:12 2023 daemon.err dockerd[3494]: chmod /opt/docker: read-only file system
Fri Feb  3 22:41:12 2023 authpriv.warn dropbear[3593]: Failed listening on '22': Error listening: Address not available
Fri Feb  3 22:41:12 2023 authpriv.info dropbear[3593]: Not backgrounding
Fri Feb  3 22:41:12 2023 user.notice dnsmasq: found already running DHCP-server on interface 'br-lan' refusing to start, use 'option force 1' to override
Fri Feb  3 22:41:12 2023 daemon.info avahi-daemon[2168]: Server startup complete. Host name is OpenWrt.local. Local service cookie is 932091712.
Fri Feb  3 22:41:17 2023 daemon.err smbd[3759]: [2023/02/03 22:41:17.017432,  0] ../../source3/passdb/secrets.c:66(secrets_init_path)
Fri Feb  3 22:41:17 2023 daemon.err smbd[3759]:   Failed to open /etc/samba/secrets.tdb
Fri Feb  3 22:41:17 2023 daemon.err smbd[3759]: [2023/02/03 22:41:17.017560,  0] ../../source3/passdb/secrets.c:66(secrets_init_path)
Fri Feb  3 22:41:17 2023 daemon.err smbd[3759]:   Failed to open /etc/samba/secrets.tdb
Fri Feb  3 22:41:17 2023 daemon.err smbd[3759]: [2023/02/03 22:41:17.017590,  0] ../../lib/util/become_daemon.c:121(exit_daemon)
Fri Feb  3 22:41:17 2023 daemon.err smbd[3759]:   exit_daemon: daemon failed to start: smbd can not open secrets.tdb, error code 13
Fri Feb  3 22:41:17 2023 daemon.notice ttyd[3864]: [2023/02/03 22:41:17:1307] N: ttyd 1.6.3 (libwebsockets 4.3.2-20221001-584-g1daa71c00)
Fri Feb  3 22:41:17 2023 daemon.notice ttyd[3864]: [2023/02/03 22:41:17:1308] N: tty configuration:
Fri Feb  3 22:41:17 2023 daemon.notice ttyd[3864]: [2023/02/03 22:41:17:1308] N:   start command: /bin/login
Fri Feb  3 22:41:17 2023 daemon.notice ttyd[3864]: [2023/02/03 22:41:17:1308] N:   close signal: SIGHUP (1)
Fri Feb  3 22:41:17 2023 daemon.notice ttyd[3864]: [2023/02/03 22:41:17:1308] N:   terminal type: xterm-256color
Fri Feb  3 22:41:17 2023 daemon.notice ttyd[3864]: [2023/02/03 22:41:17:1312] N:    /usr/lib/libwebsockets-evlib_uv.so
Fri Feb  3 22:41:17 2023 daemon.notice ttyd[3864]: [2023/02/03 22:41:17:1312] N: lws_create_context: LWS: 4.3.2-20221001-584-g1daa71c00, NET CLI SRV H1 H2 WS ConMon IPV6-off
Fri Feb  3 22:41:17 2023 daemon.notice ttyd[3864]: [2023/02/03 22:41:17:1312] N: elops_init_pt_uv:  Using foreign event loop...
Fri Feb  3 22:41:17 2023 daemon.notice ttyd[3864]: [2023/02/03 22:41:17:1313] N: __lws_lc_tag:  ++ [wsi|0|pipe] (1)
Fri Feb  3 22:41:17 2023 daemon.notice ttyd[3864]: [2023/02/03 22:41:17:1313] N: __lws_lc_tag:  ++ [vh|0|netlink] (1)
Fri Feb  3 22:41:17 2023 daemon.notice ttyd[3864]: [2023/02/03 22:41:17:1313] N: __lws_lc_tag:  ++ [vh|1|default|br-lan|br-lan|7681] (2)
Fri Feb  3 22:41:17 2023 daemon.notice ttyd[3864]: [2023/02/03 22:41:17:1314] N: [vh|1|default|br-lan|br-lan|7681]: lws_socket_bind: source ads 192.168.1.112
Fri Feb  3 22:41:17 2023 daemon.notice ttyd[3864]: [2023/02/03 22:41:17:1315] N: __lws_lc_tag:  ++ [wsi|1|listen|default|br-lan|7681] (2)
Fri Feb  3 22:41:17 2023 daemon.notice ttyd[3864]: [2023/02/03 22:41:17:1315] N:  Listening on port: 7681
Fri Feb  3 22:41:22 2023 daemon.err smbd[4189]: [2023/02/03 22:41:22.034478,  0] ../../source3/passdb/secrets.c:66(secrets_init_path)
Fri Feb  3 22:41:22 2023 daemon.err smbd[4189]:   Failed to open /etc/samba/secrets.tdb
Fri Feb  3 22:41:22 2023 daemon.err smbd[4189]: [2023/02/03 22:41:22.034606,  0] ../../source3/passdb/secrets.c:66(secrets_init_path)
Fri Feb  3 22:41:22 2023 daemon.err smbd[4189]:   Failed to open /etc/samba/secrets.tdb
Fri Feb  3 22:41:22 2023 daemon.err smbd[4189]: [2023/02/03 22:41:22.034620,  0] ../../lib/util/become_daemon.c:121(exit_daemon)
Fri Feb  3 22:41:22 2023 daemon.err smbd[4189]:   exit_daemon: daemon failed to start: smbd can not open secrets.tdb, error code 13
Fri Feb  3 22:41:22 2023 daemon.info procd: - init complete -
Fri Feb  3 22:41:22 2023 daemon.err dockerd[4295]: chmod /opt/docker: read-only file system
Fri Feb  3 22:41:27 2023 daemon.err smbd[4369]: [2023/02/03 22:41:27.052471,  0] ../../source3/passdb/secrets.c:66(secrets_init_path)
Fri Feb  3 22:41:27 2023 daemon.err smbd[4369]:   Failed to open /etc/samba/secrets.tdb
Fri Feb  3 22:41:27 2023 daemon.err smbd[4369]: [2023/02/03 22:41:27.052571,  0] ../../source3/passdb/secrets.c:66(secrets_init_path)
Fri Feb  3 22:41:27 2023 daemon.err smbd[4369]:   Failed to open /etc/samba/secrets.tdb
Fri Feb  3 22:41:27 2023 daemon.err smbd[4369]: [2023/02/03 22:41:27.052585,  0] ../../lib/util/become_daemon.c:121(exit_daemon)
Fri Feb  3 22:41:27 2023 daemon.err smbd[4369]:   exit_daemon: daemon failed to start: smbd can not open secrets.tdb, error code 13
Fri Feb  3 22:41:32 2023 daemon.err smbd[4524]: [2023/02/03 22:41:32.067240,  0] ../../source3/passdb/secrets.c:66(secrets_init_path)
Fri Feb  3 22:41:32 2023 daemon.err smbd[4524]:   Failed to open /etc/samba/secrets.tdb
Fri Feb  3 22:41:32 2023 daemon.err smbd[4524]: [2023/02/03 22:41:32.067338,  0] ../../source3/passdb/secrets.c:66(secrets_init_path)
[... 15 lines cut ...]
Fri Feb  3 22:41:37 2023 daemon.err smbd[4612]:   Failed to open /etc/samba/secrets.tdb
Fri Feb  3 22:41:37 2023 daemon.err smbd[4612]: [2023/02/03 22:41:37.085854,  0] ../../lib/util/become_daemon.c:121(exit_daemon)
Fri Feb  3 22:41:37 2023 daemon.err smbd[4612]:   exit_daemon: daemon failed to start: smbd can not open secrets.tdb, error code 13
Fri Feb  3 22:41:37 2023 daemon.info procd: Instance samba4::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
Fri Feb  3 22:41:39 2023 daemon.info samba4-server: io_uring module found, enabling VFS io_uring. (also needs Kernel 5.4+ Support)
Fri Feb  3 22:41:39 2023 daemon.err smbd[4752]: [2023/02/03 22:41:39.512702,  0] ../../source3/passdb/secrets.c:66(secrets_init_path)
Fri Feb  3 22:41:39 2023 daemon.err smbd[4752]:   Failed to open /etc/samba/secrets.tdb
Fri Feb  3 22:41:39 2023 daemon.err smbd[4752]: [2023/02/03 22:41:39.512803,  0] ../../source3/passdb/secrets.c:66(secrets_init_path)
Fri Feb  3 22:41:39 2023 daemon.err smbd[4752]:   Failed to open /etc/samba/secrets.tdb
Fri Feb  3 22:41:39 2023 daemon.err smbd[4752]: [2023/02/03 22:41:39.512832,  0] ../../lib/util/become_daemon.c:121(exit_daemon)
Fri Feb  3 22:41:39 2023 daemon.err smbd[4752]:   exit_daemon: daemon failed to start: smbd can not open secrets.tdb, error code 13
Fri Feb  3 22:41:39 2023 daemon.info procd: Instance samba4::instance1 s in a crash loop 7 crashes, 0 seconds since last crash
Fri Feb  3 22:41:39 2023 kern.err kernel: [   35.297032] squashfs: Unknown parameter 'discard'
Fri Feb  3 22:41:39 2023 kern.info kernel: [   35.298788] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x89e45589)
Fri Feb  3 22:41:39 2023 kern.err kernel: [   35.301033] F2FS-fs (sda128): Can't find valid F2FS filesystem in 1th superblock
Fri Feb  3 22:41:39 2023 kern.info kernel: [   35.303508] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x2e48937b)
[... cut ...]
Fri Feb  3 22:41:39 2023 kern.err kernel: [   35.308432] exFAT-fs (sda128): invalid boot record signature
Fri Feb  3 22:41:39 2023 kern.err kernel: [   35.310175] exFAT-fs (sda128): failed to read boot sector
Fri Feb  3 22:41:39 2023 kern.err kernel: [   35.311983] exFAT-fs (sda128): failed to recognize exfat type
Fri Feb  3 22:41:48 2023 kern.info kernel: [   44.706332] hv_balloon: Max. dynamic memory size: 1048576 MB
Fri Feb  3 22:42:22 2023 daemon.err dockerd[6321]: chmod /opt/docker: read-only file system
Fri Feb  3 22:50:09 2023 daemon.err uhttpd[1963]: sh: write error: Broken pipe
`
A：

zhangguanzhang, 2023-02-15:
不是报错 sda 没容量了吗

Maxswang, 2023-11-28:
@kkkevin07  openwrt 重启后文件系统变成 read-only， docker守护进程无法写磁盘，从而无法完成启动

closed 2025-08-21 by kkkevin07