coolsnowwolf/lede #10856: SSRP启动NaiveProxy节点一直重启

firstep, 2023-02-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：自行编译的固件，SSRP选择naiveproxy协议的节点是无法启动，日志一直重启，请大佬指示

### (2) 路由器型号和固件版本

A：X86，固件版本R23.2.14

### (3) 详细日志

A：
Fri Feb  3 16:28:20 2023 kern.info kernel: [ 1157.880148] traps: naive[16447] trap int3 ip:563d25a22515 sp:7ffebab402e0 error:0 in naive[563d2564a000+4a3000]
Fri Feb  3 16:28:20 2023 kern.info kernel: [ 1157.881846] traps: naive[16449] trap int3 ip:562318b75515 sp:7ffd3af79740 error:0 in naive[56231879d000+4a3000]
Fri Feb  3 16:28:52 2023 user.notice shadowsocksr: ssrplus redir tcp error.restart!
Fri Feb  3 16:28:56 2023 kern.info kernel: [ 1193.369809] traps: naive[17475] trap int3 ip:55b308ff1515 sp:7fff9944ede0 error:0 in naive[55b308c19000+4a3000]
Fri Feb  3 16:28:56 2023 kern.info kernel: [ 1193.370362] traps: naive[17477] trap int3 ip:557ca4fb7515 sp:7ffd2722f4e0 error:0 in naive[557ca4bdf000+4a3000]

littleghost2016, 2023-04-12:
X86，固件版本R23.4.1，同样的问题
```
Wed Apr 12 15:00:39 2023 kern.info kernel: [  121.048885] traps: naive[8805] trap int3 ip:5643a64f6935 sp:7ffffb2419a0 error:0 in naive[5643a6102000+4c5000]
Wed Apr 12 15:00:39 2023 kern.info kernel: [  121.049768] traps: naive[8807] trap int3 ip:55c051dc5935 sp:7ffd9b1f4950 error:0 in naive[55c0519d1000+4c5000]
```

coolsnowwolf, 2026-09-05:
日志是旧版 naiveproxy 二进制运行时触发 `SIGTRAP`，没有 CPU 特性、naiveproxy 版本、当前包来源或可复现配置；当前无法归因于 LEDE 源码或形成通用修复。

closed 2026-09-05 by coolsnowwolf