coolsnowwolf/lede #10857: Golang版本偏低

bi4tms, 2023-02-03:
最新的Xray-core的编译需要Go1.20以上的编译环境，而feeds中的golang版本低于1.20，致反复编译无法通过。

可否百忙之中抽空升级一下golang版本，不胜感激。

@coolsnowwolf

guhill1, 2023-02-03:
+1, 我也遇到相同问题，麻烦更新一下吧，急需

WYC-2020, 2023-02-04:
等我消息

0xDion, 2023-02-04:
同问题，Xray-core编译报错

bi4tms, 2023-02-04:
> 等我消息

坐等大佬公布调整方法

xiaotan8, 2023-02-04:
好像镜像都没发布1.20版。

WYC-2020, 2023-02-04:
试过了 升级倒是可以升，并且用1.20的确实能编译通过，但是其他很多用go的都会编译错误，因为1.20 还没普及，pw那边 有提交xray patch 建议暂时可以用那个方式搞，等大部分用go的包都升级到1.20 在升级go 要不然得不偿失，因为一个包导致其他包都的打patches

WYC-2020, 2023-02-07:
[fix_go_120.tar.gz](https://github.com/coolsnowwolf/lede/files/10671245/fix_go_120.tar.gz)
用这个直接覆盖 就可以了 @bi4tms

guoguo163, 2023-02-08:
> [fix_go_120.tar.gz](https://github.com/coolsnowwolf/lede/files/10671245/fix_go_120.tar.gz) 用这个直接覆盖就可以了@bi4tms

[image]
我这个是因为没删掉010-compatible-with-go-1.18+.patch 吗

WYC-2020, 2023-02-08:
> > [fix_go_120.tar.gz](https://github.com/coolsnowwolf/lede/files/10671245/fix_go_120.tar.gz) 用这个直接覆盖就可以了@bi4tms
> 
> [image] 我这个是因为没删掉010-compatible-with-go-1.18+.patch 吗

是的 最好是吧原来的都删除 然后用我这个 直接放过去，当然你也可以 用那个patch 不用我这个，那个patch 修复的比较全 包含很多版本，我没修复那么多

jmmes, 2023-02-09:
> > > [fix_go_120.tar.gz](https://github.com/coolsnowwolf/lede/files/10671245/fix_go_120.tar.gz) 用这个直接覆盖就可以了@bi4tms
> > 
> > 
> > [image] 我这个是因为没删掉010-compatible-with-go-1.18+.patch 吗
> 
> 是的 最好是吧原来的都删除 然后用我这个 直接放过去，当然你也可以 用那个patch 不用我这个，那个patch 修复的比较全 包含很多版本，我没修复那么多

请问patch在哪下呢？同遇到类似问题了！

closed 2023-02-23 by bi4tms