coolsnowwolf/lede #10858: package/lean/default-settings compile 之后无法编译成固件

deplives, 2023-02-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

本地编译或云编译概率性在 package/lean/default-settings compile  之后无法继续（或者说后续进度很慢很慢）
慢到云编译超过 6 小时被取消，慢到本地编译这里我最长等过 13 个小时还没结束，受不了手动给停了。

上一个我提的 issue 因为没有详细日志被关闭，我这里贴上详细日志
[image]
[image]

从10:05:31 time: package/lean/default-settings/compile#0.26#0.06#0.40 结束后，一直到最后 action 任务超过 6 小时被取消，中间有超过两个小时时间，一直有日志，但是速度貌似很慢很慢

本地编译也是卡在这里，最长的时候

这个是完整日志
https://pipelines.actions.githubusercontent.com/serviceHosts/4db607d6-b813-4b14-8494-4c0825a1c121/_apis/pipelines/1/runs/679/signedlogcontent/2?urlExpires=2023-02-03T12%3A50%3A19.8410584Z&urlSigningMethod=HMACV1&urlSignature=m0sl%2Bt%2FsJemEXdhLqq5mQ3bguJ7ADyBdEdF2IcxO5mI%3D

zheshifandian, 2023-02-04:
云编译别带node，编译node太费时间

deplives, 2023-02-05:
> 云编译别带node，编译node太费时间

本地编译也卡在这里啊，卡了 13 个小时还没结束

tanst, 2023-02-19:
> 云编译别带node，编译node太费时间

大佬能详细说一下怎么不带node吗？对编译有影响吗？

closed 2023-02-24 by aiamadeus