coolsnowwolf/lede #10860: R22.10.1无法正常安装dockerd 20.10.22-1

huangwb8, 2023-02-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

通过opkg install无法正确地安装dockerd (20.10.22-1)。原因是linux内核与dockerd (20.10.22-1)不兼容，其最低的Linux要求为5.15.90-1。希望有大佬提供一个下载旧版本dockerd/docker/docker-compose的ipk包的方法，因为我找了好久都没找到 (ฅ´ω`ฅ)

### (2) 路由器型号和固件版本

我目前使用的固件版本是：openwrt-x64-R22.10.1-CW-X86-P1-squashfs-combined-efi.img.gz
linux内核是： Linux OpenWrt 5.15.68 #0 SMP Tue Sep 20 12:20:53 2022 x86_64 GNU/Linux

### (3) 详细日志

```
...
* satisfy_dependencies_for: Cannot satisfy the following dependencies for dockerd:
 *      kernel (= 5.15.90-1-c76eae9a0375f342f15112dd27ce89a1)
 * opkg_install_cmd: Cannot install package dockerd.
```

firkerword, 2023-02-06:
刷新版固件不就得了

huangwb8, 2023-02-06:
@firkerword 我用的是这个固件，似乎已经是最新版的了？

firkerword, 2023-02-06:
> @firkerword 我用的是这个固件（ https://github.com/coolsnowwolf/lede/releases/tag/20221001 ），似乎已经是最新版的了？

最新的是230214，我是自己编译的，你也可以自己尝试编译或者去恩山找个合适的

huangwb8, 2023-02-06:
@firkerword 谢谢，那我去了解一下！我还以为这个仓库的release已经是最新的了  (ฅ´ω`ฅ)

closed 2023-02-06 by huangwb8

Gsam333, 2023-02-23:
opkg install docker是可以安装成功的，docker -v:Docker version 20.10.22, build 3a2c30b.但docker是没有启动，运行/etc/init.d/docker-ash: /etc/init.d/docker: not found。运行其他docker命令都是提示Cannot connect to the Docker daemon at unix:///var/run/docker.sock. Is the docker daemon running?

huangwb8, 2023-02-23:
@Gsam333 

我个人对该项目的理解是，其提供了构建OpenWrt社区的最重要基础；但不一定是最好用的发行版。比如，在这个固件（ https://github.com/coolsnowwolf/lede/releases/tag/20221001 ）中，由于内核版本问题无法正常安装dockerd，你只安装docker是不行的。

我的解决方案是不用这个版本。我直接使用的是这个项目，即利用在线编译定制专属固件： https://github.com/kiddin9/OpenWrt_x86-r2s-r4s-r5s-N1 。

仅供参考，希望有帮助哈！

Gsam333, 2023-02-23:
@huangwb8 基于对lede大佬膜拜，又有的CW-X86-P1特别优化固件，特别我正好cw-x86-p1设备，docker是能正常安装，能查到docker版本号，但不晓得为何无法启动？

huangwb8, 2023-02-23:
> @huangwb8 基于对lede大佬膜拜，又有的CW-X86-P1特别优化固件，特别我正好cw-x86-p1设备，docker是能正常安装，能查到docker版本号，但不晓得为何无法启动？

因为还需要有这一步： `opkg install dockerd`。但在R22.10.1固件版本中是没办法成功安装的，因为它内核版本太低。

Gsam333, 2023-02-23:
@huangwb8 
opkg install dockerd
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (libpthread and libpthread) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (librt and librt) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (libpthread and libpthread) providing same name marked HOLD or PREFER. Using latest.
Unknown package 'dockerd'.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-d282af7706ae22b64a072a63c482db12) for kmod-crypto-hash
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-d282af7706ae22b64a072a63c482db12) for kmod-crypto-crc32c
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-d282af7706ae22b64a072a63c482db12) for kmod-lib-crc32c
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-d282af7706ae22b64a072a63c482db12) for kmod-crypto-acompress
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-d282af7706ae22b64a072a63c482db12) for kmod-lib-lzo
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-d282af7706ae22b64a072a63c482db12) for kmod-lib-zlib-inflate
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-d282af7706ae22b64a072a63c482db12) for kmod-lib-zlib-deflate
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-d282af7706ae22b64a072a63c482db12) for kmod-lib-raid6
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-lib-raid6 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-d282af7706ae22b64a072a63c482db12) for kmod-lib-xor
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-lib-xor found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-d282af7706ae22b64a072a63c482db12) for kmod-lib-zstd
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-lib-zstd found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-d282af7706ae22b64a072a63c482db12) for kmod-fs-btrfs
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-fs-btrfs found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-d282af7706ae22b64a072a63c482db12) for kmod-nf-reject
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-d282af7706ae22b64a072a63c482db12) for kmod-nf-ipt
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-d282af7706ae22b64a072a63c482db12) for kmod-nf-log
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-d282af7706ae22b64a072a63c482db12) for kmod-ipt-core
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-d282af7706ae22b64a072a63c482db12) for kmod-nfnetlink
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-d282af7706ae22b64a072a63c482db12) for kmod-nf-reject6
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nf-reject6 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-d282af7706ae22b64a072a63c482db12) for kmod-nf-conntrack
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-d282af7706ae22b64a072a63c482db12) for kmod-nf-conntrack6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-d282af7706ae22b64a072a63c482db12) for kmod-nf-nat
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-d282af7706ae22b64a072a63c482db12) for kmod-nf-log6
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nf-log6 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-d282af7706ae22b64a072a63c482db12) for kmod-nft-core
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nft-core found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-d282af7706ae22b64a072a63c482db12) for kmod-nft-compat
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nft-compat found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency libnftnl11 for xtables-nft
 * pkg_hash_fetch_best_installation_candidate: Packages for xtables-nft found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-d282af7706ae22b64a072a63c482db12) for kmod-ipt-extra
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-d282af7706ae22b64a072a63c482db12) for kmod-nf-ipt6
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nf-ipt6 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-d282af7706ae22b64a072a63c482db12) for kmod-ip6tables
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ip6tables found, but incompatible with the architectures configured
[... cut ...]
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-d282af7706ae22b64a072a63c482db12) for kmod-veth
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-veth found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency uci-firewall for dockerd
 * pkg_hash_fetch_best_installation_candidate: Packages for dockerd found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package dockerd.

Gsam333, 2023-02-23:
> 

谢谢我试试自己编译包

huangwb8, 2023-02-23:
@Gsam333 好的。我博客的学习地图有[NAS系列](https://blognas.hwb0307.com/map)写过一些文章。在[NAS系列 准备安装PVE系统](https://blognas.hwb0307.com/nas/3337)中我有讲到过一些和OpenWrt安装相关的东西。使用过程暂时还没有出教程。感兴趣的话也可以多加关注喔！

zenglianjie, 2023-03-23:
> 

> opkg install docker是可以安装成功的， docker -v Docker version 20.10.22, build 3a2c30b. 但docker是没有启动，运行/etc/init.d/docker start -ash:/etc/init.d/docker: not found。 运行其他docker命令都是提示Cannot connect to the Docker daemon at unix:///var/run/docker.sock. Is the docker daemon running?

请问你最后是怎么解决的呢?