coolsnowwolf/lede #10866: 编译带的qbittorrent-nox无法启动

danuonuo, 2023-02-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译带的qbittorrent-nox无法启动，webui打不开

### (2) 路由器型号和固件版本

A：小米路由器3G  OpenWrt SNAPSHOT, r5725-60178fd03  OpenWrt R23.2.14 / LuCI Master (git-23.025.27564-bfa4697)

### (3) 详细日志

A：QObject::connect: signal not found in QLocalServer
QObject::connect: signal not found in QTimer
QObject::connect: signal not found in Application
QObject::connect: signal not found in QTimer
QObject::connect: signal not found in QNetworkAccessManager
QObject::connect: signal not found in QNetworkAccessManager
QObject::connect: signal not found in QTimer
QObject::connect: signal not found in QTimer
QObject::connect: signal not found in QTimer
QObject::connect: signal not found in QNetworkConfigurationManager
QObject::connect: signal not found in QNetworkConfigurationManager
QObject::connect: signal not found in QNetworkConfigurationManager
QObject::connect: signal not found in QNetworkConfigurationManager
QObject::connect: signal not found in QThread
QObject::connect: signal not found in QTimer
QObject::connect: signal not found in DownloadHandlerImpl
QObject::connect: signal not found in QNetworkReplyHttpImpl
QObject::connect: signal not found in QFileSystemWatcher
QObject::connect: signal not found in QTimer
QObject::connect: signal not found in QTimer
QObject::connect: signal not found in QThread
QObject::connect: signal not found in QTimer
QObject::connect: signal not found in QThread
QObject::connect: signal not found in QThread
QObject::connect: signal not found in QTimer
QObject::connect: signal not found in QThread
QObject::connect: signal not found in QTimer

******** Information ********
To control qBittorrent, access the WebUI at: http://localhost:8080

The Web UI administrator username is: admin
The Web UI administrator password has not been changed from the default: adminadmin
This is a security risk, please change your password in program preferences.
QObject::connect: signal not found in QThread
QObject::connect: signal not found in QTcpSocket
QObject::connect: signal not found in QTcpSocket
QObject::connect: signal not found in QTcpSocket
QObject::connect: signal not found in QTcpSocket
QObject::connect: signal not found in QTcpSocket
QObject::connect: signal not found in QTcpSocket

WYC-2020, 2023-02-07:
l大的qBittorrent  依赖早就该更新了 他依赖的是 rblibtorrent，这个不知道多老了,我这边是改成依赖libtorrent-rasterbar，同时增加boost,我这边按这样编译出来运行正常
[image]
[image]

danuonuo, 2023-03-04:
> l大的qBittorrent 依赖早就该更新了 他依赖的是 rblibtorrent，这个不知道多老了,我这边是改成依赖libtorrent-rasterbar，同时增加boost,我这边按这样编译出来运行正常 [image] [image]

大佬可以考虑提交一个pr修复这个问题吗