coolsnowwolf/lede #10871: 编译出错

hahazhw, 2023-02-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译出错

### (2) 路由器型号和固件版本

A：新三 d2

### (3) 详细日志

A：net/sched/Kconfig:45: warning: menuconfig statement without prompt
net/Kconfig:478:warning: multi-line strings not supported
  CALL    scripts/atomic/check-atomics.sh
  CALL    scripts/checksyscalls.sh
  CHK     include/generated/compile.h
  CC      lib/decompress.o
  AR      lib/lib.a
  EXPORTS lib/lib-ksyms.o
  AR      lib/built-in.a
  GEN     usr/initramfs_data.cpio.lzma
v2ray ../init.d/v2ray 777 0 0' line 483
make[6]: *** [usr/Makefile:63: usr/initramfs_data.cpio.lzma] Error 255
make[5]: *** [Makefile:1739: usr] Error 2
make[5]: Leaving directory '/home/harvey/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.230'
make[4]: *** [Makefile:45: /home/harvey/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.230/.image] Error 2
make[4]: Leaving directory '/home/harvey/lede/target/linux/ramips'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/home/harvey/lede/target/linux'
time: target/linux/install#60.90#53.36#95.24
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory '/home/harvey/lede'
make[1]: *** [target/Makefile:24: /home/harvey/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/harvey/lede'
make: *** [/home/harvey/lede/include/toplevel.mk:231: world] Error 2

346673164, 2023-02-07:
拉取最新代码，遇到了类似的问题
vmlinux vmlinuz modules
make[5]: Entering directory '/data/workspace/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.231'
scripts/kconfig/conf  --syncconfig Kconfig
net/sched/Kconfig:45: warning: menuconfig statement without prompt
net/Kconfig:478:warning: multi-line strings not supported
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  CHK     include/generated/compile.h
  GEN     usr/initramfs_data.cpio.lzma
Unrecognized dir format '/etc/rc.d/S99^Mv2ray ../init.d/v2ray 777 0 0' line 388
make[6]: *** [usr/Makefile:63: usr/initramfs_data.cpio.lzma] Error 255
make[5]: *** [Makefile:1739: usr] Error 2
make[5]: Leaving directory '/data/workspace/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.231'
make[4]: *** [Makefile:45: /data/workspace/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.231/.image] Error 2
make[4]: Leaving directory '/data/workspace/lede/target/linux/ramips'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/data/workspace/lede/target/linux'
time: target/linux/install#6.12#2.96#26.16
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory '/data/workspace/lede'
make[1]: *** [target/Makefile:24: /data/workspace/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/data/workspace/lede'
make: *** [/data/workspace/lede/include/toplevel.mk:231: world] Error 2

bytesyard, 2023-02-13:
rm -rf /home/uba/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.10.167/usr/initramfs_data.cpio*
make -C /home/uba/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.10.167 KCFLAGS="-fmacro-prefix-map=/home/uba/openwrt/build_dir/target-mips_24kc_musl=target-mips_24kc_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/uba/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mips-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sun Feb 12 03:28:05 2023" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/uba/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.10.167 CC="mips-openwrt-linux-musl-gcc"  all modules
make[5]: Entering directory '/home/uba/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.10.167'
  SYNC    include/config/auto.conf.cmd
net/sched/Kconfig:45: warning: menuconfig statement without prompt
net/Kconfig:486:warning: multi-line strings not supported
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  CHK     include/generated/compile.h
  GEN     usr/initramfs_data.cpio
v2ray ../init.d/v2ray 777 0 0' line 319
make[6]: *** [usr/Makefile:82: usr/initramfs_data.cpio] Error 255
make[6]: *** Deleting file 'usr/initramfs_data.cpio'
make[5]: *** [Makefile:1819: usr] Error 2
make[5]: Leaving directory '/home/uba/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.10.167'
make[4]: *** [Makefile:41: /home/uba/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.10.167/.image] Error 2
make[4]: Leaving directory '/home/uba/openwrt/target/linux/ath79'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/home/uba/openwrt/target/linux'
time: target/linux/install#11.79#5.96#17.33
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory '/home/uba/openwrt'
make[1]: *** [target/Makefile:24: /home/uba/openwrt/staging_dir/target-mips_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/uba/openwrt'
make: *** [/home/uba/openwrt/include/toplevel.mk:231: world] Error 2

coolsnowwolf, 2026-09-05:
失败发生在自定义 v2ray init 脚本生成 initramfs（`v2ray ... line 483`），不是内核 Kconfig 警告导致，且缺少脚本内容/配置，无法修复当前仓库，关闭为 not planned。

PR #10620 (2022-12-15): tools/dosfstools: fix PKG_SOURCE
Both mirrors provided in the Makefile only serve gzipped tarballs.

Fixes: #10871
Fixes: 9edfe7dd13d9 ("source: Switch to xz for packages and tools where possible")

Q：你知道这是`pull request`吗？(使用 "x" 选择)
* [x] 我知道
commit 4952f51d (2022-12-15): tools/dosfstools: fix PKG_SOURCE (#10620)

Both mirrors provided in the Makefile only serve gzipped tarballs.

Fixes: #10871
Fixes: 9edfe7dd13d9 ("source: Switch to xz for packages and tools where possible")
closed 2026-09-05 by coolsnowwolf